esphome/feature-requests #371: Tickless Idle/Automatic Light Sleep

silverchris, 2019-08-22:
**Describe the problem you have/What new integration you would like**
Tickless Idle support/Automatic Light Sleep

**Please describe your use case for this integration and alternatives you've tried:**
I am looking at this to reduce power consumption of my ESPHome nodes when they aren't doing anything. Currently I have them down to about 34mA, and I would like to see if I can get them lower. From a cursory glance through the code, it doesn't look like the esp power management is implemented at all? (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/power_management.html#overview).

Trying to avoid deep sleep, as I would like a fairly quick reaction if I need the nodes to do something.

**Additional context**
I am thinking about playing around with this myself, but figured I would ask first if anyone had looked into it previously. I am assuming some changes would need to be made to the scheduler, to ensure tasks are run?

OttoWinter, 2019-08-31:
From my testing, the main loop (which is running at 60Hz) is not the issue here. The main power usage comes from the WiFi peripheral.

You can try this by installing a development copy of esphome and changing the main loop frequency to something much lower like 0.1Hz - the power use will not drop noticably. See also `power_save_mode` for wifi block

silverchris, 2019-08-31:
Hey @OttoWinter, I don't think the copy of the Arduino environment for the ESP32 includes tickless idle, and automatic sleep by default, so I don't think reducing the main loop frequency would change power consumption too much, as the cores would just be running a NOP loop or something?

From other peoples research, it does look like it can get very low on the power usage https://blog.voneicken.com/2018/lp-wifi-esp32-dfs/

I have been trying to play with platformio to get it to build with these features turned on so I can try it out, though this is my first major experience with platformio, and it has a bit of a learning curve for me.

silverchris, 2019-09-01:
So, I managed to get a test done with this.
Measuring across a 1.2ohm resistor in line with the ESP32 3.3V in pin:
(Averaged by my Fluke 179 Multimeter)

Without PM enabled:
46.5mV/1.2ohms =  38.75mA

With PM Enabled:
8.2mV/1.2ohms = 8.3mA

The one notable hack, that probably makes this more of an artificial test is that I set the loop delay to a fixed 1 second interval.

silverchris, 2019-09-02:
Pushed a bit of my testing code to here https://github.com/silverchris/esphome/tree/auto-sleep

This [sdkconfig.h](https://gist.github.com/silverchris/a1c71150b7011d21257ff4cbb44bb353) needs to be manually dropped into the src folder when compiling a device

this is the yml file I am using for my testing.
```
esphome:
  name: test
  platform: ESP32
  board: esp32dev
  arduino_version: dev
  platformio_options:
          platform: https://github.com/platformio/platform-espressif32.git
          version: https://github.com/platformio/platform-espressif32.git#90747cc8b3d65149f3fc9c66515391a362de9fed
          board : esp32dev
          framework: arduino, espidf
wifi:
  ssid: ""
  password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
        port: 80
light:
  - platform: binary
    name: Light
    output: led_output
    id: led

# Example output entry
output:
  - platform: gpio
    id: led_output
    pin: GPIO2
```

A notable issue is that when the device goes into a light sleep, the GPIO pins all turn off. Which is why I added a bit of code to the binary output. Not sure if that is the best way to do it?

OttoWinter, 2019-10-13:
Oh that is really interesting! I did not expect such improvements!

I will look into this after 1.14 ships (don't have time before then). Yeah the GPIO switch thing will be an issue (and ideally it would be solved in esphal.cpp). I'm still a bit suspicious of how well this will work, I've had some bad experiences with Espressif's APIs (for example remote APIs breaking WiFi) so it will require a lot of testing to make sure it works well, but if it does it would be awesome!

JimmyJones79, 2020-01-28:
Curious if anything has been done with this.  I have a LILYGO TTGO T-Camera ESP32 WROVER that I am running off of batteries that I would love to get the power consumption down on so this post caught my eye.

silverchris, 2020-05-26:
I did a couple of things to clean it up a little bit in https://github.com/silverchris/esphome/tree/auto-sleep

Probably not ready for a pull request or anything, but is working on the few devices I am testing with. They only use a toggle switch input, and the LEDC output, so nothing complex. 

```
esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit
  platformio_options:
    framework: espidf, arduino
    platform: espressif32@1.11.2

wifi:
  ssid: ""
  password: ""

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""
  
pm:
  min_freq_mhz: 40
  max_freq_mhz: 80

light:
  - platform: monochromatic
    name: ${upper_devicename} Light
    output: led_output
    id: led

# Example output entry
output:
  - platform: ledc
    id: led_output
    pin: GPIO32
    frequency: 32000 hz
    max_power: 0.60
    
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO33
      mode: INPUT_PULLUP
    name: ${upper_devicename} Switch
    on_press:
      then:
        - light.toggle: led
```

closed 2022-07-01 by nagyrobi

nagyrobi, 2022-07-01:
You provided it above. Feel free to make a PR.