esphome/feature-requests #372: Expose BLE devices as seperate devices

deece, 2019-08-22:
**Describe the problem you have/What new integration you would like**
If a room is assigned to an ESPHome BLE gateway in Home Assistant, all Xiaomi BLE temperature devices on that gateway are assigned that room. If no room is assigned, there is no way to assign a room to the individual entities.

**Please describe your use case for this integration and alternatives you've tried:**
The suggestion from the Home Assistant community is for ESPHome to expose the BLE devices as independent devices so they can each be assigned a room.

**Additional context**
Related issue for Home Assistant: https://github.com/home-assistant/home-assistant/issues/25089

antoni-k, 2019-08-24:
This is exactly the same problem that I have. When collecting data from several sensors located in several different rooms, it would make sense to assign them to different areas in HASS. I would suggest a feature to group ESPhome devices (sensors/switches/etc) into groups that would be displayed as separate devices in HASS, which would make integrating external BLE sensors much easier and intuitive.

OttoWinter, 2019-08-31:
This is more an issue with HA.

Quite a few integrations in HA suffer from this problem where a device that happens to be located in several rooms exists. And currently, there's no way to override the "Device" an entity belongs to.

Fixing it on the HA side would only be a simple matter of creating a new config file system to manually re-assign devices.

In ESPHome, you would not only have to do that, you would also need to teach esphome about HA's concept of devices etc. And that's just too much bloat to include in the esphome codebase for something that should be solved on the other side.

closed 2019-08-31 by OttoWinter