esphome/feature-requests #377: Feature Request: Device Documentation in Dashboard

danielschramm, 2019-08-25:
**Describe the problem you have/What new integration you would like**
I really like the ESPHome Dashboard but I'm missing one functionality.
I would like to see the actual entity names or functionality there since it is not always clear from my device names.
(Example Sonoff 4 Ch providing 4 different controls. If you have 10 of them it's a mess to remember which channel is providing which signal or an external documentation is needed.)

**Please describe your use case for this integration and alternatives you've tried:**
Two possibilities to implement it:
1. **is my favorite:** just include a "Comment" Field in "esphome:" section which can be used to add custom text / html to the overview. This gives huge degree of freedom for all kind of documentation without breaking the approach to keep the dashboard as simple as possible
2. automatically determine all entity names provided by a device and show them beside the device name. I think this is to specific and complicated as you already commented for #132 .

**Additional context**

danielschramm, 2019-08-26:
Seems to be very similar to #353.

OttoWinter, 2019-08-31:
Closing as duplicate of #353

closed 2019-08-31 by OttoWinter