esphome/feature-requests #378: Support

dedepene, 2019-08-26:
**Describe the problem you have/What new integration you would like**

Not so much a feature request but a desperate attempt to get some info I couldn't source elsewhere.

Most importantly is the ShellyRGBW2 flashable with esphome? Is the procedure pretty much similar to the other Shellys?

My main issue is trying to connect the Shelly running esphome to a CCT (ww+cw) led strip - only cold and warm white channels. My understanding is that in order to use the cwww platform I need to set up two output components first for each channel. My question - how do I identify the pin numbers to set those up, please? 
**Please describe your use case for this integration and alternatives you've tried:**

Use a CCT led strip with ShellyRGBW2 and esphome.
**Additional context**

glmnet, 2019-08-26:
If you cannot find information about it anywhere, you can tear apart the device and put detailed pictures here and we can move forward in identifying how it works.

dedepene, 2019-08-26:
> If you cannot find information about it anywhere, you can tear apart the device and put detailed pictures here and we can move forward in identifying how it works.

I'm trying to get in touch with the manufacturer, but maybe you can help me formulate the right question I need to be asking them. Please take a look at this [wiring diagram](https://shelly.cloud/wp-content/uploads/2019/02/DC_4xWhite_led_strip_shelly.png) . In order to figure out the output component configuration I need to know which ESP GPIO the individual R,G,B,W connection terminals correspond to, right? Is this what I should be asking them? Thanks!

glmnet, 2019-08-26:
> I need to know which ESP GPIO the individual R,G,B,W connection terminals correspond to, right?

yes, assuming they are driven directly by esp8266 pwm outputs, like in the H801 (very likely)

glmnet, 2019-08-26:
you can also just try all the valid values until you find which one is each.

glmnet, 2019-08-26:
I just googled it for you: https://bitekmindenhol.blog.hu/2019/03/30/shelly_rgbw2

dedepene, 2019-08-26:
> I just googled it for you: https://bitekmindenhol.blog.hu/2019/03/30/shelly_rgbw2

Ah, the google-fu is strong with you! Thank you! I swear I searched hard before posting here...

dedepene, 2019-08-27:
OK, I just bought this device and trying to flash esphome as a generic ESP8266 i get: 
Any ideas how to work around this?

> A fatal error occurred: MD5 of file does not match data in flash!

`Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/shellyrgbw2/.pioenvs/shellyrgbw2/firmware.elf
Checking size /data/shellyrgbw2/.pioenvs/shellyrgbw2/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.7% (used 31712 bytes from 81920 bytes)
PROGRAM: [===       ]  30.8% (used 315792 bytes from 1023984 bytes)
========================= [SUCCESS] Took 9.51 seconds =========================
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /data/shellyrgbw2/.pioenvs/shellyrgbw2/firmware.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:66:1c:be
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 2MB
Flash params set to 0x0330
Compressed 319936 bytes to 224604...
Wrote 319936 bytes (224604 compressed) at 0x00000000 in 20.0 seconds (effective 128.1 kbit/s)...
File  md5: d4c789de6a52f47a8782a4a6bffbf5dc
Flash md5: 8697620246a9c94155377b3383276099
MD5 of 0xFF is 326796cd86772179c72917d6041cd1c0

A fatal error occurred: MD5 of file does not match data in flash!`

OttoWinter, 2019-08-27:
Ok, this appears to be resolved since it's already supported. Feel free to create a guide for this device in the docs.

closed 2019-08-27 by OttoWinter

dedepene, 2019-08-28:
Yes, I can confirm the Shelly RGBW2 flashed just fine with ESPHOME, the GPIO mapping from the Hungarian blog post is accurate and the light platforms work. Thanks for the help @glmnet !