esphome/feature-requests #380: Support for output.ledc.set_frequency

DasBootU96, 2019-08-30:
**Describe the problem you have/What new integration you would like**

Support for output.esp8266_pwm.set_frequency exists, but output.ledc.set_frequency does not. Can this be added?

**Please describe your use case for this integration and alternatives you've tried:**

I can get a single tone, but then tried this to get multiple tones, but since "set_frequency" is appears not implemented for ledc, I get error: Unable to find action with the name 'output.ledc.set_frequency'.

output:
  - platform: ledc
    pin: GPIO19
    id: 'buzzer'

switch:
  - platform: template
    name: buzzzzzz
    turn_on_action:
      then:
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 800Hz
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer
        - delay: 200ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 600Hz
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer

**Additional context**

Want buzzer to generate a few Star Wars- The Imperial March tones.

tdamsma, 2019-09-04:
I was about to file a feature request for exactly this, but see the work has already started. I am impressed once again, keep up the god work!

closed 2019-10-18 by OttoWinter

martin3000, 2019-10-30:
what version of esphome do I ned to use set_frequency?

glmnet, 2019-10-31:
> what version of esphome do I ned to use set_frequency?

Not available in 1.13. You need 1.14 beta or dev

DasBootU96, 2019-10-31:
Tested with dev and It works quite well with a cheap passive buzzer, thanks a 1000000! 

```
script:
  - id: ImperialMarch
    then:
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 440
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 440
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 440
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer
        - delay: 20ms        
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 349
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 350ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 523
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 150ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 440
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 500ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 349
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 350ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 523
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 150ms
        - output.turn_off: buzzer
        - delay: 20ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 440
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 650ms
        - output.turn_off: buzzer

  - id: SuperMario
    then:
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 660 
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 150ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 660 
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 300ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 660
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 300ms        
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 510
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 100ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 660
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 300ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 770
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer
        - delay: 550ms
        - output.ledc.set_frequency:
            id: buzzer
            frequency: 380
        - output.set_level:
            id: buzzer
            level: 50%
        - delay: 100ms
        - output.turn_off: buzzer

switch:
  - platform: template
    name: Buzz Imperial March
    icon: "mdi:bell-ring"
    turn_on_action:
      - if:
          condition:
            - script.is_running: ImperialMarch
          then:
            - script.stop: ImperialMarch
            - script.execute: ImperialMarch
          else:
            - script.execute: ImperialMarch    

  - platform: template
    name: Buzz Super Mario
    icon: "mdi:bell-ring"
    turn_on_action:
      - if:
          condition:
            - script.is_running: SuperMario
          then:
            - script.stop: SuperMario
            - script.execute: SuperMario
          else:
            - script.execute: SuperMario
```

glmnet, 2019-10-31:
Some ESPHome cooked code for imperial March and Super Mario!

Now I need more buzzers!!!

OttoWinter, 2019-10-31:
Dammit, that's awesome 🥇 

Now I need to make this easier 🤣:

```yaml
# Imperial March
- output.ledc.song:
    id: buzzer
    active_level: 50%
    pattern:
      - {note: A, duration: 500ms}
      - {space: 20ms}
      - {note: A, duration: 500ms}
      - {space: 20ms}
      - {note: A, duration: 500ms}
      - {space: 20ms}
      - {note: F, duration: 350ms}
      - {space: 20ms}
      - {note: C5, duration: 150ms}
      - {space: 20ms}
      - {note: A, duration: 500ms}
      - {space: 20ms}
      - {note: F, duration: 350ms}
      - {space: 20ms}
      - {note: A, duration: 650ms}
```

martin3000, 2019-10-31:
Is this also available in HA as a service call?

OttoWinter, 2019-10-31:
@martin3000 Please see https://esphome.io/components/api.html#user-defined-services