esphome/feature-requests #381: Support for Xiaomi Cleagrass e-ink sensor

bistory, 2019-08-30:
**Describe the problem you have/What new integration you would like**

The recent Xiaomi Cleagrass e-ink sensor is not supported by ESPHome while Xiaomi LYWSD02 is in the latest commits.

**Please describe your use case for this integration and alternatives you've tried:**

I have already tried to use Mijia and LYWSD02 implementations but none of them work, data are always empty.

**Additional context**

sermayoral, 2019-09-06:
I would like to get this device working on ESPHome too. Thanks!!!

Fenixekb, 2019-09-07:
Greetings! Really looking forward to this integration!

zsiti, 2019-09-09:
+1 I'd much appreciate the integration of the Xiaomi ClearGrass temperature and humidity sensor with e-Ink display!

Kvasenok, 2019-09-13:
Well, I have this sensor and an ESP32 lying around. I could test the code if someone can explain me how))

sermayoral, 2019-09-13:
Thanks @Kvasenok.

I suppose you must compile the full code. I'm not sure how to do it, but i'm sure anybody here could help us.

Attached is the code. The configuration to get data of this sensor is the following:

# Example configuration entry
esp32_ble_tracker:

sensor:
  - platform: xiaomi_cleargrass
    mac_address: XX:XX:XX:XX:XX:XX
    temperature:
      name: "Xiaomi Cleargrass Temperature"
    humidity:
      name: "Xiaomi Cleargrass Humidity"
    battery:
      name: "Xiaomi Battery Humidity"

[esphome-dev.zip](https://github.com/esphome/feature-requests/files/3609281/esphome-dev.zip)

kammerjaeger, 2019-09-14:
Hi,
I have four of them and could try it.
I tried to get your code running but could not find any files in the archive with the string "xiaomi_cleargrass".
Are you sure that this is the right archive? 
As far as I understood the documentation there should be a folder esphome\components\xiaomi_cleargrass
(I also have problems to set up the local development so I cannot compile it anyway)

sermayoral, 2019-09-14:
You are right. Wrong zip. Sorry...

I'm gonna fix the problem.

sermayoral, 2019-09-14:
Here is the correct code :-)

[esphome-dev-xiaomi-cleargrass.zip](https://github.com/esphome/feature-requests/files/3613188/esphome-dev-xiaomi-cleargrass.zip)

kammerjaeger, 2019-09-20:
Thanks for the file.
It is not working perfectly yet one problem was the line:

bool is_cleargrass = (raw[1] & 0x30) == 0x30 && raw[2] == 0x47 && raw[3] == 0x03;

Also it seams that the newest firmware works (In Xiaomi Home 1.1.2_0020) but the 17 version did not work. I'm still trying to figure out what the problem is.

sermayoral, 2019-09-20:
Hi @kammerjaeger.

Try this:

bool is_cleargrass = raw[2] == 0x47 && raw[3] == 0x03;

If it is works, i will try to fix this part too

kammerjaeger, 2019-09-20:
Oh sorry I did not write it clearly.
The line that I wrote is working, you had:
bool is_cleargrass = (raw[1] & 0x20) == 0x20 && raw[2] == 0x47 && raw[3] == 0x03;
which I think should work too but is not what is send (and at the time I changed it, I still had the problem with the firmware)

It ran now for some time and I can say that it is showing values for the 1.1.2_0020 Version, what irritated me that the frames did only come irregular. But it seams that the device only sends updates on a value change, or something like that.

sermayoral, 2019-09-20:
Oh, thats great! I think this behaviour it's even better because you dont need to know that the device read the same temperature/humidity, saving as energy as posible.

Ok so, for the 1.1.2_0020 Version, what does it work? Everything? I mean, Temperature, Humidity & Battery information?

Thanks @kammerjaeger

kammerjaeger, 2019-09-20:
Yes everything works (temperature, humidity and battery) and I updated one of the sensors to the firmware 20 to test again that it does not work with the firmware version 17.

[00:55:00][D][xiaomi_ble:134]: Got Xiaomi Cleargrass (58:xx:xx:xx:xx:xx):
[00:55:00][D][xiaomi_ble:137]:   Temperature: 25.2°C
[0:55:00][D][xiaomi_ble:140]:   Humidity: 56.0%
sometimes it is also only temperature or humidity and not both
and
[00:28:08][D][xiaomi_ble:134]: Got Xiaomi Cleargrass (58:xx:xx:xx:xx:xx):
[00:28:08][D][xiaomi_ble:143]:   Battery Level: 65%

Thanks for your work.

sermayoral, 2019-09-20:
Great! You are welcome!!

I have the Xiaomi Mijia version and it has the same behaviour of the Xiaomi Cleargrass version. Sometimes it only receive Temperature, Humidity or both. Battery once in the blue moon... ☺️

So, Im going to prepare a Pull Request to upload this tested code @ottowinter @glmnet

sermayoral, 2019-09-23:
PR created #734

sermayoral, 2019-09-23:
[https://github.com/esphome/esphome/pull/734](url)

kammerjaeger, 2019-09-23:
Hi sermayoral,

I saw your PR, tank you fo taking the effort.
One question, was it intended to use the 0x20 in line 87 in the file xiaomi_ble.cpp?
The device sends 0x30.
[23:05:43][VV][esp32_ble_tracker:302]: Parse Result:
[23:05:43][VV][esp32_ble_tracker:319]:   Address: 58:XX:XX:XX:XX:XX (PUBLIC)
[23:05:43][VV][esp32_ble_tracker:321]:   RSSI: -36
[23:05:43][VV][esp32_ble_tracker:322]:   Name: ''
[23:05:43][VV][esp32_ble_tracker:330]:   Ad Flag: 6
[23:05:43][VV][esp32_ble_tracker:335]:   Manufacturer data: (0)
[23:05:43][VV][esp32_ble_tracker:336]:   Service data: 0x50.0x30.0x47.0x03.0x57.0xXX.0xXX.0xXX.0xXX.0xXX.0x58.0x06.0x10.0x02.0xDA.0x01.0x0A.0x10.0x01.0x62 (20)
[23:05:43][VV][esp32_ble_tracker:339]:   Service Data UUID: FE:FE95
[23:05:43][VV][esp32_ble_tracker:343]: Adv data: 0x02.0x01.0x06.0x17.0x16.0x95.0xFE.0x50.0x30.0x47.0x03.0x57.0xXX.0xXX.0xXX.0xXX.0xXX.0x58.0x06.0x10.0x02.0xDA.0x01.0x0A.0x10.0x01.0x62 (27)

sermayoral, 2019-09-23:
@kammerjaeger 

Fixed in <https://github.com/esphome/feature-requests/issues/381>

maccan-tech, 2019-09-27:
Hi.
Cant get my "ClearGrass Temperature and Humidity Monitor" to work.
Model: CGG1
Firmware 2.04 (got it from ClearGrass App)

[08:04:37][VV][esp32_ble_tracker:319]:   Address: 58:XX:XX:XX:XX:XX (PUBLIC)
[08:04:37][VV][esp32_ble_tracker:321]:   RSSI: -76
[08:04:37][VV][esp32_ble_tracker:322]:   Name: ''
[08:04:37][VV][esp32_ble_tracker:330]:   Ad Flag: 6
[08:04:37][VV][esp32_ble_tracker:335]:   Manufacturer data: (0)
[08:04:37][VV][esp32_ble_tracker:336]:   Service data: 0x08.0x01.0xXX.0xXX.0xXX.0xXX.0xXX.0x58.0x01.0x04.0xF8.0x00.0x3D.0x02.0x02.0x01.0x56 (17)
[08:04:37][VV][esp32_ble_tracker:339]:   Service Data UUID: FD:FDCD
[08:04:37][VV][esp32_ble_tracker:343]: Adv data: 0x02.0x01.0x06.0x14.0x16.0xCD.0xFD.0x08.0x01.0xXX.0xXX.0xXX.0xXX.0xXX.0x58.0x01.0x04.0xF8.0x00.0x3D.0x02.0x02.0x01.0x56 (24)

sermayoral, 2019-09-27:
@maccan-tech

I think you have a firmware problem. You get 24 bytes of data when @kammerjaeger receives 27 bytes of data.

@kammerjaeger, can you tell us how do you update the Cleargrass firmware? 

Thanks

kammerjaeger, 2019-09-27:
Hi,
I have the two apps installed:
"Cleargrass air" version 1.1.7 (122) and "Xiaomi Home" version 5.5.66.
In the cleargrass app the firmware update is mostly not working for me, it even shows a different firmware version then the xiaomi app. The Xiaomi app works a lot better. To see the device in the xiaomi app you have to use the Chinese mainland server which I don't like. The other server don't show the cleargrass devices for me. I hope that this will change at some point.
If you have added the devices in there you can go to settings and update the firmware.
Tell me what worked for you.

vevsvevs, 2019-10-13:
Thanks for your efforts, @sermayoral! Is there any chance to get [735 ](https://github.com/esphome/esphome/pull/735) being merged?

sermayoral, 2019-10-13:
Hi @vevsvevs.

I hope @ottowinter can review it soon.

closed 2019-10-14 by OttoWinter