esphome/feature-requests #382: Output Minimum On Power

alexyao2015, 2019-09-01:
**Describe the problem you have/What new integration you would like**

When and only when an output component is on, the minimum power will be the specified amount.
**Please describe your use case for this integration and alternatives you've tried:**

For an output component, it would be extremely helpful to have a minimum power output when the output is on. In my use case with lights, the lights have be at a specific minimum brightness in esp8266_pwm for the lights to both turn on and not flicker.

Currently, there is no way to have the brightness be at 1% and have the light on and not flickering with my particular light.

**Additional context**

The current min_power affects the output at all times, not just when it is on.

OttoWinter, 2019-09-01:
That sounds like a pretty clear case of custom code here. Please see custom output component, with that it should be possible.

closed 2019-09-01 by OttoWinter

alexyao2015, 2019-09-01:
It's a standard [Tuya](https://blakadder.github.io/templates/feit_rgb_a19.html) bulb that I am having these issues with. I could only believe that other Tuya bulbs have similar issues since they use software pwm. It would be really beneficial if this was part of the core of esphome.

glmnet, 2019-09-01:
Outputs have a min and a max, min value = off I believe.
This looks more like a light component feature than an output component.
I haven't played with them much yet but you have [gamma_corrent](https://esphome.io/components/light/index.html) there which should help.

alexyao2015, 2019-09-01:
Gamma_correct totally fixed it.