esphome/feature-requests #383: RGBWW with FastLED

alexyao2015, 2019-09-01:
**Describe the problem you have/What new integration you would like**

I would like to combine a FastLED RGB light with a WW light to create an RGBWW light. 

**Please describe your use case for this integration and alternatives you've tried:**

Many Tuya bulbs use FastLED for the RGB component of the bulb ([ex](https://blakadder.github.io/templates/feit_rgb_a19.html)). Currently, there is no way to create an RGBWW light unless all outputs are pwm.

**Additional context**

OttoWinter, 2019-09-01:
FastLED will not support that - FastLED's developers are not developing much anymore, so this is probably not going to happen.

The NeoPixelBus library also doesn't support that yet, so it also won't happen that way right now.

As for the bulb, if it uses the sm16716 chip that will already be supported starting with the next version

closed 2019-09-01 by OttoWinter

kvvoff, 2019-10-05:
@ottowinter I think it means that 2 lights are connected: 1 fastled (rgb) and 2 through regular PWM.  In esphome and home assistant there will be 2 different "lights", but you want to combine them into one item.  And control from one place.  That is, just combine them into one (fastled + pwm) for ease of management.  It will be very helpful.