esphome/feature-requests #384: Add Mitusbishi Electric AV IR climate component

psnszsn, 2019-09-02:
**Describe the problem you have/What new integration you would like**

I would like to use ESPHome to control my Mitsubishi Electric AC, model msz-hr35vf.

**Please describe your use case for this integration and alternatives you've tried:**

I can control the AC using the Xiaomi Mi Remote app on my phone.
**Additional context**

https://github.com/ToniA/arduino-heatpumpir/blob/master/MitsubishiHeatpumpIR.cpp describes 4 different models, but i do not know which is the right one.

Lopton, 2019-09-02:
If you're interested in diving deeper you can actually control these Mitsubishi Units via an ESP32 and the serial port connection inside the unit. I have a repository here https://github.com/Lopton/mitipump . I'm looking at diving into EspHome in the near future and hoping I can add this directly to this project in the future.

glmnet, 2019-09-02:
Can you play a little with https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino and figure out which model it is? Then a custom IR sender for ESPHome can be made.

psnszsn, 2019-09-04:
> Can you play a little with https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino and figure out which model it is? Then a custom IR sender for ESPHome can be made.

I managed to get it to work using the link you provided after a few tries (i had to use a transistor for the led). It seems that there is only one Mitsubishi Electric model in the IRremoteESP8266 implementation.

glmnet, 2019-09-04:
Great, let me put together an ESPHome implementation for it then so you can integrate it with HA

closed 2019-12-31 by glmnet