esphome/feature-requests #386: [web server] No states on web server

froido, 2019-09-02:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Hi,

there is a problem with showing the states on the web server. There are no states shown in the web frontend, even though they are updating in the serial console. I tried this with a couple of sensors and binaries.

Config: Adafruit ESP32 Feather as standalone webserver

### YAML-File:
```
esphome:
  name: pump
  platform: ESP32
  board: featheresp32

web_server:
  port: 80

wifi:
  ap:
    ssid: "esp32-pump"
    password: "esp32-pump!"
  reboot_timeout: 0s
  
# Enable logging
logger:

sensor:
  - platform: esp32_hall
    name: "ESP32 Hall Sensor"
    update_interval: 2s
    id: bla

```

### Serial-Log:
[image]

### Web-Frontend:
[image]

brandond, 2019-09-02:
Is this a feature request, or a bug report?

froido, 2019-09-02:
This is a big report!

brandond, 2019-09-02:
You're in the 'feature requests' issue tracker, you should take this to https://github.com/esphome/issues/issues

closed 2019-09-02 by brandond