esphome/feature-requests #391: Support for more 1-Wire devices

randybb, 2019-09-05:
**Describe the problem you have/What new integration you would like**

Support for more Dallas/1-Wire devices. Few more are already supported in HA https://www.home-assistant.io/components/onewire/, some fancy are not:
- DS2406 | Dual Addressable Switch Plus 1Kb Memory
- DS2408 | 1-Wire 8-Channel Addressable Switch
- DS2413 | 1-Wire Dual Channel Addressable Switch
- DS2438 | Smart Battery Monitor
- DS2890 | 1-Wire Digital Potentiometer
full list: https://www.maximintegrated.com/en/pl_list.cfm/filter/21

For example on Loxone platform it is used often for intelligent switches:
- https://www.loxone.com/enen/kb/wiring-1-wire-devices/
- https://sedtronic.eu/en/unica-1wire/8-31-unica-1-wire-type-u1wtvd.html

**Please describe your use case for this integration and alternatives you've tried:**

Other options are bus extenders hooked to SPI or I2C bus that are supported already. Big disadvantage of these buses is limited distance. Sure, in case of I2C it is possible to use range extenders/bus buffer (P82B96), but this adds complexity to both sites of the bus.

**Additional context**

JayElDubya, 2019-10-08:
Especially interested in the addressable switches. I have a few applications where I only have three wires going to a low-voltage switch box. Would like to have power, ground, and a robust 1-wire bus to an addressable switch to control multiple relays.

randybb, 2019-10-08:
This is exactly what I want to achieve - these 8 channels switches will be used for controlling of remote relays mounted in distribution panel and the smaller version with 2 GPIOs for controlling status LEDs ("power on" and "locator" - as we have on servers / network switches) in sockets.
With a strong pull-up (needs to be supported by uC or DS2482 i2c to 1-wire bridge can handle it automatically), you can lower it to 2 wires with reliable communication up to 200 m. Pretty neat!

OttoWinter, 2019-10-17:
Hi, please only one device per feature request. ESPHome already has a onewire base (for dallas temperature sensors), so integrating them wouldn't be *that* hard. But of course you first need the hardware to test it etc.

closed 2019-10-17 by OttoWinter