esphome/feature-requests #392: Implement HVAC Action for Climate devices

marcelveldt, 2019-09-06:
**Describe the problem you have/What new integration you would like**
The current climate implementation let's you set the HVAC mode but it's current state (if it's cooling, heating or idle) is not represented in the hass state object. While exploring esphome code (bang_bang climate implementation) the info is actually there but private. Home assistant actually supports filling the hvac_action property for that: https://developers.home-assistant.io/docs/en/entity_climate.html

At the same time I see that the hvac "AUTO" mode is translated to "HVAC_MODE_HEAT_COOL" in the hass implementation but I think (personal preference I guess) that HVAC_MODE_AUTO is a better fit as the bang_bang implementation actually has logic to determine if it should cool or heat.

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to show the actual state of the climate entity in hass so I know if it's cooling, heating or idle. At the same time I think that the hvac mode heat_cool should be replaced with auto, at least for the bang_bang component.

**Additional context**
I will take a look at the code myself too but as this request impacts the whole chain, I leave it here for discussion.

glmnet, 2019-09-26:
> At the same time I see that the hvac "AUTO" mode is translated to "HVAC_MODE_HEAT_COOL" in the hass implementation but I think (personal preference I guess) that HVAC_MODE_AUTO is a better fit as the bang_bang implementation actually has logic to determine if it should cool or heat.

This discussion will never end, but they come up with [this](https://github.com/home-assistant/home-assistant/blob/3efdf29dfa178838002201eaea258db78e6fa2ba/homeassistant/components/climate/const.py#L16) so it is documented in AUTO mode you cannot set a temperature set point.

marcelveldt, 2019-09-26:
> This discussion will never end, but they come up with [...]

Ah, never saw that. In that case ignore my ramblings and leave it as HVAC_MODE_HEAT_COOL to stay with the Hass defaults.

closed 2019-10-18 by OttoWinter