esphome/feature-requests #393: full_update_every" option in Waveshare EINK 4.2inch.

melwinek, 2019-09-07:
**Describe the problem you have/What new integration you would like**

Please add the "full_update_every" option in Waveshare EINK 4.2inch.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

melwinek, 2019-09-08:
At https://esphome.io/components/display/waveshare_epaper.html it is written that esphome uses the library "Arduino Waveshare E-Paper library by Yehui (@soonuse)".

Soonuse wrote in readme: "However, these 3 e-paper displays: 1.54" B / 7.5 "/7.5" B cannot support partial update of the frame memory (that is, unable to completely update one frame by updating partial of it several times ). Therefore the Arduino examples for them can only display static images. The following products are not affected: 1.54 "/2.13" /2.13 "B / 2.7" /2.7 "B / 2.9" /2.9 "B / 4.2" /4.2 B. "

So partial update support in 4.2 should not be a problem.
Can it be done?

melwinek, 2019-09-20:
Can anything be done about it?

Yazzcat, 2019-10-10:
I would very much welcome this as well!