esphome/feature-requests #397: Implement BLE temperature sensor - Inkbird

danps1, 2019-09-09:
**Describe the problem you have/What new integration you would like**

The Inkbird IBT-6X is a pretty popular BBQ thermometer that has 6 probes and transmits temperature data over BLE to a phone app. 

[https://www.amazon.com/stores/page/0546CE73-22C5-4AF8-AF34-01A3A4EF3F4D?ingress=0&visitId=939e9d41-b8b3-44d2-93f2-d58471c6bcfd&lp_slot=auto-sparkle-hsa-tetris&store_ref=SB_A0622099Z7RNH0U4IISR](url)

Both the device and the app are good, but it would be even better if we could use ESPHome to harvest this data into HA and then run automations etc!

I have captured the bluetooth packets, and decoded the frames, but lack the programming skills to implement this into ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

This is the capture of a typical packet...

[image]

The frame headers seem to be the same each time, with the part that changes being the value at the end. These 12 bytes seem to correspond to the temperature reading of the 6 probes in celsius. Each 2 byte pair appears to be swapped, and then converted to decimal gives the temperature (10x) in °C.

For example, in this capture, I have probes 1 and 3 connected, and probe 1 is at 4°C and probe 3 is at 194°C...
```
28 00 - Probe 1 - 0x0028 = 40 decimal = 4°C
f6 ff - Probe 2 - disconnected
94 07 - Probe 3 - 0x0794 = 1,940 decimal = 194°C
f6 ff - Probe 4 - disconnected
f6 ff - Probe 5 - disconnected
f6 ff - Probe 6 - disconnected
```

**Additional context**

I imagine the syntax in esphome then being similar to a xiaomi mijia...
```
esp32_ble_tracker:

sensor:
  - platform: inkbird
    mac_address: 7A:80:8E:19:36:BA
    probe1:
      name: "Inkbird Probe 1"
    probe2:
      name: "Inkbird Probe 2"
    probe3:
      name: "Inkbird Probe 3"
    probe4:
      name: "Inkbird Probe 4"
    probe5:
      name: "Inkbird Probe 5"
    probe6:
      name: "Inkbird Probe 6"
```

dcec, 2020-01-21:
this project if useful?
https://github.com/dereulenspiegel/ibbq-gateway

jshridha, 2020-05-04:
@danps1 I just put together a project that can accomplish this, but it's written in python and works great on a raspberry pi. Check it out here:
https://github.com/jshridha/inkbird

It's still extremely beta since I just started on it a few days ago, but you may find it useful.

Dbroek, 2020-06-08:
i back this feature request. I have a ESPhome in the garden. Would be nice to integrate this way into Home Assistant. I use this meter most of the time in the garden close to the barbeque :D

AlBrough, 2020-08-18:
> i back this feature request. I have a ESPhome in the garden. Would be nice to integrate this way into Home Assistant. I use this meter most of the time in the garden close to the barbeque :D

I have my Inkbird linked through the Tuya app rather than the InkbirdSmart one, this way I have been able to integrate it with Home Assistant using Tuya integration. The issue I have is that the values are all 10x the temperature, 16.6 reads as 166 as other contributors have observed

anacruza, 2020-11-18:
This would be very nice indeed!

nrandell, 2021-04-20:
I've been experimenting with this and here's a very rough and rudimentary configuration that appears to be working with lots of debug!

I'll tidy up over the next few days and hopefully test it at the weekend on a proper bit of cooking!

```
sensor:
  - platform: template
    name: "iBBQ Temperature Probe 1"
    expire_after: 60s
    id: ble_sensor_1
    unit_of_measurement: "°C"
    accuracy_decimals: 1

  - platform: template
    name: "iBBQ Temperature Probe 2"
    expire_after: 60s
    id: ble_sensor_2
    unit_of_measurement: "°C"
    accuracy_decimals: 1

  - platform: template
    name: "iBBQ Temperature Probe 3"
    expire_after: 60s
    id: ble_sensor_3
    unit_of_measurement: "°C"
    accuracy_decimals: 1

  - platform: template
    name: "iBBQ Temperature Probe 4"
    expire_after: 60s
    id: ble_sensor_4
    unit_of_measurement: "°C"
    accuracy_decimals: 1

esp32_ble_tracker:
  on_ble_advertise:
    - lambda: |-
        if (x.get_name() != "iBBQ") return;
        ESP_LOGI("ble_adv", "New BLE device");
        ESP_LOGI("ble_adv", "  address: %s", x.address_str().c_str());
        ESP_LOGI("ble_adv", "  name: %s", x.get_name().c_str());
        ESP_LOGI("ble_adv", "  Advertised service UUIDs:");
        for (auto uuid : x.get_service_uuids()) {
            ESP_LOGI("ble_adv", "    - %s", uuid.to_string().c_str());
        }
        ESP_LOGI("ble_adv", "  Advertised service data:");
        for (auto data : x.get_service_datas()) {
            ESP_LOGI("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
        }
        ESP_LOGI("ble_adv", "  Advertised manufacturer data:");
        for (auto data : x.get_manufacturer_datas()) {
            ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
            if (data.uuid.contains(0, 0)) {

              int probe0 = (data.data[9] << 8) + data.data[8];
              int probe1 = (data.data[11] << 8) + data.data[10];
              int probe2 = (data.data[13] << 8) + data.data[12];
              int probe3 = (data.data[15] << 8) + data.data[14];
              ESP_LOGI("ble_data", "    - %f %f %f %f", probe0 / 10.0, probe1 / 10.0, probe2 / 10.0, probe3 / 10.0);
              if (probe0 < 60000) {
                id(ble_sensor_1).publish_state(probe0 / 10.0);
              }
              if (probe1 < 60000) {
                id(ble_sensor_1).publish_state(probe1 / 10.0);
              }
              if (probe2 < 60000) {
                id(ble_sensor_1).publish_state(probe2 / 10.0);
              }
              if (probe3 < 60000) {
                id(ble_sensor_1).publish_state(probe3 / 10.0);
              }

            }
        }
```

aegjoyce, 2021-04-26:
@nrandell wow, this is fantastic! Great work!

Had to make a slight tweak to your code otherwise all temps get published to the same sensor:

```
              if (probe0 < 60000) {
                id(ble_sensor_1).publish_state(probe0 / 10.0);
              }
              if (probe1 < 60000) {
                id(ble_sensor_2).publish_state(probe1 / 10.0);
              }
              if (probe2 < 60000) {
                id(ble_sensor_3).publish_state(probe2 / 10.0);
              }
              if (probe3 < 60000) {
                id(ble_sensor_4).publish_state(probe3 / 10.0);
              }
```

Otherwise this works perfectly! Tested on a new IBT-4XS and an ESP32-WROOM-32 from AZ-Delivery

nrandell, 2021-04-26:
Yes I found that and was going to post an update! I also found the expiry at 60 seconds was too short.

My next step is to work out how to create an esphome component and get this working!

swbouman, 2021-05-21:
Thanks for the setup! Very useful en welcome addition to home assistant! (ESPHome) my ibbq didn’t make it to connect if I was in the kitchen en the egg was already on. 

I was already looking for a solution. A stand-alone pi is really overkill. So this is the solution. 

A question I have is;

does the device update temperature after initial discovery? Mine stays stuck at initial values. The bt discovery is set at 300 s. The device doesn’t update in this 300s nor after the 300s. 

any idea? 

Is there an option to add the device as a connected bt device?

nrandell, 2021-05-22:
I've used mine a couple of times. I don't connect it with the inkbird app and in that mode it just appears to send out advertisments every 90 seconds or so that contain the temperature - which is all I need.

I'm sure there are ways to connect more fully and do things like control the alarms, but I was just looking for monitoring only.

So I can't help too much at the moment, but if someone does identify more details about the protocol and if the UK weather keeps letting us down I may have some rainy weekends to play 😁

swbouman, 2021-05-22:
I found this info on GitHub from another project. To receive realtime data we need to login first. I'm very new in programming and don't understand how to send data from the esp to ink bird.

initiate-login :: 
  write @CredentialsMessage to characteristic @AccountAndVerify
enable-realtime-data ::
  enable notifications on @ClientCharacteristicConfigurationDescriptor
  write @RealtimeDataEnableMessage to characteristic @SettingsData
set-unit :: Fahrenheit or Celsius
  when Fahrenheit write @UnitsFahrenheitMessage to characteristic @SettingsData
  when Celsius    write @UnitsCelsiusMessage to characteristic @SettingsData
request-battery-level ::
  write @RequestBatteryLevelMessage to characteristic @SettingsData
set-target-temp ::
  write @SetTargetTempMessage to characteristic @SettingsData

swbouman, 2021-05-22:
Here are some hardcoded messages:

@CredentialsMessage: (:byte[]) { 0x21, 0x07, 0x06, 0x05, 0x04, 0x03, 0x02, 0x01, 0xb8, 0x22, 0x00, 0x00, 0x00, 0x00, 0x00}
@RealtimeDataEnableMessage: (:byte[]) { 0x0B, 0x01, 0x00, 0x00, 0x00, 0x00}
@UnitsFahrenheitMessage: (:byte[]) { 0x02, 0x01, 0x00, 0x00, 0x00, 0x00}
@UnitsCelsiusMessage: (:byte[]) { 0x02, 0x00, 0x00, 0x00, 0x00, 0x00}
@RequestBatteryLevelMessage: (:byte[]) { 0x08, 0x24, 0x00, 0x00, 0x00, 0x00 }
A message exists to set target temperatures (low and high, the closest of which will be displayed on the screen)

@SetTargetTempMessage: (:byte[]) { 0x01, probe-nr, low[0:7bits], low[8:15bits], high[0:7bits], high[8:15bits] }
probe-nr is the probe to select, from 0 onwwards
low is a int16 signed integer for the lower temperature, in 10^-1 Celsius
high is a int16 signed integer for the higher temperature, in 10^-1 Celsius

gabrio79, 2021-05-27:
hey guys, I'm trying your suggestion, but i receive error message during validation process:
sensor.template: [source /config/esphome/iBBQ-Gateway.yaml:28]
  platform: template
  name: iBBQ Temperature Probe 1
  
  **This option requires component mqtt.
  expire_after: 60s [source /config/esphome/iBBQ-Gateway.yaml:30]**
  id: ble_sensor_1
  unit_of_measurement: °C
  accuracy_decimals: 1

any suggestions?

arekuominae, 2021-05-31:
> I've been experimenting with this and here's a very rough and rudimentary configuration that appears to be working with lots of debug!
>
> I'll tidy up over the next few days and hopefully test it at [...]

This is working great, thank you for posting this.  Would you know how to capture fahrenheit as opposed to celsius?

gabrio79, 2021-06-01:
I also trying your code, but I receve and error message relaterd to following command
`expire_after: 60s`
Any siggestion? (below error message)

INFO Reading configuration /config/esphome/iBBQ-Gateway.yaml...
Failed config

sensor.template: [source /config/esphome/iBBQ-Gateway.yaml:28]
  platform: template
  name: iBBQ Temperature Probe 1
  
  This option requires component mqtt.
  expire_after: 60s [source /config/esphome/iBBQ-Gateway.yaml:30]
  id: ble_sensor_1
  unit_of_measurement: °C
  accuracy_decimals: 1

herik63, 2021-07-11:
Hallo everyone, I also have a 4 probes model  so I tried the code above, it worked great, except that the number of the probe is sequential, a single probe attached to the 4th port is reported as probe1... not a great issue.
The funny thing is happening if i power on my previous 2 probes model... the temperature is reported as single number on the same probe1 :D
[image]
[image]

jooprzol, 2021-07-12:
> This is working great, thank you for posting this. Would you know how to capture fahrenheit as opposed to celsius?

You'd want to add a lambda function to each sensor:

`sensor:
  - platform: template
    name: "Solis Temperature Probe 1"
    filters:
      - lambda: return x * (9.0/5.0) + 32.0;
    id: ble_sensor_1
    unit_of_measurement: "°F"
    accuracy_decimals: 1`

VanFlipsen, 2021-07-13:
Based on the config from @nrandell i made an config for the Inkbird IBT-6x thermometer

if device is off or out of range values turning to "0" otherwise sensor is not available in Home Asisstant.

Full Yaml, change the values ​​to your likings.

Thanks @nrandell for making this possible.

````
substitutions:
  esp_name: esp32blt
  
esphome:
  name: ${esp_name}
  platform: esp32
  board: nodemcu-32s
  on_boot:
    priority: -10
    then:
      - switch.turn_off: annoyingled
      - lambda: |-
          {
          id(ble_sensor_1).publish_state(0);
          id(ble_sensor_2).publish_state(0);
          id(ble_sensor_3).publish_state(0);
          id(ble_sensor_4).publish_state(0);         
          id(ble_sensor_5).publish_state(0);
          id(ble_sensor_6).publish_state(0);         
          }
      - delay: 1 sec
      - switch.turn_on: annoyingled
      - delay: 1 sec
      - switch.turn_off: annoyingled
wifi:
  ssid: !secret WIFI_SSID2
  password: !secret WIFI2
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${esp_name} Fallback Hotspot"
    password: !secret WIFI_hotspot

captive_portal:

web_server:
  port: 80

logger:

api:

ota:

time:
  - platform: homeassistant
  
esp32_ble_tracker:
  on_ble_advertise:
    - lambda: |-
        if (x.get_name() != "iBBQ") return;
        ESP_LOGI("ble_adv", "New BLE device");
        ESP_LOGI("ble_adv", "  address: %s", x.address_str().c_str());
        ESP_LOGI("ble_adv", "  name: %s", x.get_name().c_str());
        ESP_LOGI("ble_adv", "  Advertised service UUIDs:");
        for (auto uuid : x.get_service_uuids()) {
            ESP_LOGI("ble_adv", "    - %s", uuid.to_string().c_str());
        }
        ESP_LOGI("ble_adv", "  Advertised service data:");
        for (auto data : x.get_service_datas()) {
            ESP_LOGI("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
        }
        ESP_LOGI("ble_adv", "  Advertised manufacturer data:");
        for (auto data : x.get_manufacturer_datas()) {
            ESP_LOGI("ble_adv", "    - %s: (%s)", data.uuid.to_string().c_str(), hexencode(data.data).c_str());
            if (data.uuid.contains(0, 0)) {

              int probe0 = (data.data[9] << 8) + data.data[8];
              int probe1 = (data.data[11] << 8) + data.data[10];
              int probe2 = (data.data[13] << 8) + data.data[12];
              int probe3 = (data.data[15] << 8) + data.data[14];
              int probe4 = (data.data[17] << 8) + data.data[16];
              int probe5 = (data.data[19] << 8) + data.data[18];              
              ESP_LOGI("ble_data", "    - %f %f %f %f %f %f", probe0 / 10.0, probe1 / 10.0, probe2 / 10.0, probe3 / 10.0, probe4 / 10.0, probe5 / 10.0);
              if (probe0 < 60000) {
                id(ble_sensor_1).publish_state(probe0 / 10.0);
              } else {
                id(ble_sensor_1).publish_state(0);                
              }
              if (probe1 < 60000) {
                id(ble_sensor_2).publish_state(probe1 / 10.0);
              } else {
                id(ble_sensor_2).publish_state(0);                
              }
              if (probe2 < 60000) {
                id(ble_sensor_3).publish_state(probe2 / 10.0);
              } else {
                id(ble_sensor_3).publish_state(0);                
              }
              if (probe3 < 60000) {
                id(ble_sensor_4).publish_state(probe3 / 10.0);
              } else {
                id(ble_sensor_4).publish_state(0);                
              }
              if (probe4 < 60000) {
                id(ble_sensor_5).publish_state(probe4 / 10.0);
              } else {
                id(ble_sensor_5).publish_state(0);
              }
              if (probe5 < 60000) {
                id(ble_sensor_6).publish_state(probe5 / 10.0);
              } else {
                id(ble_sensor_6).publish_state(0);
              }

            }
        }

switch:
  - platform: gpio
    name: "${esp_name}-Onboard-LED"
    pin: 2
    inverted: False
    id: annoyingled
  - platform: restart
    name: ${esp_name}-restart
    id: restart_switch
    
    
sensor:
  - platform: uptime
    name: "${esp_name}_Uptime Sensor"
    
  - platform: wifi_signal
    name: "${esp_name} WiFi Signal"
    update_interval: 60s

  - platform: template
    name: "iBBQ Temperature Probe 1"
    id: ble_sensor_1
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 2"
    id: ble_sensor_2
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 3"
    id: ble_sensor_3
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 4"
    id: ble_sensor_4
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 5"
    id: ble_sensor_5
    unit_of_measurement: "°C"
    accuracy_decimals: 0

  - platform: template
    name: "iBBQ Temperature Probe 6"
    id: ble_sensor_6
    unit_of_measurement: "°C"
    accuracy_decimals: 0
````

ajobbins, 2021-08-12:
> Based on the config from @nrandell i made an config for the Inkbird IBT-6x thermometer

Thanks for this - it works really well!

Does anyone know if the battery status of the device is also being pushed and could be captured? I get a battery status in the BBQ Go app, so assume it can communicate it.

nrandell, 2021-08-12:
The battery status can be got if we connect properly. At the moment we are only picking up the broadcast messages it sends out. If we connect then we get data more often, see battery status and can set the alarm information. However for simple monitoring some cooking remotely, I can't see much use for the additional functionality.

closed 2022-07-01 by nagyrobi