esphome/feature-requests #398: Hide build timestamp from text_sensor version

maartenpaauw, 2019-09-09:
**Describe the problem you have/What new integration you would like**

I would like to be able to remove the version build timestamp from the `version text_sensor` component.

**Please describe your use case for this integration and alternatives you've tried:**

I didn't try anything yet. But it would be nice if something like this is possible.

```yaml
text_sensor:
  - platform: version
    name: version
    hide_build_timestamp: true
```

**Additional context**

https://community.home-assistant.io/t/version-sensor-without-build-date-and-time/135895

glmnet, 2019-09-09:
May be the format string can be made configurable and have a few examples in docs

jarcikw, 2019-12-08:
@OttoWinter  is there any chance to remove timestamp from with lambda or any other way ?

closed 2020-07-11 by glmnet