esphome/feature-requests #400: Feature request, UV and Light sensors VEML 6040 / 6070 / 6075 / 7700

ways, 2019-09-09:
**Describe the problem you have/What new integration you would like**

I'd like to see these sensors supported. Apparently they are similar and can be combined in one plugin.

Plugin in ESP Easy is here: https://github.com/letscontrolit/ESPEasy/pull/2440

**Please describe your use case for this integration and alternatives you've tried:**

No alternatives tried. There doesn't seem to be any UV sensors supported.

**Additional context**

Link to sensors:
* https://www.adafruit.com/product/4162
* https://www.adafruit.com/product/2899
* https://www.adafruit.com/product/3964

rradar, 2019-09-13:
Partly duplicate of #217

OttoWinter, 2019-10-21:
Yeah, I will close this for https://github.com/esphome/feature-requests/issues/217 - Please only one device per feature request!

closed 2019-10-21 by OttoWinter