esphome/feature-requests #404: Running command line ESPHome works in Windows

robcazzaro, 2019-09-11:
**Describe the problem you have/What new integration you would like**
This is more of a documentation bug than a feature request, but the system auto-closed my bug due to the lack of a template (which in that case was rather annoying). Anyways...

I wanted to play with ESPHome on my Windows 10 machine before installing HASS on my RPi. The "Getting Started through Command Line" explicitly says no Windows, but everything works just fine under WSL, the Windows Linux Subsystem. Including the Dashboard. OTA flash works just fine, too. For local flashing, I simply used one of the many ESP flashing tools and the generated binary

Same installation steps, same everything, just under WSL. I think it would be good to add a "Windows 10" section, and remove the "(sorry, no windows)." from the Dashboard section

I also suggest adding, in the Installation section, the need to edit the PATH variable to include the ESPHome binary directory, otherwise things don't work properly (while searching I found a lot of people reporting bugs simply due to not having set PATH properly). It's critical for WSL

**Please describe your use case for this integration and alternatives you've tried:**
Needlessly limiting the audience. Since everything already works, it's a good idea to document it

**Additional context**

glmnet, 2019-09-11:
I use the esphome command line in windows 10 without wsl 
Python works on windows 
Esphome used to do some stuff with symlinks which windows did not like though. 

The dashboard is another story and it is very welcome a setup guide to get it working on wsl on windows 10, I hadn’t tried but id like to get it working. So did you know you can contribute just by using the link at the bottom “edit this page” go ahead and share the love.

robcazzaro, 2019-09-12:
I tried editing (using GITHub Desktop) and creating a pull request, but I  must have done something wrong or not have the right permissions

I'm enclosing my edited file here (renamed to txt as GitHub doesn't allow .rst files). It needs to be checked for formatting, but otherwise it's ready to go, if you can be so kind to commit this

[getting_started_command_line.rst.txt](https://github.com/esphome/feature-requests/files/3603144/getting_started_command_line.rst.txt)

As for using the dashboard, as long as you install everything from the start under WSL, it works just like the equivalent in Linux. The only downside is that COM port mapping in WSL is somewhat problematic, so the USB adapter won't show in the dashboard. But I flash locally in Windows using ESPHome-flasher, pointing at the bin file (in most recent Windows 10 WSL, you can access the WSL filesystem from Explorer and every system file selector using \\wsl$\Ubuntu-18.04\home\ (or starting "explorer.exe ." from within WSL