esphome/feature-requests #406: Exception/coredump handler and decoder

lobradov, 2019-09-13:
**What new integration you would like**

If device crashes for whatever reason, exception/core dump is lost unless device is connected via serial monitor. This makes is hard to debug software and hardware issues for devices that are "deployed" in their target locations (wall, roof...)

Having the ability to store exceptions / core dumps and analyse them afterwards could be of great help to pinpoint the problem (and improve software quality).

Analyzing dumps requires access to code in order for debugger to provide meaningful trace - something quite easily accessible with Hassio/CLI ecosystem provided by esphome.

**Please describe your use case for this integration and alternatives you've tried:**

Ideal use-case: 
- device crashes (3.6, not good but not terrible)
- upon next boot, native API informs Hassio plugin that there's dump available. 
- Hassio plugin shows "Decode crash" option in the device context 
- Once triggered, this option pulls the last crash dump, decodes it and displays summary for novice users (hardware / software problem ) and details that might be usable for bug report.

In parallel, CLI tool for same functionality could/should be provided for users not using Hassio plugin.

**Additional info**

ESP8266 exception handling "ecosystem": 
https://github.com/krzychb/EspSaveCrash
https://github.com/me-no-dev/EspExceptionDecoder

ESP32 core dump info: 
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_core_dump.h
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html

OttoWinter, 2019-10-22:
Yeah I looked at something similar some time ago (around the time we got the stacktrace decoder). As for the way it's presented to the user what you're proposing it a bit too complex (adding new API commands etc, especially considering that would not work for MQTT users)

A much simpler option for that would be to just print the stack values via the normal log mechanism - the client can then choose to decode that similar to how it's done right now over the serial interface. When the user performs an OTA upload, clear the fault.

But the main problem I encountered when I looked at it was where to store the crash logs. Yes, it needs to be stored in flash somewhere, but where exactly is hard to define. If I remember correctly stack traces could be up to 4kb long (default stack size), and that would need some region in flash where it could be stored without interfering with some other parts that need that space. For example, some stuff like restoring light state etc is already stored there. And to the other end of the flash space we soon reach the OTA partition (and RF calib).

rudshep, 2020-01-05:
Is the low level IP stack still vailable after a "crash"? What about dumping to Syslog? Syslog could be a useful debugging tool anyway - can monitor devices where Serial connection is not an option and API monitoring might be down with the main program.

OttoWinter, 2020-01-12:
@rudshep It is not (or not guaranteed to be). Lots of crashes are even caused by it (or, at least by WiFi-related events).
Plus we can't wait forever until the WiFi packet has been sent.
It has to be stored somewhere (RTC memory on ESP32, flash on ESP8266) and then dumped when it starts up again.