esphome/feature-requests #407: A platform type of "template" for fan component.

billchurch, 2019-09-13:
**Describe the problem you have/What new integration you would like**
I have an IR controlled fan that I would like to show up in HA as a fan, instead of a switch. Since it's IR controlled, I can't use any of the current platforms with the "fan" component.

**Please describe your use case for this integration and alternatives you've tried:**
For fans and similar devices which may be controlled by IR or RF and need to appear in HA properly

**Additional context**
Example Config of what I would want it to work with:
```
remote_transmitter:
  pin: D2
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: Dyson Fan Power
    id: dyson_power
    turn_on_action: 
    - remote_transmitter.transmit_raw:
        code: [2248, -808, 782, -1560, 782, -1560, 782, -787, 782, -786, 782, -778, 783, -773, 782, -769, 782, -825, 782, -799, 782, -795, 782, -791, 782, -782, 782, -778, 783, -773, 782, -1535, 782]
        carrier_frequency: 38khz
```

I still need to define speed and rotation, and some how map those back to a template function for fan as well...

glmnet, 2019-09-16:
I guess your request is fair.
There are workarounds like [this one ](https://github.com/esphome/esphome-docs/pull/218) you can get it working soon

nagyrobi, 2022-07-07:
Duplicate of https://github.com/esphome/feature-requests/issues/60