esphome/feature-requests #410: LinkIT 7697 platform support

Valcob, 2019-09-19:
**Describe the problem you have/What new integration you would like**

I would be so nice to have along with eps8266 and esp32 another not as well known platform MT 7697 which is so alike esp32 and in some cases even better.

**Please describe your use case for this integration and alternatives you've tried:**

It would be nice to reflash all xioami lights that have this module and bring them into ESPHOME family :) 
**Additional context**

As a good point of start would be the ref to Arduino lib and how Tos:
[Arduino Documentaion](https://docs.labs.mediatek.com/resource/linkit7697-arduino/en/developer-guide/limitations-of-linkit-7697)
and a PDF about the chip itself:
[MT 7697 Datasheet](https://s3-ap-southeast-1.amazonaws.com/mediatek-labs-imgs/downloads/1442bc4eeda95ee334e14bf3be262862.pdf?response-content-disposition=inline%3B%20filename%3DMT7697_Datasheet.pdf&X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Security-Token=AgoJb3JpZ2luX2VjEMb%2F%2F%2F%2F%2F%2F%2F%2F%2F%2FwEaDmFwLXNvdXRoZWFzdC0xIkcwRQIhAMTNCSBbBN2tJu11mI4rJon%2BaevCi6dlbRfWM%2FB9a2SpAiA8ETx9AtgRQeKa8K%2FEAGfc7qAqznKW9wTBiyo0X3ayyirtAwiP%2F%2F%2F%2F%2F%2F%2F%2F%2F%2F8BEAEaDDQ0OTY5Njk5MjA2NiIMdo9WTi3orXqxI9BLKsEDvJVaAkZ%2ForUrs5ixGz96t%2F4rZ67QXxNHcxN%2Bc7ZifAJwEfbTB%2BK2Y566fF90%2FjXZ9tGlDoeKGpe7rRbg4wtakjPSbXVrveNU3L539LTnfs9nIm8BAmKrpYkAeTA09vbPiyIHhTTVxxEfsUXn30ft3aAIKkXN%2Bcx5Rilhs1%2FL%2B1vmUgzw6StLfyxBiq7po%2FocZO%2B0U%2FbJ%2FRV2R4zUl0YsNaYZr7CRMHmWVdjo4lpQroNKDnQIE%2FgiS48scA%2Bxddd%2BcbDeeo8q%2FI27YR%2FGC85t%2FtK2X8vauwlRwweyz4Q3MFXCJZgnRldW4Rrjs43BYYzKrMTh3nky475JgqTSVSUoHlto1FTzxkHMpJIiYts2Y2G4UHJ5qWLzqGnE8oIc0TmYnBZJSGOU2MvhTDhEL8jplWtaRPkfK3bTwrCk%2B0K%2BfcdQMJ0ETsD89XMX%2BD1OaoMNH4E5ljR3OJ4h931texd566pGO4juMMxMZxHpHJk5GJMmwIrnuqaXQxhkKr%2BJ%2BBFlX83UzNosmR2n2tcs2XpF6DtIZ9Ri5kqsOPHkPRPgFTllz2jY7%2FhlLgS%2F3MovB7hFWGuX26JuMngH5i%2Bu4DjEEBYwieeI7AU6tAHVqSedtmn%2BjmUfql0L%2B%2F3lOeysLeMRcluQ5I%2BcJUhjVws5JHTsSRsYrcAkUFoBQy%2F7HochbjiJ0P52SM1m%2BqAjxDP9yaFmO%2Bj42NFbIMgunluuojjo1K35l5dC3CyjKFcoQ72N9U%2BnnzLH7ZRIqBtJXWvd8KXKJwgAHOnoGGWeQUQWnT4cWk9oL8dF%2FhpatRzdQnjNVkUiMbDJlcgVv%2F17T7xPGD0ZLbsUlV4K7FnL1xOQDWc%3D&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=ASIAWRNAHRNBAQY4XS63%2F20190918%2Fap-southeast-1%2Fs3%2Faws4_request&X-Amz-Date=20190918T143929Z&X-Amz-SignedHeaders=host&X-Amz-Expires=600&X-Amz-Signature=fc02d0ffedd6d2d2638c88d96454fe77ec620becdde76b84c9bb445b319ee290)

randybb, 2019-09-19:
It needs to be supported by PlatformIO first https://github.com/platformio/platformio-core/issues/992 (open since July 2017) and then, probably, it should not be a big problem.

OttoWinter, 2019-10-21:
No, ESPHome will only support ESPs (it's in the name of the project even).

Supporting the WiFi stack and some peripherals on the ESP8266/ESP32s is already enough work. Adding more platforms is not a path this project should go down.

closed 2019-10-21 by OttoWinter