esphome/feature-requests #411: No password for OTA via HTTP web page

tripplet, 2019-09-21:
**Describe the problem you have/What new integration you would like**
Enabling the HTTP component allows for an OTA update via the web page, this kind of defeats the purpose of having and OTA password and seems a bit unsecure by default.
A http basic protection for the OTA feature would be nice or the option to disable it for HTTP.

**Please describe your use case for this integration and alternatives you've tried:**
At least the documentation should mention this, for me this was kind of unexpected.
Generally I an option to protect the web page with a password would be nice.

**Additional context**

Posted as issue before [here](https://github.com/esphome/issues/issues/626)

randybb, 2019-09-26:
...and have option to enable OTA via web.

OttoWinter, 2019-10-20:
Closing per https://github.com/esphome/esphome/pull/674

closed 2019-10-20 by OttoWinter