esphome/feature-requests #414: Please add support for themes from Home Assistant

cjramseyer, 2019-09-30:
**Describe the problem you have/What new integration you would like**
Please add integration/support for HASSIO dark themes into ESPHome

**Please describe your use case for this integration and alternatives you've tried:**
Home Assistant support themes for most of its components, there is not currently an option to make ESPHome dark theme

**Additional context**
Dark themed content is much easier to look at when handling multiple things and having to negotiate many lines of script/text/code

OttoWinter, 2019-10-20:
Well, I personally just use dark reader - a plugin for my browser to convert pages to dark mode automatically. The esphome dashboard looks quite good with it.

I personally don't think this is high priority - yes it's a nice-to-have thing but not really worth spending lots of time on.

If someone wants to implement this, feel free.

cjramseyer, 2019-10-20:
The problem is that the rest of home assistant supports dark mode.  Esphome 
does not.  So I have to switch the dark mode plugin that I have on and off 
when switching back and forth between esphome and so.ething else in home 
assistant.

Most of the other plugins support the theme for. Hi.e assistant.

closed 2020-02-23 by cjramseyer