esphome/feature-requests #415: Is there an ESPHome equivalent to the ESPEasy Serial Server?

JunShine, 2019-09-30:
**Describe the problem you have/What new integration you would like**

Is there an ESPHome equivalent to the ESPEasy Serial Server?

**Please describe your use case for this integration and alternatives you've tried:**

I want connect a CC2530 coordinator via an ESP8266。like that https://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html
I wondered whether it would be possible to do the same using ESPHome。How to fill in the ESPHome configuration entry？

**Additional context**

minsuke, 2019-09-30:
You mean like ser2net component? i wish this can be applied serial devices, it will be connected with old devices which use old serial connection. Great fuction !!

JunShine, 2019-09-30:
Yes！ser2net component！

brandond, 2019-09-30:
ESPHome is primarily meant to work with Home Assistant, either via MQTT or API. While many of the components can be used standalone, some of the dedicated single-purpose functionality like acting as a generic serial gateway is absent. You might be able to cobble something together with lambdas and mqtt on_message triggers but it wouldn't be pretty.

If you had a specific use case that involved using a CC2530 to connect various devices to Home Assistant that might find more traction.

JunShine, 2019-10-01:
> ESPHome is primarily meant to work with Home Assistant, either via MQTT or API. While many of the components can be used standalone, some of the dedicated single-purpose functionality like acting as [...]
Yes，use zigbee2mqtt (https://www.zigbee2mqtt.io/  or  https://github.com/Koenkk/zigbee2mqtt)，Allows you to use your Zigbee devices without the vendors bridge or gateway.It bridges events and allows you to control your Zigbee devices via MQTT. In this way you can integrate your Zigbee devices with whatever smart home infrastructure you are using. Currently 380 devices are supported from 84 different vendors.( here https://www.zigbee2mqtt.io/information/supported_devices.html)
Zigbee2mqtt needs to communicate with zigbee devices via CC2531/CC2530. and I found that use the serial port in esp8266 to connect cc2531/cc2530 and use it as a coordinator to send data to the mqtt gate. on  https://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html

randybb, 2019-10-08:
This would be a great addition for these who run BTLE and RF433 gateways.

OttoWinter, 2019-10-17:
The big problem with that is that most protocols ESPHome supports (MQTT, native API, HTTP) are *not* streaming protocols, so a serial stream cannot be represented in them.

Also, while in principle I don't have anything against this, this would be an integration that would soon get uber-generic. Maybe some user wants to split packets somehow (based on some terminating character), maybe someone wants a different protocol etc.

This would lend itself much better to custom code (which ESPHome does support quite well). I'd be happy to accept a cookbook contribution with the code required to create a serial server. That way, if users want to customize behavior, they have full access to the code.

closed 2019-10-17 by OttoWinter