esphome/feature-requests #417: FastLED Gradient support

Snabbie, 2019-09-30:
**Describe the problem you have/What new integration you would like**
I would like to see support in ESPHome for the FastLED library gradient color palettes. See here: https://github.com/FastLED/FastLED/wiki/Gradient-color-palettes

**Please describe your use case for this integration and alternatives you've tried:**
I would like to use this to be able to quickly and maybe on the fly create static color gradients on ledstrips with individial pixel control. For example it starts red at the beginning but gradually changes and it will be blue at the end. 

Even better would be to be able to change this on the fly with home assistant, but right now the only possible solution (as far as I know) is to cut up the strip in individual partitions and set the color manually, but that takes more processing power and is not really user friendly.

**Additional context**

None

Dustin-Harrell, 2019-10-06:
Gradient color palettes would be an excellent addition to ESPHome.
Ambient lighting with subtle color/brightness shifts can easily be implemented with Palettes and a simple noise function:
https://github.com/FastLED/FastLED/blob/master/examples/NoisePlusPalette/NoisePlusPalette.ino

The ability to define Palettes in a ESPHome YAML and then use the across multiple Effects would be great.
PaletteKnife has a lot of great resources for predefined palettes:
http://fastled.io/tools/paletteknife/

Thanks for all the great work in ESPHome!

vins84, 2023-05-16:
Any update on this ? It would be really good to be able to set a gradient colour

a0s, 2023-05-24:
This was made with duct tape and bubblegum (c)
https://github.com/a0s/esphome-light-gradient