esphome/feature-requests #418: Allow hyphens in node name

dannysauer, 2019-09-30:
**Describe the problem you have/What new integration you would like**
Allow hyphens in node name.

**Please describe your use case for this integration and alternatives you've tried:**
Node names currently allow underscore, which is in the set of valid mDNS service identifier and DHCP client id characters, but not in the set of (strictly) valid DNS hostname characters.  Some people have DHCP servers which create dynamic DNS names, and DNS servers which are configured to strictly verify valid hostnames.

Rather than altering network topology to accommodate mDNS, I would like to simply use DNS name resolution to reach ESP nodes.  Cosmetically, I'd like to use a separator in the node hostnames.  Hyphen is the only functional character for this.

**Additional context**
There are, realistically, other options for resolving this issue:
 * I could use a subdomain specific to the ESP devices rather than a naming convention.  That's moderately inconvenient, as it means an additional configuration step rather than just letting the DHCP server hand out an address wherever on the network the device attaches
 * I could alter the DNS server to accept underscores.  The drawback to this is that future devices could create hostnames with underscores, and underscores are not valid in real hostnames, dang it. :D 
 * I could just mash the hostnames together instead of using hyphens.  But that decreases legibility, and reduced legibility increases the possibility of mistakes later on.
 * I could rearrange my network so multicast DNS worked reliably.  That's kind of a hassle, though. :)

After a quick glance, it looks like this is just a matter of adding a hyphen to the string at https://github.com/esphome/esphome/blob/7ad593d674e488e3abbfb3208326e3cbcd404d24/esphome/config_validation.py#L46 - I'll wait to open a PR in a few hours, because Hacktoberfest. 👕

dannysauer, 2019-10-01:
...and a matter of making C++ accept non-underscore characters in variable names.  Grumble. :D

brandond, 2019-10-02:
I asked about this a while back and Otto said no.

BirdAPI, 2019-10-05:
I recently started using esphome and this limitation was quite surprising to me coming from a PIO + esp-idf/arduino background. PlatformIO supports hyphens in the env name, so does DNS/mDNS, MQTT, etc. Sounds like a limitation of esphome's generated code.

martin3000, 2020-05-19:
when I use "name: test_5", the dns-name becomes "test-05". this is also confusing and maybe a reason why hyphens cannot be used because there would be a conflict.

closed 2020-07-29 by OttoWinter