esphome/feature-requests #419: Limit rate of state updates to HA

thoj, 2019-10-02:
I may have missed something but I've been searching all the docs and code I could find and nothing solves this without doing something special with lambadas for each sensor/controller. 

I have have a case where I'm using the bang bang climate controller on a very fast heating pad. I need to read the sensor every 100ms to avoid to much overshoot. The problem is that esphome is sending updates to HA every 100ms. I would like to limit the number of state updates to HA to like every 10 seconds or something. This avoids filling HA with to much unnecessary data. (The history graphs get so slow) 

I could maybe solve this by using lambadas to implement the bang bang controller myself and only sending updates when I want.  But I think a generic state update rate limiting function for components would be a good thing to have. 

I also considered using two bang bang controllers. One for HA and one fast internal one and just copying max/min and command over to the internal one. But it really seems like a hassle compared to a generic rate limit.

brandond, 2019-10-03:
You probably want one of these, with an eye towards the send_every option.
https://esphome.io/components/sensor/index.html#sliding-window-moving-average-exponential-moving-average

Note that you can still access the most recent value (as `id(sensor).raw_state` in lambdas) from other automation if you need to use it for a control loop.

thoj, 2019-10-03:
Thanks for trying to help. I know about the filters and use them extensively. the problem is that for the bang bang controller to work fast enough it needs values at short intervals. There is no way (afaik) to filter the state updates from the bang bang controller.

My test config below, this works great except for the thing i asked about. It send way to many state updates to HA: 

```yaml
logger:
  level: debug
  logs:
    switch: ERROR
    adc: ERROR
    sensor: ERROR

climate:
# how to filter state updates from this
  - platform: bang_bang
    sensor: pad1_temp
    default_target_temperature_low: 0 °C
    default_target_temperature_high: 100 °C
    id: pad1_controller
    name: Pad1 Controller
    heat_action:
      - switch.turn_on: pad1_heat
    idle_action:
      - switch.turn_off: pad1_heat
      
interval:
  - interval: 10ms
    then:
      - switch.turn_on: pad1_vcc
      - component.update: pad1_adc
      - switch.turn_off: pad1_vcc

sensor:
  - platform: resistance
    sensor: pad1_adc
    id: pad1_res
    configuration: DOWNSTREAM
    resistor: 75kOhm

  - platform: ntc
    sensor: pad1_res
    id: pad1_temp
    calibration:
      b_constant: 3950
      reference_temperature: 25°C
      reference_resistance: 100kOhm
    name: Pad 1 Int Temp
    
  - platform: adc
    id: pad1_adc
    pin: GPIO32
    attenuation: 11db
    update_interval: never
    filters:
     - exponential_moving_average: #Interval 10ms * 10 = 100ms interval on bang bang
         alpha: 0.1
         send_every: 10

switch:
  - platform: gpio
    id: pad1_heat
    pin: 2
  - platform: gpio
    id: pad1_vcc
    pin: 15
```

And again i do understand that i could implement the controller myself in lambadas to make this work but the climate components are very nice to use.

thoj, 2019-10-03:
After having a think on the filter chain. A combination of throttle and delta is not perfect but would substantially reduce the amount of data in my case: 

```yaml
    filters:
     - exponential_moving_average: #Interval 10ms * 10 = 100ms interval on bang bang
          alpha: 0.1
          send_every: 10
     - or:
        - throttle: 10s
        - delta: 0.5
```

brandond, 2019-10-03:
Just to be clear, it's specifically the climate component that's spamming HA, not the sensors - since those and the switches are nameless?

thoj, 2019-10-03:
Yes it's the climate component.  It seems to be sending at 10Hz, even if the state is the same.

glmnet, 2019-10-09:
As an ugly workaround you can have two `climate: ` components one does the work, the other faces HA, and uses a throttled template sensor.

thoj, 2019-10-09:
I was thinking of doing that, but how would you transfer the settings to the fast controller? Like mode, start temp and end temp

glmnet, 2019-10-09:
> I was thinking of doing that, but how would you transfer the settings to the fast controller? Like mode, start temp and end temp

https://esphome.io/components/climate/index.html#lambda-calls

OttoWinter, 2019-10-17:
Hmm, I don't think this is a feature to implement. Sure I understand the reason behind it, but it would be very messy architecture-wise.

Sensors are sensors and have specialized utilities just for that - sensor filters. I don't want to introduce the additional complexity of sensor filters into climate devices too. 

I do think there's working solutions to the problem here (delta filter on the base sensor, two climate devices, ...) - yes they're hacky but less hacky than the internals of the climate component would get.

closed 2019-10-17 by OttoWinter