esphome/feature-requests #422: OTA/Dashboard with a target IP not mDNS

Nornode, 2019-10-09:
**Describe the problem you have/What new integration you would like**

I'm unable to run updates OTA or see status in dashboard as my IOT network is separated from my LAN. The mDNS reliance is blocking my updates/log/status checks.

Background: my mDNS isn't working as expected and I have two separate (v)LAN's, a Standard and one for all IoT's and ESPs.

I can reach and add the flashed ESPs to Homeassistant, all fine.
But ESPhome command line and dashboard (running in a container) can't see the devices so I'm unable to update them OTA/see in the dashboard if they are online.

 
I tried to edit the json file in .esphome/config.yaml.json to have:

{
  "arduino_version": "espressif8266@1.8.0",
  "address": "192.168.12.51",
  "storage_version": 1,
  "esphome_version": "1.13.6",
...
...
However that didn't work and I still can't e.g. read logs from the dashboard... it still says: (LOG OUTPUT BELOW)

I also tried to run the container with the variable -e ESPHOME_DASHBOARD_USE_PING=true
Yet, I'm still not able to get things working properly.
The modified ".esphome/config.yaml.json" shows status=green but I can't read logs from, and the other modules are just red but can sometimes read logs?

I'd like to specify the ESP8266 microcontrollers ip in the dashboard and in the command line for direct access withouth an mDNS lookup and instead a direct IP connection like:

esphome config.yaml logs --ip 192.168.12.52

I have been unable to find any documentation on additional arguments the 'esphome' command can take.
Thanks!

glmnet, 2019-10-09:
you can OTA compile then upload and see logs via command line using ip:

`esphome file.yaml run --upload-port <ip>:<port>`

you can also just upload a previously compiled file using `upload` instead of `run`

you cannot see logs though, logs does not support 

in the other hand the dashboard is a different story, so this is why I say you are making a confusing request, it is not easy to track multi purpose issues / feature requests.

> I have been unable to find any documentation on additional arguments the 'esphome' command can take.

The getting started from command line will give you a coarse intro, but more detailed and updated info is available if you use `--help` after the verb

e.g.

`esphome file.yaml run --help`

OttoWinter, 2019-10-17:
You can also use the `use_address` option. See also https://esphome.io/components/esphome.html?highlight=use_address#changing-esphome-node-name

closed 2019-10-17 by OttoWinter