esphome/feature-requests #423: Allow DNS names in Yaml for such things as mqtt broker

aenertia, 2019-10-12:
**Describe the problem you have/What new integration you would like**

There are several places where variable of type 'Network' is required in the YAML - many if not all of these should also accept FQDN DNS entries. 

Here is an example for an mqtt broker of an error:

`mqtt: [source heatpumpcontroller.yaml:15]
  
  Component mqtt requires component network.
  broker: openhab.ae.net.nz
api: [source heatpumpcontroller.yaml:61]
`

**Please describe your use case for this integration and alternatives you've tried:**
dotted quad style IPv4 addresses can be used - have not attempted IPv6 entries as the esp's v6 stack is a bit immature. 

**Additional context**
FQDN's are valid and desirable for many types of configuration. There are many reasons to do this; to allow Highly Available endpoints, to allow Shifiting targets (i.e dynamic). For example my MQTT broker should always get populated in end devices by DNS FQDN so that I can Load-balance and/or move the IP it sits on simply.

OttoWinter, 2019-10-15:
> Here is an example for an mqtt broker of an error:
> Component mqtt requires component network.

That doesn't have anything to do with the broker setting here actually.

You just don't have any network connection configured (no wifi or ethernet entry) :)

closed 2019-10-15 by OttoWinter

aenertia, 2019-10-15:
That was just a snippet. I have a complete working config. It's to do with
the parser considering FQDN NOT a network - which seems to be a dotted quad
v4 IP

OttoWinter, 2019-10-15:
As I said, that error message means that you don't have any network source in your configuration.

Please look in your configuration and check if you have a wifi: section in there.

aenertia, 2019-10-15:
There is the standard wifi stanza (which works if I ommit or use a dotted
quad IP in the broker section ; so am not sure what you mean?

OttoWinter, 2019-10-17:
You're saying you get an error `Component mqtt requires component network.`, right? 

The only way that can turn up is if the `network` component is not loaded. And `network` is auto-loaded by wifi.

This error message is generated before the MQTT section is even validated. I bet if you put an IPv4 there instead you will see the same error.

Please post your full config (with secrets removed of course).