esphome/feature-requests #425: Single Temperature Bang-Bang Thermostat

rjmurph2241, 2019-10-12:
**Describe the problem you have/What new integration you would like**

Currently, the only way to set up a thermostat is to use the bang bang controller.  However, when this integrates with the Google home, it doesn't work very well (for example, no temperature set point is shown in heat mode)

It would be very convenient to be able to set a temperature and and internally set a range around that temperature for the bang bang.  For example, the temperature is set to 70F and the range is 1F so the bang bang controller has a range of 69-71F, but the UI only .  

**Please describe your use case for this integration and alternatives you've tried:**

I know this would be possible if I exposed the temperature and relay to Hassio and just used the generic thermostat in Hassio but I would much rather have the thermostats work without Hassio if possible. 
**Additional context**

AalianKhan, 2019-12-14:
@rjmurph2241 if you can you share your config for the generic thermostat?

Cirrus94, 2020-04-28:
I would also love to have the option to set only one specific target temperature instead of a range. Is there a chance that this is implemented?

Guyohms, 2020-06-11:
I agree, I wanted to migrate all my `generic_thermostat` from Home Assistant to ESPHome but this weird behavior of having 2 setpoint to change each time stopped me in the process.

I think the `bang_bang`thermostat should replicate what's done with the `generic_thermostat`

Cirrus94, 2020-06-12:
> I agree, I wanted to migrate all my `generic_thermostat` from Home Assistant to ESPHome but this weird behavior of having 2 setpoint to change each time stopped me in the process.
> 
> I think the `bang_bang`thermostat should replicate what's done with the `generic_thermostat`

There is already somebody (@kbx81) who is enhancing the component and adding a single setpoint feature here: https://github.com/esphome/esphome/pull/1061

kbx81, 2020-06-12:
I'm not 100% sure if what's implemented in the updated `bang_bang` controller that I'm working on does exactly what's described here, but I think it's really close if it's not an exact match.

If only a single action (heat **or** cool) is defined in the config, the component will only register a single available set point in HA; as such, HA only displays UI for adjusting a single set point. It also implements hysteresis as you've described.

Is that what you are looking for?

Cirrus94, 2020-06-12:
For me it‘s exactly what i was looking for! Thats the only reason why my DIY esphome thermostats wasn‘t exceeding the prototype phase so far...

kbx81, 2020-06-12:
Yeah, I'm working on the same thing and...yeah. 😄
I guess I'll link this as issue as "fixed" in my PR.

Guyohms, 2020-06-14:
Nice then !
I'll wait for your PR to be merged !
Thanks for the work done

glmnet, 2020-06-17:
Closing as pr got merged. Feel free to open another if needed.

closed 2020-06-17 by glmnet