esphome/feature-requests #430: Support for DIGOO DG-R8H weather module

dmonizer, 2019-10-14:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I'd like ESPHome to have functionality for DG-R8H weather sensors (implementing this would _probably_ mean support for other chinese cheap weather sensors as many of these probably share internals.
Link to device: https://www.banggood.com/Digoo-DG-R8H-433MHz-Wireless-Digital-Hygrometer-Thermometer-Weather-Station-Sensor-for-TH11300-8380-p-1178108.html?rmmds=search&cur_warehouse=CN

I am running ESPHome with on Sonoff RF mini, and have number of DG-R8H's available, so I am willing to give any info that I can and also test any future code.

Thank you.

randybb, 2019-10-15:
Have you tried decode these packets with digoo.cpp from https://github.com/dgomes/homeGW?

dmonizer, 2019-10-15:
No, despite numerous searches this has not poppedon my radar. Will give it a try asap.

If it works, can it be integrated into esphome?

dmonizer, 2019-10-16:
ordered RF receiver/transmitter for testing. takes a few days.

dmonizer, 2019-11-01:
Ok, modules finally arrived and got tested. That library seems to work well in the first glance.
Tested with 2 different sensors, both work well.
This is the output from my code:
22:47:25.979 -> digooid: 13 ch: 2 batt: 1 temp: 23.30 hum:58 raw:0xd90e9f3a
22:47:26.112 -> digooid: 13 ch: 2 batt: 1 temp: 23.30 hum:58 raw:0xd90e9f3a
22:47:26.277 -> digooid: 13 ch: 2 batt: 1 temp: 23.30 hum:58 raw:0xd90e9f3a
22:47:26.410 -> digooid: 13 ch: 2 batt: 1 temp: 23.30 hum:58 raw:0xd90e9f3a
22:47:26.576 -> digooid: 13 ch: 2 batt: 1 temp: 23.30 hum:58 raw:0xd90e9f3a
22:47:39.360 -> digooid: 6 ch: 1 batt: 1 temp: 25.00 hum:58 raw:0x680faf3a
22:47:39.492 -> digooid: 6 ch: 1 batt: 1 temp: 25.00 hum:58 raw:0x680faf3a
22:47:39.659 -> digooid: 6 ch: 1 batt: 1 temp: 25.00 hum:58 raw:0x680faf3a
22:47:39.791 -> digooid: 6 ch: 1 batt: 1 temp: 25.00 hum:58 raw:0x680faf3a
22:47:39.957 -> digooid: 6 ch: 1 batt: 1 temp: 25.00 hum:58 raw:0x680faf3a

code itself being simple:
------------- digoodump.ino ------------------
#include <homeGW.h>
#include <digoo.h>

HomeGW gw(1);
digoo digooStation;
uint64_t prev_p = 0;

#define RF_RECEIVER_PIN 2

void setup() {
  Serial.begin(115200);

  pinMode(RF_RECEIVER_PIN, OUTPUT);
  digitalWrite(RF_RECEIVER_PIN, LOW);
  pinMode(RF_RECEIVER_PIN, INPUT);
  digitalWrite(RF_RECEIVER_PIN, LOW);
  gw.setup(RF_RECEIVER_PIN);

  gw.registerPlugin(&digooStation);
}

void loop() {
  uint64_t p = 0;
  String output = "";
  if (digooStation.available())
    if ((p = digooStation.getPacket())) {
      if (p == prev_p) {
        Serial.print("digoo ");
        Serial.print("id: ");
        Serial.print(digooStation.getId(p));
        Serial.print(" ch: ");
        Serial.print(digooStation.getChannel(p));
        Serial.print(" batt: ");
        Serial.print(digooStation.getBattery(p));
        Serial.print(" temp: ");
        Serial.print(digooStation.getTemperature(p));
        Serial.print(" hum:");
        Serial.print(digooStation.getHumidity(p));
        Serial.print(" raw:");
        Serial.println(digooStation.getString(p));
        p = 0;
      }
      prev_p = p;
    }
}

------------- /digoodump.ino ------------------

what else could I do to get this included in esphome? I do not feel even close to confident enough to write the support myself - my main language is java and i barely managed this much of C :)

waiet, 2019-11-09:
I also needed to add support for dg-r8h, but also don't know very good C.
I think we need first to add sensor.py to /esphome/components/remote_receiver. And after then to add protocol for digoo to /esphome/components/remote_base. Can someone help us with that?

OttoWinter, 2019-11-09:
I would recommend looking at `climate_ir` types, for example `coolix` - it's basically also for receiving IR data.

Only difference is that coolix is for climate devices, but adapting it should not be too hard, as sensors just have one main method that needs to be called `publish_state()`

waiet, 2019-11-10:
Thank you @OttoWinter, but I was looking for coolix, but this component is only transmitting data. My main issue is how retrieve data. Could you help me how get it?  Should I create new component or edit remote-receiver?

Sorry for my questions but for me it's quite hard for understanding(C and python), because I'm java developer.

OttoWinter, 2019-11-12:
@waiet No, it is also receiving data, see `on_receive`

waiet, 2019-11-16:
I did some changes with that. github.com/waiet/esphome/commit/fb4756bf0ffb93a787a630730c160601da935eba
But I need help, because I don't know how to get packet in digoo_protocol.cpp.

dmonizer, 2019-11-17:
@waiet i purchased this https://www.banggood.com/433Mhz-RF-Transmitter-With-Receiver-Kit-For-Arduino-ARM-MCU-Wireless-p-74102.html for testing. But in general i'd like to have it integrated so I could use my Sonoff RF bridge.

dmonizer, 2019-11-24:
@waiet have you had any success developing it further?

waiet, 2019-11-24:
@dmonizer not. Can you check my changes https://github.com/waiet/esphome/commit/fb4756bf0ffb93a787a630730c160601da935eba ?
I need help with function decode. 
But I have issues with receiving data with raw protocol.

dmonizer, 2019-12-03:
I did look through, but to be honest, my C/python is severly lacking...

how could I "install" this branch? i did clone the repo, switched to branch, did "setup.py build" but trying to "setup.py install" fails ("You are attempting to install a package to a directory that is not
on PYTHONPATH and which Python does not read ".pth" files from.  The
installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was
    /home/erik/esphome-digoo/lib/python/
and your PYTHONPATH environment variable currently contains:
    '/home/erik/esphome-digoo'") ....

Gravitate1, 2020-02-28:
Has there been any progress with this? I have just ordered a dg-r8h and I am hoping to get it working with a sonoff rf bridge and esphome.

sergeolkhovik, 2020-03-11:
I purchased this sensor as well, hope ESPHome will support that before I receive the parcel, or will do what I can to help with the code.

waiet, 2020-03-12:
I have no progress. I don't know how decode signal.
@sergeolkhovik  what kind of 433mhz receiver do you have? rxb6?

sergeolkhovik, 2020-03-12:
No, I have Sonoff RF flashed with ESPHome.

sergeolkhovik, 2020-03-12:
I can buy any receiver and attach it to Arduino or ESP for decoding. Can you recommend me any?

dzkahn, 2020-04-16:
Some people have gotten the Digoo dg-r8h working with Sonoff bridge, maybe someone would like to add the protocol to Esphome.

Nexus433
https://github.com/aquaticus/nexus433
the protocol is defined in the code

this one is about getting it to work with Tasmota and costume rf firmware called  Portisch /
RF-Bridge-EFM8BB1 
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/119

The weather module works with OpenMQTTGateway but you can not read the weather module and door sensors at the same time, use the pilight gateway
https://docs.openmqttgateway.com/

https://community.openmqttgateway.com/t/digoo-dg-r8s-433mhz-battery-operated-temperature-and-humidity-sensor-with-embedded-display/287

drcommons, 2020-05-12:
@dzkahn I have been looking for examples of people getting the Digoo dg-r8h working with Sonoff RF Bridge but not able to find any. 
Are you able to post some links please because from what I have read, it does not yet work?

dzkahn, 2020-05-12:
No the Digoo dg-r8h does not work with Esphome.  It looks like people have gotten it to work using other software.  Such as this guy:
https://1technophile.blogspot.com/2019/04/sonoff-rf-bridge-pilight-or-how-to.html

Someone said that this works
https://github.com/merbanan/rtl_433

drcommons, 2020-05-13:
ok thanks, i'll check out those links

denisjoshua, 2020-10-17:
Hi there all, 
it there news about the DIGOO DG-R8H please ? 
I have 8 pieces and because I not understand this thinks I just wait that someone make it work. 
Thanks in advance

randybb, 2020-10-17:
8? :D Just be aware these 433 devices are really ancient - with huge power consumption (2xaaa batteries will least around 10 months), you can have only 3 connected devices, since you can select only channel 1-3, and other 433 devise will cause a loot of interference (like door bells, remote for garage doors,...). 
Better option are zigbee sensors (for example from xiaomi - I have these in every room) - much smaller, powered by cr2032 battery that will least more than 1 year and every devices has his unique ID , max number of devices depends from topology (you will need more "routers" connected to one "coordinator").

denisjoshua, 2020-10-18:
Thanks a lot for the answer Randy, 
but if you talk about [this one](https://www.amazon.it/Aqara-Temperatura-Ambientale-Intelligente-Connessione/dp/B07R9VCQ5N/ref=sr_1_1?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=xiaomi+sensore+temperatura&qid=1603041444&sr=8-1) I think it too much 17 EU. for every sensor I must install... I will need more than 8... about 40-50 pieces. 

So you tell me that I can't put more than 3 433 sensors in same place ? 
If I understand everyone have an ID (ID that change when change battery) 

Maybe Is better if I make from zero all the sensors by DIY... using temperature/humidity sensors and RF trasmiter with arduino. 

Thanks again

randybb, 2020-10-18:
Just checking other forums (openmqttgateway) - so I was wrong, it is transmitting its own ID, so this should be fine. But if you need 50 this could be a bit problem.

Xiaomi has two types of zigbee temperature sensors - the rectangular ones (temp, hum, pressure - I have this outside) for around 12 Euro, or older round ones (temp, hum - and these inside) is even cheaper. Or if you want to see temperature/humidity (more advanced alternative to Digoo R8S), then you can choose BTLE version (esphome can act as a gateway for these) with a display (some models have a clock as well). I use two types of zigbee gateways - one is a zigbee gateway from xiaomi (local API) and the second one is just a usb zigbee dongle directly connected to my HA server. Both are visible in HA => esphome can read any entity_id from HA.

I fully understand - these 433 devices are cheap, but these sensors are really ancient and after few tests I threw it to my e-waste bin.

denisjoshua, 2020-10-18:
Can you give me a link to this BTLE version please ? 
I can't find it on net. 
How much distance is work please ? cause I need also out of home in my greenhouse. 
Thanks again 
Denis

randybb, 2020-10-18:
You can find all supported bluetooth sensors here https://esphome.io/components/sensor/xiaomi_ble.html You can buy these Aliexpress / Bangoods / Gearbest.

denisjoshua, 2020-10-19:
This is greate ... thanks a lot 
But last question please... for example the "CGG1 Hygro thermometer, round body, e-ink display" how much distance
can transmit ? 
I ask you cause I need to put some sensor in my greenhouse and the greenhouse is at about 20 MT from my house. 

Thanks a lot again