esphome/feature-requests #432: Support for DS2413 (1-Wire Dual Channel Addressable Switch)

randybb, 2019-10-17:
**Describe the problem you have/What new integration you would like**

Besides of temperature sensors 1-wire bus supports much more interesting devices, like EEPROM's, addressable switches (GPIO), digital potentiometers (ADC),... 

**Please describe your use case for this integration and alternatives you've tried:**

Unfortunately, the dallas component is not universal,is aim to temperature sensors only. Please make it universal for future expansions.

**Additional context**

This is based on closed feature request https://github.com/esphome/feature-requests/issues/391 and since I have these on stock, I can send them to you to play with.

epenet, 2020-08-27:
Hi @randy,

Do you still need this?
I've added a PR for some extra functionnality on the onewire integration: https://github.com/home-assistant/core/pull/39321

If you give me more details, then I can look at adding these also...

epenet, 2020-08-27:
Hi @randybb,
The onewire integration is being worked on (https://github.com/home-assistant/core/pull/39321), so I guess it depends if you want to use the HomeAssistant integration or esphome.

randybb, 2020-08-27:
Hi @epenet, wow. It could be handy for 1wire testing and in some automation scenarios.

Originally I was looking for a dedicated device that can handle the logic by self. I don't want to rely on HA - heating and lights need to work independently with remote control option. Because there was no demand from other people i decided to move to a different solution. ESPHome will be connected to a MCU that will act as a I2C to GPIO bridge from ESPHome site (similar to [Arduino port extender](https://esphome.io/cookbook/arduino_port_extender.html just with STM32) and on the other site will be 1wire with multiple 8 PIO switches controlling remote switches/LEDs + relays connected via I2C GPIO bridge (it is a standard in all my setups with the same modular HW that I keep in stock). Unfortunately, due lack of time I postponed it completely.

Why we need have this supported? Maybe it doesn't sounds fancy based on my description, but just imagine to have all wall switches connected to 2 wire bus. You will need a special switches, but these are commercial available https://sedtronic.eu/en/unica-1wire/30-52-unica-1-wire-tvsl.html or you can build your own HW (for example with 8 PIO) based on your needs.

epenet, 2020-08-28:
Hi @randybb,

I didn't realise that I had ended up on the esphome repository! I thought I was still on the [home assistant](https://github.com/home-assistant) repository.
If your need is definitely for esphome integration, then I can't actually help you.

JayElDubya, 2020-11-28:
I received a handful of the DS2413 in tsoc6 packaging (tiny). I "dead-buged" one onto some headers so I could prototype and test on a breadboard. Using Adafruit's test code I verified that it is working as expected -- at least in output mode.

So now I've moved onto attempting a custom component to use these things. Admittedly my coding is limited mostly to bash scripting and 101 level C so this is going to be a learning experience no doubt. For example, constructors and classes are new to me. 

So that said, would be helpful if someone had some advice to get me going. I've read the custom sensor and custom switch pages and have started trying to understand some of the example custom components I've found. Will see how far I can get and then post some code blocks.

-J

JayElDubya, 2020-11-28:
First question, are the esp_one_wire and/or dallas libraries that esphome usable in a custom component? i.e. can I leverage the functions that scan the bus for devices?

Here's what I have so far, and no it doesn't work.

in custom_component_DS2413.h:
```
#include "esphome.h"
#include "esp_one_wire.h"

namespace esphome {
namespace dallas {

#define DS2413_FAMILY_ID 0x3A
#define DS2413_ACCESS_READ 0xF5
#define DS2413_ACCESS_WRITE 0x5A
#define DS2413_ACK_SUCCESS 0xAA
#define DS2413_ACK_ERROR 0xFF

  // static const uint8_t DALLAS_MODEL_DS2413 =

class DS2413CustomComponent : public Component,  public BinaryOutput  {

  public:
    ESPOneWire oneWire(GPIOPin *pin);
    uint8_t address[8] = {0, 0, 0, 0, 0, 0, 0, 0};
    
    void setup() override {
      ESP_LOGD("custom", "Looking for a DS2413 on the bus");
      OneWire.reset_search();
    }
    void write_state(bool state) override {}
  };
} // namespace dallas
} // namespace esphome
```

and in the relevant yaml:
```
esphome:
  <<: !include templates/devices/wemos_d1_mini_core.yaml
  includes:
    - custom_component_DS2413.h
  libraries:
    - "esp_one_wire.h"

dallas:
  - pin: D3

output:
  - platform: custom
    type: binary
    lambda: |-
      auto my_custom_binary_output = new DS2413CustomComponent();
      App.reegister_component(my_custom_binary_output);
      return {my_custom_binary_output};
    outputs:
      - id: sw_01  # note that the DS2413 has two GPIO so I also need to figure out multiple outputs here
```

-J

brano1990, 2021-01-24:
hi @JayElDubya does this your code work well as a switch?
would it work as a binary sensor?1-wire

JayElDubya, 2021-02-15:
@brano1990, my code does not work yet but my intent is to create two switches with this component.

JayElDubya, 2021-02-17:
Let me first say that I'm not coding is not my day job so I could use some help cleaning this up. However, the code is working as is for using the DS2413 as a dual switch. However, it's kludgy because I couldn't figure out how to create a custom component with multiple switches which is all kinds of inefficient. It scans the bus twice on startup, I have duplicative but otherwise equal functions, etc. I'll post what I have and if anyone wants to help me clean it up I would be much appreciative. The limitations of the current state of code as I can think of are:

- only supports a single DS2413 device on the oneWire bus
- no restore on boot, you set default boot states with bool variables near the top
- scans the bus twice on setup()
- switch output mode only, no support yet for using as an input
- on bus scan, the address is printed incorrectly

With that all out of the way, here it is working:

relevant yaml:
`
esphome:
  name: ${project_name}
  platform: ESP8266
  board: d1_mini
  includes:
    - custom_component_DS2413.h
  libraries:
    - "OneWire"

switch:
  - platform: custom
    lambda: |-
      auto sw01 = new DS2413_1();
      App.register_component(sw01);
      return {sw01};
    switches:
      - id: switch_01
        name: "Switch 1"
  - platform: custom
    lambda: |-
      auto sw02 = new DS2413_2();
      App.register_component(sw02);
      return {sw02};
    switches:
      - id: switch_02
        name: "Switch 2"
`

custom_component_DS2413.h:
`
#pragma once
#include "esphome.h"
#include "OneWire.h"

#define DS2413_ONEWIRE_PIN  0 //gpio0 = d3
bool sw1 = false;  // set default boot state 
bool sw2 = false;  // set default boot state

#define DS2413_FAMILY_ID    0x3A
#define DS2413_ACCESS_READ  0xF5
#define DS2413_ACCESS_WRITE 0x5A
#define DS2413_ACK_SUCCESS  0xAA
#define DS2413_ACK_ERROR    0xFF

#define TAG "dallas"

class DS2413_1 : public Component, public Switch  {
  public:
    OneWire oneWire;  // initiate component
    uint8_t address[8] = { 0, 0, 0, 0, 0, 0, 0, 0 };
    
    DS2413_1():oneWire(DS2413_ONEWIRE_PIN) {} // constructor

    void printBytes(uint8_t* addr, uint8_t count) {
      for (uint8_t i = 0; i < count; i++) 
      {
        ESP_LOGD(TAG, "%i:", addr[i]); // working but needs conversion to hex and a string buffer
      }
[... 146 lines cut ...]
void write_state(bool state) override
{
  sw2 = state;
  bool ok = false;

  ESP_LOGD(TAG, "Writing state of: %i", sw2);

  if (sw2 && sw1) {
    ok = write(0x03);
  } else if (sw2 && !sw1) {
    ok = write(0x02);
  } else if (!sw2 && sw1) {
    ok = write(0x01);
  } else {
    ok = write(0x00);
  }

  (ok) ? publish_state(sw2) : ESP_LOGD("custom", "Wire failed");  
  
  } // write_state
  
}; // class
`

-J