esphome/feature-requests #434: Allow different manual_ip settings per ssid when configuring multiple wifi networks

davey, 2019-10-19:
**Describe the problem you have/What new integration you would like**

Please add support for having separate, per-wifi static IP settings if using multiple wifi networks/SSIDs.

**Please describe your use case for this integration and alternatives you've tried:**

I configured multiple wifi networks to have connection redundancy, want to use manual IP settings instead of DHCP and have different IP settings / subnets in these different wifi networks. 
Currently, it seems not to be possible to have different manual_ip settings for these multiple wifi ssid settings in esphome.

This is what I have now, and it only works if connected to the main-wifi, because only there the IP settings fit:

```yaml
wifi:
  networks:
  - ssid: "main-wifi"
    password: "secret1"
  - ssid: "backup-wifi"
    password: "secret2"
  manual_ip:
    static_ip: 1.11.11.11
    gateway: 10.11.11.1
    subnet: 255.255.255.0
    dns1: 10.11.11.1
    dns2: 8.8.8.8
```

This is what I think would be desirable:

```yaml
wifi:
  networks:
  - ssid: "main-wifi"
    password: "secret1"
    manual_ip:
      static_ip: 10.11.11.11
      gateway: 10.11.11.1
      subnet: 255.255.255.0
      dns1: 10.11.11.1
      dns2: 8.8.8.8
  - ssid: "backup-wifi"
    password: "secret2"
    manual_ip:
      static_ip: 10.22.22.22
      gateway: 10.22.22.1
      subnet: 255.255.255.0
      dns1: 10.22.22.1
      dns2: 8.8.8.8
```

**Additional context**

I know that using DHCP would solve this, but in my case I need to have the static/manual IP settings for multiple reasons (e.g.: don't own the backup-wifi and no DHCP server there; need the fast connection setup; ...)

BTW: Thanks a lot for esphome - I'm LOVING it! :-)

OttoWinter, 2019-10-20:
Yeah I knew this would come up some day.

The problem here is that ESPHome has *one address* it tries to reach the ESP under for things like logs, ota etc.

If you don't have a static IP, it's `<name>.local`, if you do, it's the IP.

However, it's hard to define what should happen when there are multiple (potentially different) static IPs. Should it choose the first one? Asking the user is not really simple because that would need changes in many parts of the codebase and would not be worth the effort.

The C++ codebase already supports this, the main problem is how the frontend can handle it.

davey, 2019-10-22:
Thanks for looking into this!
What about this idea: why not just set an additional 'flag' on one of the entries that marks it as the 'main' IP (for ota, logs,...)?

davey, 2019-10-22:
Even just automatically using the first static IP found in the yaml would be fine, as long as it's well documented - this leaves control to the user and avoid a lot extra vide I guess

OttoWinter, 2019-10-22:
Yeah, that's probably true. I will revisit this after 1.14 release is complete.

mihlit, 2021-02-06:
I've just tried to use this to find out it does not work yet. So here is my use case:

I have two locations where I use home assistant. I use vpn to connect second location, so I rely on static ip and not mdns. 

I prepare esphome switches/sensors at home and then I want to deploy them in second location, preferably without need to upload firmware over cable, so I don't have to take my laptop with me. For this I use two networks with static ip address in configuration and together "use_address" option. I configure and upload it at home with "use_address" for home wifi, then I would install that device in other location. It would get the second ip address and I would just change use_address in configuration, so esphome knows where to find it. As that value is used only for master configuration, it just works and I don't have to upload new firmware.

dr1rrb, 2021-05-03:
Oh :( I was under the feeling that it was already working as the editor didn't raised any red flag. 

My use case is really to have a backup WiFi to connect to before fallbacking to the internal AP/Captive portal. Even with a UniFi HD, the WiFi from my terrace is really unstable. So I've created an AP on a RPi connected by PLC, and I want the ESP32 to connect to it ... but I don't really trust that WiFi neither. My idea was that I would always have at least one WiFi accessible, and the RPi which is running some automations that are going to use the ESP32, would try both IPs (the RPi AP and my "main" WiFi), using static IPs on both to reduce points of failures.

Personally I love the idea to be able to define a custom endpoint for the ota component (I already map all my ESP in my DNS with a custom name with dashes, which are not supported by ESPHome).

BTW ... is really LOVE ESPHome ! Thanks for the great job 👍

closed 2022-07-01 by nagyrobi