esphome/feature-requests #436: IR gateway

escoand, 2019-10-19:
**Describe the problem you have/What new integration you would like**
I try to free my [WiFi-IR-gateway](https://blakadder.github.io/templates/auvisio_S06.html) from the cloud and so installed esphome but struggle to get it working correctly.

The goal is to use my TV remote control to control my Logitech AV receiver to only have one remote control.

At the end a simple mapping in the yaml would be nice. So you could route IR-to-IR, IR-to-RF and RF-to-IR.

**Please describe your use case for this integration and alternatives you've tried:**
I tried two different options but both do not work. The receiving and the switch at is working without problems so IR is generally working.

```yaml
binary_sensor:
  # map volume_up from Samsung 0xE0E0E01F to NEC 0x10EF+0x58
A7
  - platform: remote_receiver
    id: samsung_volumeup
    samsung:
      data: 0xE0E0E01F
    on_press:
      - remote_transmitter.transmit_nec:
          address: 0x10EF
          command: 0x58A7

  # map volume_down from Samsung 0xE0E0D02F to NEC 0x10EF+0x
708F
  - platform: remote_receiver
    id: samsung_volumedown
    samsung:
      data: 0xE0E0D02F
    on_press:
      - switch.turn_on: volume_down

switch:
  - platform: template
    id: volume_down
    name: Volume Down
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0x10EF
          command: 0x708F
```

**Additional context**

OttoWinter, 2019-10-20:
> At the end a simple mapping in the yaml would be nice. So you could route IR-to-IR, IR-to-RF and RF-to-IR.

Well I think the existing tools (namely the binary_sensor + transmit action) solve this sufficiently already (the first entry in your config is already good, no need for a switch in between)

> I tried two different options but both do not work.

Please explain. Add logs, explain what device is sending the remote codes, and which should be receiving them.

escoand, 2019-10-23:
Hmm, the logs look good but the AV receiver does not recognize the message. Could it be that the repetition of the Samsung remote interfere with the Logitech signal which is too noisy of this?

```
[23:12:44][D][binary_sensor:033]: 'samsung_volumeup': Sending state ON
[23:12:44][D][remote_transmitter:066]: Sending remote code...
[23:12:44][D][binary_sensor:033]: 'samsung_volumeup': Sending state OFF
[23:12:44][D][binary_sensor:033]: 'samsung_volumeup': Sending state ON
[23:12:44][D][remote_transmitter:066]: Sending remote code...
[23:12:44][D][binary_sensor:033]: 'samsung_volumeup': Sending state OFF

[23:12:57][D][binary_sensor:033]: 'samsung_volumedown': Sending state ON
[23:12:57][D][switch:021]: 'Volume Down' Turning ON.
[23:12:57][D][remote_transmitter:066]: Sending remote code...
[23:12:57][D][binary_sensor:033]: 'samsung_volumedown': Sending state OFF
[23:12:57][D][binary_sensor:033]: 'samsung_volumedown': Sending state ON
[23:12:57][D][switch:021]: 'Volume Down' Turning ON.
[23:12:57][D][remote_transmitter:066]: Sending remote code...
[23:12:57][D][binary_sensor:033]: 'samsung_volumedown': Sending state OFF
```

escoand, 2019-10-23:
I found some anomalies in the Logitech remote codes, it seems to be not 100% NEC compatible.
```
[23:33:15][D][remote.nec:068]: Received NEC: address=0x10EF, command=0x58A7
[23:33:15][D][remote.raw:041]: Received Raw: 9078, -2217, 559
[23:33:22][D][remote.nec:068]: Received NEC: address=0x10EF, command=0x708F
[23:33:22][D][remote.raw:041]: Received Raw: 9081, -2214, 559
# and as raw
[23:39:05][D][remote.raw:028]: Received Raw: 9048, -4472, 578, -557, 587, -549, 583, -553, 581, -1645, 610, -525, 609, -508, 629, -507, 602, -533, 583, -1667, 605, -1666, 581, -1692, 583, -510, 624, -1667, 561, -1664, 606, -1668, 584, -1666, 605, -531, 609, -1663, 558, -555, 582, 
[23:39:05][D][remote.raw:041]:   -1690, 583, -1665, 582, -556, 558, -556, 582, -531, 602, -1668, 607, -529, 584, -1666, 609, -527, 584, -532, 580, -1668, 581, -1691, 580, -1671, 581
[23:39:05][D][remote.raw:041]: Received Raw: 9043, -2228, 603
[23:39:06][D][remote.raw:028]: Received Raw: 9074, -4495, 582, -531, 578, -538, 600, -557, 558, -1667, 605, -555, 557, -556, 580, -556, 580, -557, 557, -1669, 578, -1693, 585, -1667, 606, -528, 580, -1692, 559, -1668, 578, -1694, 582, -1691, 553, -559, 608, -1665, 582, -1665, 582, 
[23:39:06][D][remote.raw:041]:   -1669, 580, -531, 631, -531, 556, -537, 600, -554, 580, -1671, 579, -533, 603, -558, 532, -582, 579, -1692, 558, -1669, 579, -1672, 604, -1690, 554
[23:39:06][D][remote.raw:041]: Received Raw: 9040, -2249, 584
```

IR from ESPHome to Logitech is currently not working at all. But I'm sure the switch worked some days ago. But I was able to capture parts of my own NEC signals. So the transmit GPIO seems to work generally:
```
[00:01:12][D][binary_sensor:033]: 'samsung_volumedown': Sending state ON
[00:01:12][D][remote_transmitter:066]: Sending remote code...
[00:01:12][D][binary_sensor:033]: 'samsung_volumedown': Sending state OFF
[00:01:12][D][remote.raw:041]: Received Raw: 582, -1632, 606, -512, 609, -1656, 583, -512, 608, -513, 607, -512, 609, -512, 608, -512, 608, -513, 608, -1656, 582, -512, 609, -1656, 582, -1657, 582, -1655, 583, -1656, 582
[00:01:12][D][binary_sensor:033]: 'samsung_volumedown': Sending state ON
[00:01:12][D][remote_transmitter:066]: Sending remote code...
[00:01:12][D][binary_sensor:033]: 'samsung_volumedown': Sending state OFF
```

I tried to send the additional codes as well, but this did not work:
```yaml
switch:
  - platform: template
    id: volume_down
    name: Volume Down
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0x10EF
          command: 0x708F
      - remote_transmitter.transmit_raw:
          code: [9080, -2215, 560]
```

I'm not sure what to do next...

escoand, 2019-10-23:
To answer you second question: I have a Samsung UE40ES5700 and a Logitech Z-5500. And my goal is to control the latter with the remote of the first.

OttoWinter, 2019-10-24:
If you think it may not be 100% NEC compatible, I would recommend you to only dump the raw codes (`dump: raw`) and use those instead.

escoand, 2019-10-24:
There are 2 rows of "Received Raw:". Do they have to send in one transmit action?

OttoWinter, 2019-10-24:
Yes

closed 2020-05-04 by escoand