esphome/feature-requests #437: Enhance !include

Joshfindit, 2019-10-19:
**Describe the problem you have/What new integration you would like**

I would like to use includes as templates and rely on substitutions instead of 'standard' changes.

## Allow multiple 'trunks' in included yaml files

### Example 1
Splitting out the touch sensor in to `capacitivetouch.yaml`

```yaml
esp32_touch:
  setup_mode: false

binary_sensor:
  - platform: esp32_touch
    name: $touchname
    id: mainbutton
    pin: $touchpin
    threshold: 500

    on_click:
      then:
        - light.turn_off: mainlight
```

### Example 2
Encapsulating everything to do with wifi in to `mywifisettings.yaml`

```yaml
wifi:
  ssid: $wifiNameToJoin
  password: $wifiPasswordToJoin

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $accesspointSSID
    password: "esphomeisgreat"

captive_portal:
```

## Allow mixing `!include`d effects with in-file effects

### Example 1

```yaml
light:
    ...
    effects:
      - flicker:
          name: Flicker
          alpha: 95%
          intensity: 1.5%
      - strobe:
          name: Strobe
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 100%
              blue: 100%
              white: 100%
              duration: 80ms
            - state: False
              duration: 80ms
      !include includes/genericSingleLEDEffects.yaml
      !include includes/testing/lambdaEffect1.yaml
```

### Example 2 (this one may go a little far)
*Allows for the `!include`s to interact with the tree-based data object itself, not just as text*

`lamp4.yaml`
```yaml
substitutions:
  devicename: lamp4
  upper_devicename: Lamp 4
  ledpin: GPIO18
  upper_devicename2: Lamp 4 (2)
  ledpin2: GPIO19
  onboardledpin: GPIO22

switch:
  - platform: gpio
    pin: $onboardledpin
    name: "Onboard LED"
    id: onboard_led
    internal: true
    inverted: yes

light:
  - platform: monochromatic
    name: $upper_devicename
    output: output_component1
    default_transition_length: 500ms

output:
  - platform: ledc
    id: output_component1
    pin: $ledpin
    frequency: 50000
    bit_depth: 10

!!include includes/twoOutputs.yaml # '!!' is a notation I just made up, meaning 'overwrite any conflicting entries with the ones from the included file'.
```

`twoOutputs.yaml`
```yaml
light:
  - platform: monochromatic
    name: $upper_devicename
    output: output_component1
    default_transition_length: 500ms
  - platform: monochromatic
    name: $upper_devicename2
    output: output_component2
    default_transition_length: 500ms

output:
  - platform: ledc
    id: output_component1
    pin: $ledpin
    frequency: 50000
  - platform: ledc
    id: output_component2
    pin: $ledpin2
    frequency: 50000
```

**Please describe your use case for this integration and alternatives you've tried:**

I want to be able to create dozens of lights based on ESPHome, and 'centrally' manage the settings/properties.
If I find a better touch threshold, or a light transition that I particularly like, then it's just a matter of modifying it in one place and uploading to all devices.

**Additional context**

I know that this goes further than HomeAssistant's yaml import, but that's designed to allow splitting up a monolithic single config, whereas ESPHome is about configuring multiple devices.

OttoWinter, 2019-10-20:
The main problem here is that we must work with the YAML syntax.

If we enhance the YAML syntax by new features we will break all YAML validators (IDEs, ...) and our YAML parses. The only way we can add new features to the YAML parser is through adding new `!tag` s - and those have a very clearly defined syntax: https://yaml.org/spec/1.2/spec.html.

So Example 1 in `Allow mixing !included` will already not work because the list syntax can't combine list items and lists itself.

Example 2 also can't work because it's expecting a key-value pair at the last line - but it's only given a key (so it's invalid YAML).

Over the last few months I've thought a lot about this, and there are some solutions to this.

One is a system similar to HA's `package` system - YAML configs are independently loaded and combined in the end. Problem: The load order of all these things (YAML, substitutions, packages, ...) is hard to define.

A second option is something similar to what Ansible does: jinja2 in values that are evaluated at compile time.

There's also a ton of other ideas that I'm evaluating, and unfortunately don't have time to write down right now. But rest assured I'm working on something.

Joshfindit, 2019-10-20:
Awesome, I’m looking forward to seeing where this goes.

docmattman, 2019-12-23:
The idea of using HA `packages` sounds pretty good.  Right now, I include a lot of "boilerplate" text sensor data on every component.  Something like this:

```
# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Doorbell ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Doorbell IP
    ssid:
      name: Doorbell SSID
    bssid:
      name: Doorbell BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Doorbell Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: Doorbell WiFi Signal
    update_interval: 60s
```

However, if one of my components needs to have an additional `text_sensor`, I can't include a file with the "boilerplate" ones and add/extend ones specific to the component.  If we could define packages and still all the common stuff in as a package and get the config to merge with the component-specific config, that would be idea.  It would mean much less config-duplication across various components.

theone11, 2020-05-04:
Waiting for something like this too.

I have many devices that use the "common.yaml" file but also need specific text_sensor/binary_sensor/sensor and can't include it in the specific yaml file becuase exist in the common yaml file.

corvis, 2020-05-05:
Hey guys,

I would propose the following:

1. Implement `packages:` in the way similar to what HomeAssitant have. Definition in config should prevail over definition in package so lists and dicts should be merged in the way when definition in config file overrides definition in package.

2. Disable feature. It would be also good to have a way to disable some component definition in compile time via corresponding substitutions. So you could describe config for the most complete setup and just disable features with simple configuration defined in one single place.

Example from my practice - I have a heating controller which includes temperature sensor, relay output and bang bang controller for each of the channels (1 to 4) and also optional display.
There are a couple modifications of device (different qty of channels, display module) and it would be really great to manage it as follows:

```yaml
substitutions:
  display_module: true
  enable_zone1: true
  enable_zone2: true
  enable_zone3: false

packages:
    network_config: <<: !include common/wifi.base.yaml
    heating_base: <<: !include common/heating.base.yaml
```

The way it cold be achieved is to add boolean property to component base schema and ignore disabled components on config validation stage.

WDYT @OttoWinter ?

corvis, 2020-05-07:
Hey everyone, 

I've implemented packages feature in [my fork](https://github.com/esphome/esphome/compare/dev...corvis:fb-packages). This basically allows configuration as follows:

config.yaml
```yaml
substitutions:
  zone_name: Living room

packages:
  wifi: !include test_packages/test_packages_package_wifi.yaml
  heating_base: !include test_packages/test_packages_package1.yaml

esphome:
  name: test1

uart:
  rx_pin: GPIO12
  tx_pin: GPIO13
  baud_rate: 9600

sensor:
  - platform: mhz19
    co2:
      name: "CO2"
    temperature:
      name: "Temperature"
    update_interval: 60s
    automatic_baseline_calibration: false
```
test_packages_package1.yaml
```yaml
esphome:
  board: nodemcu-32s
  platform: ESP32
  build_path: build/test1

i2c:
  sda: 21
  scl: 22
  scan: True

sensor:
  - platform: adc
    pin: A0
    name: "$zone_name Brightness"
    attenuation: 2.5db
    unit_of_measurement: "°C"

  - platform: esp32_hall
    name: ESP32 Hall Sensor
```
test_packages_package_wifi.yaml
```yaml
wifi:
  networks:
  - ssid: 'MySSID'
    password: 'password1'
```

After config validation it transforms into:
```yaml
substitutions:
  zone_name: Living room
esphome:
  name: test1
  board: nodemcu-32s
  platform: ESP32
  build_path: build/test1
  includes: []
  arduino_version: espressif32@1.11.0
  platformio_options: {}
  libraries: []
uart:
- rx_pin: 12
  tx_pin: 13
  baud_rate: 9600
  stop_bits: 1
sensor:
- platform: mhz19
  co2:
    name: CO2
    unit_of_measurement: ppm
    icon: mdi:periodic-table-co2
    force_update: false
    accuracy_decimals: 0
  temperature:
    name: Temperature
    unit_of_measurement: °C
    icon: mdi:thermometer
    force_update: false
    accuracy_decimals: 0
  update_interval: 60s
  automatic_baseline_calibration: false
- platform: adc
  pin: 36
  name: Living room Brightness
  attenuation: 2.5db
  unit_of_measurement: °C
  update_interval: 60s
  icon: mdi:flash
  force_update: false
  accuracy_decimals: 2
- platform: esp32_hall
  name: ESP32 Hall Sensor
  update_interval: 60s
  unit_of_measurement: µT
  icon: mdi:magnet
  force_update: false
  accuracy_decimals: 1
wifi:
  networks:
  - ssid: MySSID
    password: password1
    priority: 0.0
  reboot_timeout: 15min
  domain: .local
  power_save_mode: LIGHT
  fast_connect: false
  use_address: test1.local
i2c:
- sda: 21
  scl: 22
  scan: true
  frequency: 50000.0
```

@OttoWinter If you find it reasonable I'll be happy to update docs and submit pull request.

theone11, 2020-05-07:
@corvis In your implementation can substitutions also exist in packages?

corvis, 2020-05-07:
Yes, @theone11, this basically the key point which allows to split configuration efficiently

Joshfindit, 2020-05-07:
@corvis Am I wrong in seeing your suggetsion as meeting every one of the examples on my wishlist, including the one that “may go a little far”?

corvis, 2020-05-07:
@theone11 I think you are correct and all use cases you mentioned in the original request are covered.
Aside note on your example 2 it seems like this is achievable even now with `<<: ` syntax check the example below:

main config
```yaml
light:
  - platform: cwww
    name: "TEST Light CWWW"
    cold_white: pwm2
    warm_white: pwm1
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    effects:
      - flicker:
          name: Flicker
          alpha: 95%
          intensity: 1.5%
      - <<: !include common/simple-light-effect.yaml
``` 

simple-light-effect.yaml
```yaml
flicker:
  name: Flicker 2
  alpha: 59%
  intensity: 1.3%
```

theone11, 2020-06-29:
When is this planned to be released to the dev branch? Can't wait to use it :)

corvis, 2020-06-29:
Good question. My pull request is still under review and I never get any feedback. This makes me think it will not happen any time soon. Maybe we could ask @glmnet to take a look.

closed 2020-07-13 by glmnet