esphome/feature-requests #438: MHZ19B commands to set range not implemented

borpin, 2019-10-20:
Refers to Issue esphome/issues#709 
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io / Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS D1 Mini Rev 3.1.0
**Affected component:**

https://esphome.io/components/sensor/mhz19.html

**Description of problem:**
The commands to set the device range, as detailed in the most recent [datasheet ](https://www.winsen-sensor.com/d/files/MH-Z19B.pdf) for this device, have not been implemented.  The lack of an ability to set this configuration, **may** be affecting the values returned by the device.

https://github.com/esphome/esphome/blob/4310c14497be80eb47c33733910242b16aa7aa4a/esphome/components/mhz19/mhz19.cpp#L10-L15

[image]

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

evils, 2022-12-14:
as the 2000-5000ppm range is where mitigation actions are indicated, it'd be great if i could get this range out of my sensor...

closed 2023-02-10 by borpin

evils, 2023-02-10:
@borpin what completed this?