esphome/feature-requests #439: RC6 support for IR receive / transmit in Remote Transmitter

pinquin87, 2019-10-20:
**Describe the problem you have/What new integration you would like**

I would like to have the RC6 IR protocol added to ESPHome for my Philips TV.

**Please describe your use case for this integration and alternatives you've tried:**

I use a ESP8266 via mqtt with Home Assistant to turn on and off my tv and audio system via IR. The ESP8266 is currently running on Tasmota. I'm in the progress of moving my 15 or so ESP devices to ESPHome because I like the integration in Home Assistant and the yaml config. The move is going great so far, but I've just run into the issue of the missing RC6 IR protocol in the Remote Transmitter component.

**Additional context**

davidsat, 2020-01-16:
Hi,
When I run the in dump:all mode, it returns RC6 codes from my remote. But when I try to specifiy that as a protocol to dump, esphome won't let me compile it. Can this be fixed? I'm not sure what the state of RC6 support is. It seems to be partially supported?

Eknem, 2020-04-05:
Hi,
I Also need rc6 support If I try to send de raw data the rc6 receiver does not seems to recognize te signal. I tried to set the carrier_frequentie to 36Hz.

rarroyo6, 2021-03-13:
I also vote for this. I have 3 media center computers set up, using the MCE remote, and this integration would be fantastic.

andrewjswan, 2021-04-14:
I also vote for this

hExPY, 2021-06-03:
Vote 👍

vanrace, 2021-11-24:
vote!

koentelg, 2021-11-25:
vote!

john-arvid, 2022-04-16:
vote! For now I use tasmota

rradar, 2022-07-17:
available in esphome 2022.6 :tada: 

https://github.com/esphome/esphome/pull/3514

closed 2022-07-17 by rradar