esphome/feature-requests #442: Support for Sensirion SPS30 Particulate Matter Sensors for air quality measurement

valordk, 2019-10-20:
**Describe the problem you have/What new integration you would like**

Support for Sensirion SPS30 Particulate Matter (aka. air quality) sensor

**Please describe your use case for this integration and alternatives you've tried:**

I'm working on the implementation of this myself, so this is to ensure proper communication and see if others are willing to test it.

**Additional context**

matthijsberg, 2019-11-14:
I have a couple, if you need someone to test it i'll be happy to.

valordk, 2019-11-14:
Great. I'll create a pull-request this weekend.

valordk, 2019-11-18:
> 
> 
> I have a couple, if you need someone to test it i'll be happy to.

I'm not done with the documentation part for the pull request yet... but the i2c implementation can allready be tested if you'd like to build from my feature branch:
`https://github.com/valordk/esphome/tree/dev-sps30-support`

I assume you already have the required JST ZHR-5 connector.
The YAML code snippet should countain the i2c configration and the sps30 platform configuration as follows:
```
i2c:
  sda: 21 <---change this to match your setup
  scl: 22 <---change this to match your setup
  scan: False
  frequency: 50kHz  <---not critical
sensor:
  - platform: sps30
    pm_1_0:
      name: "senntemp SPS30 0x69 PM <1µm Weight concentration"
      id: "senntemp_SPS30_0x69_PM_1_0"
      accuracy_decimals: 2
    pm_2_5:
      name: "senntemp SPS30 0x69 PM <2.5µm Weight concentration"
      id: "senntemp_SPS30_0x69_PM_2_5"
      accuracy_decimals: 2
    pm_4_0:
      name: "senntemp SPS30 0x69 PM <4µm Weight concentration"
      id: "senntemp_SPS30_0x69_PM_4_0"
      accuracy_decimals: 2
    pm_10_0:
      name: "senntemp SPS30 0x69 PM <10µm Weight concentration"
      id: "senntemp_SPS30_0x69_PM_10_0"
      accuracy_decimals: 2  
    pmc_0_5:
      name: "senntemp SPS30 0x69 PM <0.5µm Number concentration"
      id: "senntemp_SPS30_0x69_PMC_0_5"
      accuracy_decimals: 2
    pmc_1_0:
      name: "senntemp SPS30 0x69 PM <1µm Number concentration"
      id: "senntemp_SPS30_0x69_PMC_1_0"
      accuracy_decimals: 2
    pmc_2_5:
      name: "senntemp SPS30 0x69 PM <2.5µm Number concentration"
      id: "senntemp_SPS30_0x69_PMC_2_5"
      accuracy_decimals: 2
    pmc_4_0:
      name: "senntemp SPS30 0x69 PM <4µm Number concentration"
      id: "senntemp_SPS30_0x69_PMC_4_0"
      accuracy_decimals: 2
    pmc_10_0:
      name: "senntemp SPS30 0x69 PM <10µm Number concentration"
      id: "senntemp_SPS30_0x69_PMC_10_0"
      accuracy_decimals: 2  
    address: 0x69
    update_interval: 10s
```
The sensor data should look like this:
[image]

regards, 
Val

joegross, 2019-11-19:
Works nicely with adafruit esp8266 huzzah

With adafruit esp32 feather I get

```
[20:59:05][C][sps30:024]: Setting up sps30...
[20:59:05][E][sps30:226]: CRC8 Checksum invalid! 0x40 != 0x81
[20:59:05][E][component:092]: Component was marked as failed.
```

valordk, 2019-11-19:
> Works nicely with adafruit esp8266 huzzah
> 
> With adafruit esp32 feather I get
> 
> ```
> [20:59:05][C][sps30:024]: Setting up sps30...
> [20:59:05][E][sps30:226]: CRC8 Checksum invalid! 0x40 != 0x81
> [20:59:05][E][component:092]: Component was marked as failed.
> ```

Can you set the i2s scanning to True and see if your module gets detected? Cold start your esp32 and the module at the same time.

matthijsberg, 2019-11-19:
Thanks! Super work. Noob here, any idea how to get those adjusted files into a docker container?  Or do I need to change the setup to a different installation?

valordk, 2019-11-19:
> 
> 
> Thanks! Super work. Noob here, any idea how to get those adjusted files into a docker container? Or do I need to change the setup to a different installation?

Thanks. You would only need to update your esphome docker if you want to be able to flash your devices OTA. I flash them locally so building from command prompt using serial cable works. Otherwise you will need to update your docker image by building my branch within that image.

matthijsberg, 2019-11-19:
Thnx. Hacked this files into the container and added the config to my setup. No luck yet. It's a Lolin D1 mini pro, connected to the SPS30 and uploaded the software. When it boots it shows: 

```[21:51:16][I][i2c:033]: Scanning i2c bus for active devices...
[21:51:16][I][i2c:040]: Found i2c device at address 0x69```

but after boot it gives met error 

```[21:54:39][W][i2c:073]: Received NACK on transmit of data for address 0x69```

is this due to the lolin board, or can this be faulty wiring? thnx!

valordk, 2019-11-19:
> Thnx. Hacked this files into the container and added the config to my setup. No luck yet. It's a Lolin D1 mini pro, connected to the SPS30 and uploaded the software. When it boots it shows: [...]

The wiring for i2c seems correct since your board sees the SPS30 @ 0x69. It seems that you have correctly shorted the 5th (SEL) pin to the 4th (GND) pin enabling the i2c mode instead of UART. All seems well here.

On longer wires I was also getting the same symptom, especially if I omit the i2c resistors or if their values are way too high (making their effect negligible). What resistor values are you using?

matthijsberg, 2019-11-19:
you're fast, thnx for that. I followed this setup (picture in the end); https://github.com/paulvha/sps30 and without the 10k too when is was not using esphome. That worked before with this lib. 

I looked into the log files a bit deeper and indeed it looks ok, I even get a serial number;

`[22:50:07][C][sps30:072]: sps30:
[22:50:07][C][sps30:073]:   Address: 0x69
[22:50:07][C][sps30:090]:   Update Interval: 10.0s
[22:50:07][C][sps30:091]:   Serial Number: 'F1CD24A1886DF018'`

matthijsberg, 2019-11-19:
Oh, btw I uploaded only the files that I thought are changed; the folder components/sps30 and the file "const.py"

valordk, 2019-11-19:
> you're fast, thnx for that. I followed this setup (picture in the end); https://github.com/paulvha/sps30 and without the 10k too when is was not using esphome. That worked before with this lib.
>
> I [...]

Glad to hear it worked.
You might have had a lower i2c frequency if it worked without the resistors. You can definitely experiment by lowering the frequency down from 50kHz to see if the i2c with this particular sensor works without the resistors.

Yes, it's correct that it's the sps30 folder and the const.py which are changed.

OttoWinter, 2019-11-20:
@valordk @matthijsberg Nice work! btw, you don't need to hack the files into the install, you can also use the `custom_components` folder (see contributing guide). Then you just need to copy files in the config dir (but constants need to be defined in the component file, not const.py)

matthijsberg, 2019-11-20:
I changed the module from a Lolin D1 mini Pro to a Lolin node mcu v3 and now it works. :-)

For a second. 

after a second boot i'm back to the `[W][i2c:073]: Received NACK on transmit of data for address 0x69` messages. tried with and without the pull up resistors, no luck either. Tried a second SPS30, same thing. Sorry

valordk, 2019-11-20:
> I changed the module from a Lolin D1 mini Pro to a Lolin node mcu v3 and now it works. :-)
>
> For a second.
>
> after a second boot i'm back to the `[W][i2c:073]: Received NACK on transmit of data for [...]

Did not you write earlier that it worked when you got the serial number "F1CD24A1886DF018" did it not start to show the sensor readings? Can you try enabling the debugging
````
logger:
  level: DEBUG
````
I have four of these sensors and all of them seem to work fine.

matthijsberg, 2019-11-21:
Sorry, I was unclear I see now. The setup works when I flash it with an example of the above mentioned github repo. When I flash the same wired setup via esphome it does not work anymore. I do see the serial number of the sensor, but not any actual data anymore. From there on I get the NACK errors. 

I tried this again yesterday with the MCU board, same results. Works on direct flash, not with ESPhome. 

I'll enable debug later on, maybe tonight, but if I don't make that it'll be later next week due to travel.

matthijsberg, 2019-11-21:
Ok, hope that the belows helps, but I can't make heads or tails of it. 

When I flash the board outside of esphome (arduino software) it works

When I flash the board via ESPhome it works:

`[16:04:17][C][i2c:028]: I2C Bus:
[16:04:17][C][i2c:029]:   SDA Pin: GPIO4
[16:04:17][C][i2c:030]:   SCL Pin: GPIO5
[16:04:17][C][i2c:031]:   Frequency: 50000 Hz
[16:04:17][I][i2c:033]: Scanning i2c bus for active devices...
[16:04:18][I][i2c:040]: Found i2c device at address 0x69
[16:04:18][C][logger:175]: Logger:
[16:04:18][C][logger:176]:   Level: DEBUG
[16:04:18][C][logger:177]:   Log Baud Rate: 115200
[16:04:18][C][logger:178]:   Hardware UART: UART0
[16:04:18][C][sps30:072]: sps30:
[16:04:18][C][sps30:073]:   Address: 0x69
[16:04:18][C][sps30:090]:   Update Interval: 10.0s
[16:04:18][C][sps30:091]:   Serial Number: '6B19E22CD975E11A'
[16:04:18][C][sps30:092]:   PM1.0 'senntemp SPS30 0x69 PM <1µm Weight concentration'
[16:04:18][C][sps30:092]:     Unit of Measurement: 'µg/m³'
[16:04:18][C][sps30:092]:     Accuracy Decimals: 2
[16:04:18][C][sps30:092]:     Icon: 'mdi:chemical-weapon'
[16:04:18][C][sps30:093]:   PM2.5 'senntemp SPS30 0x69 PM <2.5µm Weight concentration'
[16:04:18][C][sps30:093]:     Unit of Measurement: 'µg/m³'
[16:04:18][C][sps30:093]:     Accuracy Decimals: 2
[16:04:18][C][sps30:093]:     Icon: 'mdi:chemical-weapon'
[16:04:18][C][sps30:094]:   PM4 'senntemp SPS30 0x69 PM <4µm Weight concentration'
[16:04:18][C][sps30:094]:     Unit of Measurement: 'µg/m³'
[16:04:18][C][sps30:094]:     Accuracy Decimals: 2
[16:04:18][C][sps30:094]:     Icon: 'mdi:chemical-weapon'
[16:04:18][C][sps30:095]:   PM10 'senntemp SPS30 0x69 PM <10µm Weight concentration'
[16:04:18][C][sps30:095]:     Unit of Measurement: 'µg/m³'
[16:04:18][C][sps30:095]:     Accuracy Decimals: 2
[16:04:18][C][sps30:095]:     Icon: 'mdi:chemical-weapon'
[16:04:18][C][captive_portal:169]: Captive Portal:
[16:04:18][C][ota:029]: Over-The-Air Updates:
[16:04:18][C][ota:030]:   Address: home_multisensor_5.local:8266
[16:04:18][C][ota:032]:   Using Password.
[16:04:18][C][api:095]: API Server:
[16:04:18][C][api:096]:   Address: home_multisensor_5.local:6053
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <1µm Weight concentration': Sending state 6.67441 µg/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <2.5µm Weight concentration': Sending state 7.62254 µg/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <4µm Weight concentration': Sending state 7.64088 µg/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <10µm Weight concentration': Sending state 7.80903 µg/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <0.5µm Number concentration': Sending state 43.16322 #/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <1µm Number concentration': Sending state 50.53800 #/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <2.5µm Number concentration': Sending state 50.83760 #/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <4µm Number concentration': Sending state 50.84247 #/m³ with 2 decimals of accuracy
[16:04:18][D][sensor:092]: 'senntemp SPS30 0x69 PM <10µm Number concentration': Sending state 50.84530 #/m³ with 2 decimals of accuracy`

When i flash it AGAIN with exactly the same code via the same way it breaks:

Log info:

`[16:30:35][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[16:30:35][I][app:028]: Running through setup()...
[16:30:35][C][sps30:024]: Setting up sps30...
[16:30:35][W][i2c:073]: Received NACK on transmit of data for address 0x69
[16:30:35][E][component:092]: Component was marked as failed.`

and:

`[16:30:43][C][i2c:028]: I2C Bus:
[16:30:43][C][i2c:029]:   SDA Pin: GPIO4
[16:30:43][C][i2c:030]:   SCL Pin: GPIO5
[16:30:43][C][i2c:031]:   Frequency: 50000 Hz
[16:30:43][I][i2c:033]: Scanning i2c bus for active devices...
[16:30:43][I][i2c:040]: Found i2c device at address 0x69
[16:30:43][C][logger:175]: Logger:
[16:30:43][C][logger:176]:   Level: DEBUG
[16:30:43][C][logger:177]:   Log Baud Rate: 115200
[16:30:43][C][logger:178]:   Hardware UART: UART0
[16:30:43][C][sps30:072]: sps30:
[16:30:43][C][sps30:073]:   Address: 0x69
[16:30:43][W][sps30:086]: Unknown setup error!`

debug information:

`16:30:43][D][debug:023]: ESPHome version 1.15.0-dev
[16:30:43][D][debug:025]: Free Heap Size: 34592 bytes
[16:30:43][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[16:30:43][D][debug:190]: Chip ID: 0x00E41B7A
[16:30:43][D][debug:191]: SDK Version: 2.2.1(cfd48f3)
[16:30:43][D][debug:192]: Core Version: 2_5_2
[16:30:43][D][debug:193]: Boot Version=6 Mode=1
[16:30:43][D][debug:194]: CPU Frequency: 80
[16:30:43][D][debug:195]: Flash Chip ID=0x0016405E
[16:30:43][D][debug:196]: Reset Reason: External System
[16:30:43][D][debug:197]: Reset Info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000`

valordk, 2019-11-21:
I see that. Something seems wrong with the way it's build within Esphome. Could it be that the Esphome in docker has shown errors during the build which you did not notice? Alternatively, you can try as OttoWinter suggested via custom_components but I've never tried that.

valordk, 2019-11-21:
Now I wonder, if the SPS30 is sitting in a weird state when you flash it for the second time. It reminds me of another issue which I got fixed sometime ago, as the sensors could be in powersave mode where they don't respect the commands except the soft-reset, sometimes.
I will definitely recheck that.

matthijsberg, 2019-11-21:
I'll see to that ESP customer part, i'll let you know.

matthijsberg, 2019-11-21:
> `custom_components`

Hi @OttoWinter I tried this, but I run into the following; 

`INFO Reading configuration /config/esphome/home_multisensor_5.yaml...
WARNING Unable to import custom component sps30.sensor:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 140, in _lookup_module
    module = importlib.import_module('custom_components.{}'.format(domain))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/sps30/sensor.py", line 4, in <module>
    from esphome.const import CONF_ID, CONF_PM_1_0, CONF_PM_2_5, CONF_PM_4_0, CONF_PM_10_0, \
ImportError: cannot import name 'CONF_PM_4_0'
Failed config`

valordk, 2019-11-21:
I think you will need to declare those constants directly, since in this approach your const.py does not have the new constants which I introduced.

valordk, 2019-11-21:
By the way, can you try using three reverse quote characters to start and end the block? i think that way the markup will respect the carriage return at the end of the debug information you are pasting :)

valordk, 2019-11-21:
Looking at the formated output I can see that, the error is around fimrware version identification, that could support the theory that the sensor might need to be soft-reset, before initializing. I'll try to add some robustness, give me ½ hour

valordk, 2019-11-21:
Allright. I just pushed a commit which I'd like you to test out. Additionally, you can now unplug the sensor after successful initialization and reconnect it, or replace it and the device should continue reading the measurements without require a restart of the ESP32.

joegross, 2019-11-23:
With the latest update here's the output. Nothing appears after this

```
[14:02:37][C][sps30:024]: Setting up sps30...
[14:02:37][D][sps30:041]:   Firmware version v1.00
[14:02:37][E][sps30:238]: CRC8 Checksum invalid! 0xFF != 0x21
[14:02:37][E][component:092]: Component was marked as failed.
[14:02:40][I][i2c:033]: Scanning i2c bus for active devices...
[14:02:40][I][i2c:040]: Found i2c device at address 0x69
[14:02:40][C][sps30:066]: sps30:
[14:02:40][C][sps30:067]:   Address: 0x69
[14:02:40][W][sps30:080]: Unable to read sensor serial number
[14:02:40][C][sps30:093]:   Update Interval: 10.0s
[14:02:40][C][sps30:094]:   Serial Number: ''
[14:02:40][C][sps30:095]:   PM1.0 'aqm1 SPS30 0x69 PM <1µm Weight concentration'
[14:02:40][C][sps30:095]:     Unit of Measurement: 'µg/m³'
[14:02:40][C][sps30:095]:     Accuracy Decimals: 2
[14:02:40][C][sps30:095]:     Icon: 'mdi:chemical-weapon'
[14:02:40][C][sps30:096]:   PM2.5 'aqm1 SPS30 0x69 PM <2.5µm Weight concentration'
[14:02:40][C][sps30:096]:     Unit of Measurement: 'µg/m³'
[14:02:40][C][sps30:096]:     Accuracy Decimals: 2
[14:02:40][C][sps30:096]:     Icon: 'mdi:chemical-weapon'
[14:02:40][C][sps30:097]:   PM4 'aqm1 SPS30 0x69 PM <4µm Weight concentration'
[14:02:40][C][sps30:097]:     Unit of Measurement: 'µg/m³'
[14:02:40][C][sps30:097]:     Accuracy Decimals: 2
[14:02:40][C][sps30:097]:     Icon: 'mdi:chemical-weapon'
[14:02:40][C][sps30:098]:   PM10 'aqm1 SPS30 0x69 PM <10µm Weight concentration'
[14:02:40][C][sps30:098]:     Unit of Measurement: 'µg/m³'
[14:02:40][C][sps30:098]:     Accuracy Decimals: 2
[14:02:40][C][sps30:098]:     Icon: 'mdi:chemical-weapon'
[14:04:37][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

While leaving the esp32 plugged in, I detached one sensor and attached a different one. No output, but resetting the esp32 returned this different output with a serial number and errors every 10s

```
[14:06:59][I][app:028]: Running through setup()...
[14:06:59][C][sps30:024]: Setting up sps30...
[14:06:59][D][sps30:041]:   Firmware version v1.00
[14:06:59][D][sps30:060]:   Serial Number: 'F1B526305BE07341'
[14:07:02][I][i2c:033]: Scanning i2c bus for active devices...
[14:07:02][I][i2c:040]: Found i2c device at address 0x69
[14:07:02][C][sps30:066]: sps30:
[14:07:02][C][sps30:067]:   Address: 0x69
[14:07:02][C][sps30:093]:   Update Interval: 10.0s
[14:07:02][C][sps30:094]:   Serial Number: 'F1B526305BE07341'
[14:07:02][C][sps30:095]:   PM1.0 'aqm1 SPS30 0x69 PM <1µm Weight concentration'
[14:07:02][C][sps30:095]:     Unit of Measurement: 'µg/m³'
[14:07:02][C][sps30:095]:     Accuracy Decimals: 2
[14:07:02][C][sps30:095]:     Icon: 'mdi:chemical-weapon'
[14:07:02][C][sps30:096]:   PM2.5 'aqm1 SPS30 0x69 PM <2.5µm Weight concentration'
[14:07:02][C][sps30:096]:     Unit of Measurement: 'µg/m³'
[14:07:02][C][sps30:096]:     Accuracy Decimals: 2
[14:07:02][C][sps30:096]:     Icon: 'mdi:chemical-weapon'
[14:07:02][C][sps30:097]:   PM4 'aqm1 SPS30 0x69 PM <4µm Weight concentration'
[14:07:02][C][sps30:097]:     Unit of Measurement: 'µg/m³'
[14:07:02][C][sps30:097]:     Accuracy Decimals: 2
[14:07:02][C][sps30:097]:     Icon: 'mdi:chemical-weapon'
[14:07:02][C][sps30:098]:   PM10 'aqm1 SPS30 0x69 PM <10µm Weight concentration'
[14:07:02][C][sps30:098]:     Unit of Measurement: 'µg/m³'
[14:07:02][C][sps30:098]:     Accuracy Decimals: 2
[14:07:02][C][sps30:098]:     Icon: 'mdi:chemical-weapon'
[14:07:07][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:17][D][sps30:104]: Retrying to reconnect the sensor.
[14:07:17][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:17][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:17][E][sps30:221]: Error initiating measurements
[14:07:17][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:27][D][sps30:104]: Retrying to reconnect the sensor.
[14:07:27][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:27][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:27][E][sps30:221]: Error initiating measurements
[14:07:27][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:37][D][sps30:104]: Retrying to reconnect the sensor.
[14:07:37][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:37][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:37][E][sps30:221]: Error initiating measurements
[14:07:37][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:47][D][sps30:104]: Retrying to reconnect the sensor.
[14:07:47][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:47][W][i2c:070]: Received NACK on transmit of address 0x69
[14:07:47][E][sps30:221]: Error initiating measurements
[14:07:47][W][i2c:070]: Received NACK on transmit of address 0x69
```

valordk, 2019-11-24:
This looked like a case I had when added Soft-Reset to the SPS30 upon startup, and did not wait long enough before initialization. Since then there has been introducet a 50ms delay after the reset. This was not document so I had to experiment a bit.
I have just increased this delay to 300ms. Can you check if the latest commit addresses this?

valordk, 2019-11-24:
Regarding the auto-reconnection of the sensor, this would only work if the sensor was connected successfully and if the component was NOT marked as failed during the booting process.
Lowering the i2c frequency might be a useful step in troubleshooting. This is just to rule out the problems caused on the i2c bus. 10kHz is safe bet.

As an observation: on lower frequencies (under 50kHz), my SPS30 sensors work even without the pullup resistors. However auto-reconnection becomes very unreliable without the required pullup resistors.

valordk, 2019-11-24:
I've run some more tests with the extreme frequencies with two groups of two SPS30 sensors.

1. First group, both sensors with firmware v1.00 (sn: 9D68C89BF587C690 & sn: 9B1E8389ABC1F5C8) are running fine at 200kHz.
2. Second group, both sensors with firmware v2.01 (sn: F4C69FA902683887 & sn: 81D8F895E62B0A4D) are running fine only upto 100kHz.

In both cases the i2c bus had 10k pullup resistors.
I will try to get a clarification on this from the manufacturer.

joegross, 2019-11-25:
With the latest changes my esp32 with v1.00 sensor firmware are reliably working!

valordk, 2019-11-25:
> 
> 
> With the latest changes my esp32 with v1.00 sensor firmware are reliably working!

Great. Thanks for verifying it. I'll finalize the documentation and submit the pull request sometime this week.

valordk, 2019-11-30:
I've submitted the PR for the code and the documenation. Closing this issue.

closed 2019-11-30 by valordk