esphome/feature-requests #443: SM2135 support for smart led

mbo18, 2019-10-21:
**Describe the problem you have/What new integration you would like**
I bought some LSC GU10 Smart LED from Action and would like to use ithem with ESPHome. I've successfully used Tuya-convert but it seems that support for SM2135 is missing from ESPHome

**Please describe your use case for this integration and alternatives you've tried:**
It seems that some smart bulb are using this chip. Would be nice to use ESPHome on it.

**Additional context**
See this issue at Tasmota: https://github.com/arendst/Sonoff-Tasmota/issues/6495
A lot of information can be found there

pascalbuijs, 2019-11-04:
I've bought them to and have figured it out to control it via mqtt in Homeassistant, ESPHome would be better.

zarya, 2019-11-24:
I started on the initial support for the chip i made a working version here https://github.com/NURDspace/esphome/tree/sm2135/esphome/components/sm2135

tnx to https://github.com/arendst/Tasmota/blob/master/sonoff/xlgt_04_sm2135.ino for making the initial implementations.

mbo18, 2019-11-27:
Thank you @zarya !
Not enougth time to test currently but will try next week. Did someone tried it already?
Do you plan to open a PR?

zarya, 2019-11-28:
If time permits to write all docs, mine has been working 100% if just RGB but switching between the whites and rgb is still a bit of a thing and does not really work that well yet.

DirkTas67, 2019-12-21:
> If time permits to write all docs, mine has been working 100% if just RGB but switching between the whites and rgb is still a bit of a thing and does not really work that well yet.

Can you explain how to use this, I bought the LCS GU 10 light and want to put esphome on it (right now i have 'tuya-ed' it and put tasmota on it. I am using hassio and have esphome dev version installed in it.

Regards
Dirk

wdawn, 2019-12-23:
> > If time permits to write all docs, mine has been working 100% if just RGB but switching between the whites and rgb is still a bit of a thing and does not really work that well yet.
> 
> Can you explain how to use this, I bought the LCS GU 10 light and want to put esphome on it (right now i have 'untuya-ed' it and put tasmota on it. I am using hassio and have esphome dev version installed in it.
> 
> Regards
> Dirk

I would also like to know this.
Al my devices have ESPhome as firmware accept the LCS GU10

zarya, 2020-01-01:
Ofcourse they accept it fine its a ESP but the chip that controls the LED's in the bulps was/is unsupported. I wrote a driver for it but have not yet been able to finish it into a pull request into esphome

wojtek14a, 2020-01-01:
Thank you for this driver, I'm really looking forward to see a pull request.
I was just about to go with Tasmota on my GU10s but now I'll wait and stick with just esphome!
Regards
Wojtek

filmgarage, 2020-01-04:
Is it possible to add this component 'manually' to a working ESPHome config?

edvandreas, 2020-01-04:
> Is it possible to add this component 'manually' to a working ESPHome config?

Yes, create a folder custom_components in the esphome folder where all the yaml files are located. then create a folder sm2135 under custom_components and save the 4 files from the github link there. 

Should compile fine now.

I'm using the components on a couple of bulbs at the moment but it's not working well. Switching between the white levels and color is a bit buggy. The default color temperature when you turn on the light is not optimal.

But hey, I'm happy that it's working and pretty sure they will optimize the component in the future.

filmgarage, 2020-01-07:
@edvandreas Thanks! That worked!

I decided to turn it into two lights to make it more usable for my purpose:

```
  - platform: rgb
    name: "RGB light"
    restore_mode: ALWAYS_OFF
    red: output_red 
    green: output_green 
    blue: output_blue
    default_transition_length: 1s
  - platform: monochromatic
    name: "Warm White Light"
    restore_mode: ALWAYS_ON
    output: output_warmwhite
```

I did not get the "Cold White" output working, so until the component is working I will stick with a very(!) warm white light and an RGB light...

edvandreas, 2020-01-14:
@zarya Thank you for your work. Do you have some time to look for the cold white problem?

thibmaek, 2020-02-03:
This looks very nice! Really looking forward to this.

twkrol, 2020-02-06:
Hi
Thank you guys, especially @zarya 
As said previously, looking forward for complete implementation of cold/white feature.

MelvinGr, 2020-02-20:
Thank you for your work, i've got the component running on my LSC gu10 bulbs. 
I'm looking forward toward the fixes as well.

Thetmar, 2020-03-01:
Hello Melvin,

You don't get a reply in de yaml file that the component isnt found? I make a dir sm2135 in de custom components. And put there 4 files and do a reboot of HA. After it i go to esphome but i still getting the same issue. How is yours working? 

[image]

MelvinGr, 2020-03-01:
Did you put the cusotm_components folder inside the esphome folder?
Like this: ..../esphome/custom_components/sm2135

Thetmar, 2020-03-02:
Hello Melvin,

No i put is in de local folder where the other yaml files are. There is an folder with custom components. So i must replace the directory to an custom components in de esphome folder.
I’m gonne try this. And than restart HA.
I will let you know if it worked.

Kind regards, Thetmar

Thetmar, 2020-03-03:
Hi Melvin,

It worked.... is put the files in the wrong folder. Do you get the color pattern of the light or only cold  white? I used the used the code from zarya but i get only one switch. And then the gu10 bulb comes up with cold white light.

MelvinGr, 2020-03-03:
> Hi Melvin,
> 
> It worked....Thanks! is put the files in the wrong folder. Do you get the color pattern of the light or only cold white? I used the used the code from zarya but i get only one switch. And then the gu10 bulb comes up with cold white light.

No problem, yeah it is a bit buggy, but here is the code i'm using (from @filmgarage), which works fine enough for now.

```
sm2135:
  data_pin: GPIO12
  clock_pin: GPIO14

output:
  - platform: sm2135
    id: output_red
    channel: 0
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_blue
    channel: 2
  - platform: sm2135
    id: output_white
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: monochromatic
    name: "${devicename}_ww"
    output: output_warmwhite
    restore_mode: ALWAYS_ON
    
  - platform: rgb
    name: "${devicename}_rgb"
    red: output_red 
    green: output_green 
    blue: output_blue
```

T0ytoy, 2020-03-04:
Hi Melvin, thanks for sharing this config. Just to be clear, for now you only get RGB + warmwhite, so your bulb emits a really "orange-ish" color, right ? I want to use two E14 rgb as my bathroom mirror light, but I need the "cold" white mostly.

How do you use yours?

MelvinGr, 2020-03-04:
@T0ytoy Yes it is how you describe for now, altough i have made some progress today, will upload a patch soon!

thibmaek, 2020-03-04:
Nice! I'm heading out to get a GU10 bulb today, just to see how it works now. If this lands as a stable solution, I'm looking forward to replacing all my GU10 spots!

MelvinGr, 2020-03-04:
The channels are fixed now, so white and warm white are working. 
However, the brightness slider changes the blue channel for some reason (is this normal with esphome?) 
The white value slider changes the brightness. Color temperature slider works as expected.

Maybe someone with more experience with rgbww lights + esphome can comment in on this?

```
  - platform: rgbww
    name: "${devicename}_rgbww"
    red: output_red 
    green: output_green 
    blue: output_blue 
    cold_white: output_white 
    warm_white: output_warmwhite 
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
```

hiddedb, 2020-03-06:
@MelvinGr 

Did you get it fixed by any chance? I'm running into the same issue right now

MelvinGr, 2020-03-11:
@hiddedb which issue?

hiddedb, 2020-03-11:
@MelvinGr the problem with the slider that you commented on above?

"The brightness slider changes the chip to the colour channels
Edit: probably has to do with the fact that the chip can only do rgb OR white?"

RoelVideler, 2020-03-23:
I've just given it my best shot to put in some pull requests, but I'm new at github so I might have messed it up. For all you lovers out there, here's the sm2135.cpp that works for me:

```
#include "sm2135.h"
#include "esphome/core/log.h"

// Tnx to the work of https://github.com/arendst (Tasmota) for making the initial version of the driver

// Modified with address taken from this example: https://html.alldatasheet.com/html-pdf/1133367/LINKAGE/SM2135E/342/1/SM2135E.html

namespace esphome {
namespace sm2135 {

static const char *TAG = "sm2135";

static const uint8_t SM2135_ADDR_MC   =  0xC0;  // Max current register
static const uint8_t SM2135_ADDR_CH   =  0xC1;  // RGB or CW channel select register
static const uint8_t SM2135_ADDR_R    =  0xC2;  // Red color
static const uint8_t SM2135_ADDR_G    =  0xC3;  // Green color
static const uint8_t SM2135_ADDR_B    =  0xC4;  // Blue color
static const uint8_t SM2135_ADDR_C    =  0xC5;  // Cold
static const uint8_t SM2135_ADDR_W    =  0xC6;  // Warm

static const uint8_t SM2135_RGB       =  0x00;  // RGB channel
static const uint8_t SM2135_CW        =  0xC5; // 0x80;  // CW channel (Chip default)

static const uint8_t SM2135_10MA      =  0x00;
static const uint8_t SM2135_15MA      =  0x01;
static const uint8_t SM2135_20MA      =  0x02;  // RGB max current (Chip default)
static const uint8_t SM2135_25MA      =  0x03;
static const uint8_t SM2135_30MA      =  0x04;  // CW max current (Chip default)
static const uint8_t SM2135_35MA      =  0x05;
static const uint8_t SM2135_40MA      =  0x06;
static const uint8_t SM2135_45MA      =  0x07;  // Max value for RGB
static const uint8_t SM2135_50MA      =  0x08;
static const uint8_t SM2135_55MA      =  0x09;
static const uint8_t SM2135_60MA      =  0x0A;

static const uint8_t SM2135_CURRENT = (SM2135_20MA << 4) | SM2135_10MA;

void SM2135::setup() {
  ESP_LOGCONFIG(TAG, "Setting up SM2135OutputComponent...");
  this->data_pin_->setup();
  this->data_pin_->digital_write(true);
  this->clock_pin_->setup();
  this->clock_pin_->digital_write(true);
  this->pwm_amounts_.resize(5, 0);
}
void SM2135::dump_config() {
  ESP_LOGCONFIG(TAG, "SM2135:");
  LOG_PIN("  Data Pin: ", this->data_pin_);
  LOG_PIN("  Clock Pin: ", this->clock_pin_);
  ESP_LOGCONFIG(TAG, "  Total number of channels: 5");
}

void SM2135::loop() {
  if (!this->update_)
    return;

  uint8_t data[6];
  if (this->update_channel_ == 3 || this->update_channel_ == 4) {
    // No color so must be Cold/Warm
    data[0] = SM2135_ADDR_MC;
    data[1] = SM2135_30MA;
    data[2] = SM2135_CW;
    this->write_buffer_(data, 3);
    delay(1);
    data[0] = SM2135_ADDR_C;
    data[1] = this->pwm_amounts_[4];  //4  Warm
    data[2] = this->pwm_amounts_[3];  //3  Cold
    this->write_buffer_(data, 3);
	delay(2);
  } else {
    // Color
    data[0] = SM2135_ADDR_MC;
    data[1] = SM2135_45MA;
    data[2] = SM2135_RGB;
    data[3] = this->pwm_amounts_[1];  // Green
    data[4] = this->pwm_amounts_[0];  // Red
    data[5] = this->pwm_amounts_[2];  // Blue
    this->write_buffer_(data, 6);
	delay(2);
  }

  this->update_ = false;
}

}  // namespace sm2135
}  // namespace esphome
```

I've changed a number of things, the condition is fixed and there's a delay at the end so there's time for the ACK to happen. I think the chip wasn't ready for the next instruction even though new ones were being sent. Here's the YAML I'm using:

```
sm2135:
  data_pin: GPIO12
  clock_pin: GPIO14
  
output:
  - platform: sm2135
    id: output_red
    channel: 0
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_blue
    channel: 2
  - platform: sm2135
    id: output_white
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: rgbww
    name: "$devicename RGBWW"
    default_transition_length: 1s
    cold_white: output_white
    warm_white: output_warmwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    restore_mode: ALWAYS_ON
    red: output_red
    green: output_green
    blue: output_blue
```

hiddedb, 2020-03-24:
> I've just given it my best shot to put in some pull requests, but I'm new at github so I might have messed it up. For all you lovers out there, here's the sm2135.cpp that works for me:
>
> ```
> #include [...]

Thank you for this!! I will give this a try in the weekend and report back

hiddedb, 2020-03-24:
@RoelVideler 

I tried to upload to compile but I get this error

`Archiving /data/kantoor1/.pioenvs/kantoor1/libFrameworkArduino.a
Linking /data/kantoor1/.pioenvs/kantoor1/firmware.elf
/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::dump_config()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM213511dump_configEv+0x44): multiple definition of `esphome::sm2135::SM2135::dump_config()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM213511dump_configEv+0x44): first defined here
/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::loop()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM21354loopEv+0x18): multiple definition of `esphome::sm2135::SM2135::loop()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM21354loopEv+0x18): first defined here
/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::setup()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM21355setupEv+0x10): multiple definition of `esphome::sm2135::SM2135::setup()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM21355setupEv+0x10): first defined here
collect2: error: ld returned 1 exit status
*** [/data/kantoor1/.pioenvs/kantoor1/firmware.elf] Error 1
========================= [FAILED] Took 54.95 seconds =========================`

Am I doing something wrong?

`esphome:
  name: kantoor1
  platform: ESP8266
  board: esp01_1m
  includes:
    - custom_components/sm2135/sm2135.h
    - custom_components/sm2135/sm2135.cpp

sm2135:
  data_pin: GPIO12
  clock_pin: GPIO14
  
wifi:
  ssid: "blurredout"
  password: "temp123"
  
  manual_ip:
    static_ip: 192.168.50.85
    gateway: 192.168.50.1
    subnet: 255.255.255.0
    dns1: 192.168.50.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tuya Fallback Hotspot"
    password: "ObPG9RIlmLIg"

binary_sensor:
  - platform: status
    name: "kantoor1_status"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# Individual outputs
output:
  - platform: sm2135
    id: output_red
    channel: 0
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_blue
    channel: 2
  - platform: sm2135
    id: output_white
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: rgbww
    name: "kantoor1"
    default_transition_length: 1s
    cold_white: output_white
    warm_white: output_warmwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    restore_mode: ALWAYS_ON
    red: output_red
    green: output_green
    blue: output_blue`

r3mcos3, 2020-03-24:
I also tried it, for me it errors out with same error!

RoelVideler, 2020-03-24:
It looks like you might have sm2135 in both the components and custom
components folder. If you have it just in the custom components folder you
don’t need an include line.

hiddedb, 2020-03-24:
@RoelVideler 

I tried it without the include line and it gives this error.

`/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::dump_config()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM213511dump_configEv+0x44): multiple definition of `esphome::sm2135::SM2135::dump_config()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM213511dump_configEv+0x44): first defined here
/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::loop()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM21354loopEv+0x18): multiple definition of `esphome::sm2135::SM2135::loop()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM21354loopEv+0x18): first defined here
/data/kantoor1/.pioenvs/kantoor1/src/sm2135.cpp.o: In function `esphome::sm2135::SM2135::setup()':
sm2135.cpp:(.text._ZN7esphome6sm21356SM21355setupEv+0x10): multiple definition of `esphome::sm2135::SM2135::setup()'
/data/kantoor1/.pioenvs/kantoor1/src/esphome/components/sm2135/sm2135.cpp.o:sm2135.cpp:(.text._ZN7esphome6sm21356SM21355setupEv+0x10): first defined here
collect2: error: ld returned 1 exit status
*** [/data/kantoor1/.pioenvs/kantoor1/firmware.elf] Error 1
========================== [FAILED] Took 2.74 seconds ==========================`

But I wanted to try your cpp file cause of the bug fixes you described

RoelVideler, 2020-03-24:
I’m still seeing multiple locations of the cpp file, make sure you only
have one copy, same goes for the other files. You could even remove the new
one and see if it compiles, if it does you have another sm2135 driver
installed somewhere. Let me know if that doesn’t work.

hiddedb, 2020-03-25:
@r3mcos3 
@RoelVideler 

I had to create a new ESPHome device, then it worked!
So leave out the include lines Remco, create a new item from the esphome dashboard and it will work

r3mcos3, 2020-03-25:
I'ts working, great work!

the only thing to mention is the brightness off colors is low

hiddedb, 2020-03-28:
Unfortunately I just noticed on my test bulb, that it still doesn't function correctly. When I use the brightness slider in Home assistant the light goes from white to rgb colors. Any idea if this is fixable @RoelVideler ?

wojtek14a, 2020-05-06:
Is there any breakthrough in fixing issues mentioned above?
I'd really love to see the support for SM2135 in the stable version of ESPHome!

hiddedb, 2020-05-17:
> Is there any breakthrough in fixing issues mentioned above?
> I'd really love to see the support for SM2135 in the stable version of ESPHome!

As far as I'm aware, there are no current fixes at this moment :(

RoelVideler, 2020-05-18:
Sorry for the late reply, I haven’t managed to get the milliamperes part working. The max brightness is always the same on my bulb, both on the whites and the colors. As the bulb doesn’t support both rgb and white at the same time and esphome really doesn’t seem to support that, I used the idea posted previously to turn it into two bulbs, one CCW and one RGB and that works well, but it’s clunky.

RoelVideler, 2020-05-18:
This is the ESPHome YAML I'm using:
`sm2135:
  data_pin: GPIO12
  clock_pin: GPIO14
  
output:
  - platform: sm2135
    id: output_red
    channel: 0
  - platform: sm2135
    id: output_green
    channel: 1
    max_power: 0.46
  - platform: sm2135
    id: output_blue
    channel: 2
    max_power: 0.3
  - platform: sm2135
    id: output_white
    channel: 3
    max_power: 0.8
  - platform: sm2135
    id: output_warmwhite
    channel: 4
    max_power: 0.8

light:
  - platform: cwww
    name: "$devicename CWWW"
    cold_white: output_white
    warm_white: output_warmwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    restore_mode: ALWAYS_ON
    effects:
      - strobe:
      - flicker:
      - random:
          transition_length: 100ms
          update_interval: 150ms
  - platform: rgb
    name: "$devicename RGB"
    default_transition_length: 1s
    restore_mode: ALWAYS_OFF
    red: output_red
    green: output_green
    blue: output_blue
    id: light1`

hiddedb, 2020-05-18:
> This is the ESPHome YAML I'm using:
>
> ```
> sm2135:
>   data_pin: GPIO12
>   clock_pin: GPIO14
>
> output:
>   - platform: sm2135
>     id: output_red
>     channel: 0
>   - platform: sm2135
>     id: [...]

Hi Roel,

Thank you for your config, I'll give it a shot when I get home and report back!

T0ytoy, 2020-05-19:
@RoelVideler I just flashed one of my LSC E14 RGB light bulb using you method and the sm2135.cpp you shared in this thread. I had to use static IP to make it recognized in home assistant, but the issue doesn't seem to be with the firmware.

So far it's working really good for the cold white / warm white feature, I can adjust brightness and light temperature without any hiccups, excepted for the fact that the bulb turns off when the brightness sliders gets to ~25%. 
The RGB features works but the bulb is really, really dim, and you can't turn both the cwww and the rgb features at the same time.

Anyway, that's gonna be good enough for the time being, as secondary light bulbs in the bathroom. 

Thank you for your work!

sneh2j00, 2020-10-21:
I'm trying to set up a couple of new Genio 1002339 RGB+Warm White globes. 

I didn’t know they had changed internally until after I had flashed them with tuya-convert using the same ESPHome BIN I usually use. 

The light shows up as online most of time, but it occasionally goes offline in ESPHome. Unifi controller says its still connected to my wifi. OTA flashing through home assistant seems to work as it is.

I can get the light to turn on white by setting it to a yellow tone like others have mentioned, but when adding the sm2135 stuff to my yaml, I'm getting a different error to other people.. 

I'm getting an error message saying 'Component sm2135 cannot be loaded via YAML (no CONFIG_SCHEMA).'

Any ideas what to check from here?

Thanks!

BoukeHaarsma23, 2020-10-26:
> This is the ESPHome YAML I'm using:
>
> ```
> sm2135:
>   data_pin: GPIO12
>   clock_pin: GPIO14
>
> output:
>   - platform: sm2135
>     id: output_red
>     channel: 0
>   - platform: sm2135
>     id: [...]
Why not use the color_interlock feature? See: https://esphome.io/components/light/rgbw.html#rgbw-color-interlock
This way you can just use 1 light entity which disables the rgb parts when you turn on white.
My config is:

```
Light:
  - platform: rgbww
    name: "Light"
    cold_white: output_white
    warm_white: output_warmwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    default_transition_length: 0.2s
    red: output_red
    green: output_green
    blue: output_blue
    color_interlock: true
```

ryrzy, 2020-11-05:
would be nice to update this module. it doesn't work properly in esphome :) in tasmota it works perfect! and there is loot of led lights with this IC

https://github.com/arendst/Tasmota/blob/development/tasmota/xlgt_04_sm2135.ino

TheFes, 2021-02-19:
> Why not use the color_interlock feature? See: https://esphome.io/components/light/rgbw.html#rgbw-color-interlock
> This way you can just use 1 light entity which disables the rgb parts when you turn [...]

This seems to be working fine!

HA-Peet, 2021-03-12:
Hi guys,

Pretty new to homeassistant. I bought 2 GU10's recently without reading in to them not knowing there's no LSC integration. So far I installed the ESPhome add-on, made the custom_components dir and it's files. After reading this topic, I have some questions.

- How do I integrate the sm-part with home assistant?
- There's no esphome.yaml in the esphome-dir (File-editor). Do I have to make a node (where sm isn't 
   in the wizard)
- Where goes the esphome.yaml? Or is it part of the config.yaml?
- Anyone out there to describe the procedures necessary step-by-step?

TIA, Peet

ssieb, 2021-03-12:
This isn't Home Assistant.  You need to read the getting started docs for esphome.

TheFes, 2021-03-12:
> Hi guys,
>
> Pretty new to homeassistant. I bought 2 GU10's recently without reading in to them not knowing there's no LSC integration. So far I installed the ESPhome add-on, made the custom_components [...]

LSC is a rebrand of Tuya. use the Tuya app instead of the LSC app, and then you can use the Tuya integration

Thetmar, 2021-03-12:
Hi Peet,
I wrote an manual from a to z how to programm the LSC bulbs. When you sroll back you will find it. Also there are needed some drivers.

Greetings Thet

Vriendelijke groeten,
Thetmar Wiegers

hblandford, 2021-04-14:
Hi All,

thank you for your work on this.  I have a Mirabella Genio Wi-Fi Dimmable 19W BC RGB+CCT UFO LED Bulb, Item # I002930.

I flashed (took me ages as I was holding pins in place) ESPHome onto via the serial method and then I was very disappointed when I couldn't get the light to turn on properly.  I finally stumbled across this thread and am pleased to say it is now working using the custom SM2135 component.

My config is below, I initially had issues with the colors of the bulb being incorrect:

    esphome:
      name: ufo01
      platform: ESP8266
      board: esp01_1m
    
    wifi:
      ssid: !secret device_ssid
      password: !secret device_ssid_key
    
    # Enable logging
    logger:
    
    # Enable Home Assistant API
    api:
    
    ota:
    
    sm2135:
      data_pin: GPIO12
      clock_pin: GPIO14
      
    output:
      - platform: sm2135
        id: output_green
        channel: 0
      - platform: sm2135
        id: output_blue
        channel: 1
      - platform: sm2135
        id: output_red
        channel: 2
      - platform: sm2135
        id: output_white
        channel: 3
      - platform: sm2135
        id: output_warmwhite
        channel: 4
    
    light:
      - platform: rgbww
        name: "ufo01"
        default_transition_length: 1s
        cold_white: output_white
        warm_white: output_warmwhite
        cold_white_color_temperature: 6500 K
        warm_white_color_temperature: 2700 K
        red: output_red
        green: output_green
        blue: output_blue
        color_interlock: true
        # Ensure the light turns on by default if the physical switch is actuated.
        restore_mode: ALWAYS_ON
    
    sensor:
      - platform: wifi_signal
        name: "ufo01 WiFi Signal"
        update_interval: 60s

@RoelVideler being a complete beginner at this.  Can you explain why you changed

    static const uint8_t SM2135_CW        =  0xC5; // 0x80;  // CW channel (Chip default)

from 0x80 to 0xC5?  What does that do?

The white value slider does not seem to do much at all with my bulb.  Temperature and brightness are working.

Also, an example config you posted had different max_power values for the green and blue channel and none for the red.  Was that because of properties of your bulb?  Was it just because you preferred it that way.

Thanks again everyone for their work.

closed 2021-05-12 by jesserockz