esphome/feature-requests #446: add option to add hosts in /etc/hosts file in hassio config

lpt2007, 2019-10-24:
**Describe the problem you have/What new integration you would like**

Because of this error I must manualy add host in /etc/hosts file:
WARNING Error resolving IP address of witty_cloud_ldr_rgb_001.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 1 seconds

And If this in host file, error is gone:  
witty_cloud_ldr_rgb_001.local

I like to add hosts trough config under hassio addon

thx in advance

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-10-24:
1. So do I understand correctly that you already tried this with the Hass.io addon? Did you go into the container and manually edit the /etc/hosts file and confirmed that worked? Please post an example of your edited /etc/hosts file, the entry could be added to various places there.

2. Have you considered setting a static IP as the warning states? Another option is to set `use_address`

lpt2007, 2019-10-26:
Hi, 

Yes I edited /etc/hosts file manually in the container and I added ip and hostnames of esphome devices. Like this:

```
192.168.2.1  esphome_device_1
192.168.2.2 esphome_device_2
```

> Have you considered setting a static IP as the warning states? Another option is to set use_address

I don't wanna add static IP to my devices if possible.

I think the beater way to add devices is trough config inside addon. Like this:
```
{
  "hosts": [
    "192.168.2.1  esphome_device_1",
    "192.168.2.2  esphome_device_2"
  ],
}
```

OttoWinter, 2019-10-26:
See `use_address`, that is pretty much equivalent to what you're doing right now. If you set use_address, it will use that IP address for all connections without having to set a static IP.

closed 2019-10-26 by OttoWinter

lpt2007, 2019-10-26:
I will try that. Thanks.