esphome/feature-requests #447: Easydriver ESPhome support

iMike78, 2019-10-24:
**Describe the problem you have/What new integration you would like**
I'm just wondering is there any possibilities to add easydriver stepper driver support to the supported devices list? There is only two chipsets are supported, but they are for 4 phase motors.

**Please describe your use case for this integration and alternatives you've tried:**
I built a motorized roller blind, where I used an arduino code with esp8266, that's what I would like to swap with ESPhome.

**Additional context**
Any other 2 phase motor driver would be appreciated.

pstapley, 2019-11-28:
I currently have working code for the Easydriver. My question is do we want the step size configurable via ESPHome? Or just let the user pull MS1 and MS2 high or low depending on the step size they require.

coogle, 2020-12-27:
Did anything come of this? If there is Easydriver code laying around I'd be interested in maybe polishing it up. I have a project for an automatic fish feeder in need..

pauldogg, 2021-08-23:
I'd love to see this happen!
Thanks