esphome/feature-requests #451: Auto-calibration for AS3935

RichardDern, 2019-10-26:
**Describe the problem you have/What new integration you would like**

Auto-calibration implementation for AS3935 lightning sensor would be a nice feature to have.

**Please describe your use case for this integration and alternatives you've tried:**

The AS3935 can be auto-calibrated. Inspiration could be taken from Embedded Adventures and their [MOD-1016](https://github.com/embeddedadventures/MOD-1016).

Maybe the calibration could be automatically done every x days or something.

**Additional context**

OttoWinter, 2019-10-26:
As far as I see, there's two types of calibration:

First, the RCO calibration - both the TRCO and SRCO are separate oscillators on the chip that can get out of sync. A simple CALIB_RCO (0x3D) command (see page 23 of datasheet) calibrates that.

Second, there's the resonance frequency tuning - according to the datasheet (page 23) this needs to be within ±3.5% for optimal performance. It looks like the intended way is to sample at all 4 frequency division ratios and see which one gets closest to 500kHz.

But it doesn't look like the calibration would need to be done often - the datasheet says that the values are automatically compensated for temperature etc. Power cycles will clear those settings out though.

CC @TheEggi - do you know more about the AS3935 and its accuracy without tuning/calibrating?

RichardDern, 2019-10-26:
Maybe it doesn't need to be re-calibrated often, but it needs to be at least once, otherwise Embedded Adventures wouldn't provide a calibration frequency specific to each MOD-1016 (a number between 0 to 15). Fortunately, they provide the source code for auto-calibration for people like me who have lost this number 😄 

If something like scheduled auto-calibration wouldn't be done, maybe it would worth it to implement auto-calibration at boot time.

TheEggi, 2019-10-27:
@OttoWinter the intended use was for an outdoor weather station and I have to say that I never had any thunderstorms lately. My tests were pretty much all indoor by just using a lighter, which at least triggered the IRQ pin -  just by using the default values.

SparkFun did also not provide any of those functions in their libs - Some more details about the settings https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20/all

RichardDern, 2019-10-27:
That's why I linked the Embedded Adventures library: it is a bit more documented and features-rich, and works perfectly in a stand-alone Arduino project (the demo project provided in the repository).

OttoWinter, 2019-10-28:
@RichardDern Yeah, but I would say let's get some empirical data on this first. If sparkfun didn't include this in their libs it probably doesn't make that big of a difference. I would suggest let's first have some users use the AS3935 and if it's inaccurate we can still do this.

RichardDern, 2019-10-28:
Sparkfun do provide calibration example [here](https://github.com/sparkfun/SparkFun_AS3935_Lightning_Detector_Arduino_Library/blob/master/examples/Example3_Tune_Antenna_I2C/Example3_Tune_Antenna_I2C.ino).

marek-obuchowicz, 2021-07-28:
I've build a proof of concept implementation, which enables antenna trimming mode and basic calibration.
Note, that for fully automatic calibration - we should enble trimming mode, measure frequency of signal from INT pin, multiply that by `div_ratio` and check if matches 500kHz +/- 3.5%, if not - adjust parameters accordingly. At the moment I don't have time and knowledge to implement antenna trimming. I've also checked AS3935 sensors I had in my workshop - both of them seem to work in optimal range, without need of any change - so I think that even though this PR solves calibration issue only partially, it could make situation better for users, who have sensors with correctly trimmed antenna frequency.

I haven't been touching C for a while, so please review my changes thoroughly. My units have those changes in place and seem to work. Still need to wait for next thunderstorm to check, if sensor readings got better...

drawgas, 2023-07-19:
Any chance we could revive this and get @marek-obuchowicz changes merged?

mveinot, 2023-08-27:
Also requesting this code/feature be reopened/merged.

marek-obuchowicz, 2023-09-08:
I can sadly only say that I do not have time/capacity to work on this, especially that it requires access to hardware which I don't have next to me right now. If someone can pick it up, I would be very appreciated, apparently this PR brings value to community, it would be therefore pity to have it not merged as it's "almost ready" and needs only small refactoring.

mveinot, 2023-09-08:
You can follow any progress here https://github.com/esphome/esphome/pull/5366

I've merged it against the current dev branch and implemented all the requested changes. It's gotten through the automated CI tests. Just waiting for a human code review.

mveinot, 2023-09-08:
You can follow any progress here: https://github.com/esphome/esphome/pull/5366

I've merged it against the current dev branch, implemented requested changes from this PR review and gotten it through the automated CI testing.

Waiting on a peer review.

mveinot, 2023-09-08:
You can follow any progress here: esphome/esphome#5366

I've merged it against the current dev branch, implemented requested changes from this PR review and gotten it through the automated CI testing.

Waiting on a peer review.