esphome/feature-requests #452: Strikes count sensor for AS3935

RichardDern, 2019-10-26:
**Describe the problem you have/What new integration you would like**

It might be nice to have a strikes count sensor for the AS3935.

**Please describe your use case for this integration and alternatives you've tried:**

Every time the sensor publishes an update, it could also send number of strikes that happened since the last update. It would be reset to 0 after updating corresponding sensor.

**Additional context**

OttoWinter, 2019-10-26:
I don't think the AS3935 exposes that data. It mainly only exposes if a lightning storm is nearby and how far away it is. I think the sensor is mainly made for UPS and devices that need to know if there might be power surges.

See also the datasheet https://www.embeddedadventures.com/datasheets/AS3935_Datasheet_EN_v2.pdf

CC: @TheEggi (do you know if the AS3935 exposes that data?)

RichardDern, 2019-10-26:
It doesn't, but one could use a global variable increased every time the IRQ is triggered, and reset to 0 after data is sent.

OttoWinter, 2019-10-26:
In that case I would recommend using a separate `pulse_counter` sensor. Connect the IRQ pin to your ESP and count the pulses with that. It's then also in a nice format in pulses/min (or another time unit of your choosing) and you can integrate it over time with `integration`.

RichardDern, 2019-10-26:
Oh, I haven't thought about using the pulse counter. That's a much better idea !

RichardDern, 2019-10-26:
Actually, I think the pulse counter solution won't work, as you need to check for every pulse wether it's noise, human-originated or actual lightning. At least, [this is how the MOD-1016 works](https://github.com/embeddedadventures/MOD-1016/blob/master/examples/MOD1016_demo/MOD1016_demo.ino#L96).

TheEggi, 2019-10-27:
The binary sensor should only get triggered if there was a real interrupt - but I guess there was still an issue with the handling. Thanks for fixing that @OttoWinter

OttoWinter, 2019-10-28:
@TheEggi No no the previous handling was fine! I just realized when going through the datasheet that the IRQ line gets pulled high *until the data is read out*. So using a pin interrupt for that was a bit unnecessary, I just switched out the interrupt for a polling mode.