esphome/feature-requests #453: Bi-directional communication over serial, i.e. to report the status of a projector

jschwalbe, 2019-10-26:
**Describe the problem you have/What new integration you would like**

Currently I can send text via serial / UART with the `UART Switch` component, but I'd like to also receive text.

**Please describe your use case for this integration and alternatives you've tried:**

Here's my use case: Projector is downstairs. Sometimes it gets turned on accidentally and that'd not good! It doesn't turn off automatically, so I'd like to know when it's on for > x minutes so I can then send a command to turn it off.
I have used `esp-link` to successfully send commands and receive the response. I am currently unaware if ESPHome can handle this. If it can, I'm not sure how. :) But I'd prefer to use ESPHome bc of how nicely it integrates w/ hass.

**Additional context**

OttoWinter, 2019-10-26:
Duplicate of https://github.com/esphome/feature-requests/issues/213

closed 2019-10-26 by OttoWinter

jschwalbe, 2019-10-26:
I've been asking on the Discord server for a few days.. no replies. Can you give me any hints, @OttoWinter? Thanks

jschwalbe, 2019-10-26:
Also - I'd contend that it's not a duplicate, because you asked for a use-case scenario, which I provided.

danibjor, 2019-10-27:
+1 here on being able to receive text, or partial match. 

Could be nice to have a binary switch where it triggers on/off on two predefined messages on the uart port. Or a sensor, where you define a lambda to handle parsing of the incoming text and return a value. 
This would be flexible and very powerful. I could think of a lot of use cases for this.

danibjor, 2019-10-27:
Or something like
new_component:
  platform: uart
    uart_id: bus_2
    - on_message:
      starts_with: [0x42, 0x07]
      - then..
    - on_message: “power on 1”
      - then..

OttoWinter, 2019-10-27:
Looks like I linked the wrong issue, but I can't find the one I was thinking about...

So I'll just repeat what I said in that other issue I can't seem to find.

The problem is that a serial bus is an (infinite) stream, whereas text_sensor etc consist of discrete states.

UART does not have a notion of messages/frames. But we need some way to find out where a message ends to transform that into a "state".

First, about how we could transform that into states. One could simply say: a) use the linebreak character to denote a discrete message. b) Another option would be some user function that gets called to check if a message is complete. However, a) will only work for few protocols (and mostly custom-made ones). And b) will be hard to define a function that doesn't block the main thread but allows parsing.

Second, how the state is transferred. There's *no* guarantee in ESPHome that all intermediary states are sent to the instance at the other end. If you update a text sensor very quickly, the states will not be buffered anywhere (there's no memory to do so), and you will lose data. So that would make this very hard to work with on the other end, because you don't know if you received everything.

Third, there's a minor problem with the message content. If you have a binary protocol (not ASCII letters) then the messages will be sent over binary too and ESPHome will happily do that. However, if you use this with Home Assistant, it will just reject the message because everything is UTF-8 with HA, binary blobs are not valid states.

So in conclusion, I don't see a way to create such a component that can reliably work well in most scenarios people would want to use it for. That's why these people should use a custom component (extending UARTDevice) that handles the parsing logic itself. 

If you do know a way to handle all these issues well please re-open this feature request or create a new one with a proposal of how the YAML part could look like and how these issues would be addressed. 

It really isn't that hard to create a custom UART text sensor (see website, it's just a couple of lines). There's even an Arduino over i2c API example in the cookbook now, while it's not i2c the same principles apply.

danibjor, 2019-10-27:
@OttoWinter You raise many valid concerns here. I’ll have to think about these for a while..

szurr, 2019-10-27:
Maybe this will be possible with mysensors library?