esphome/feature-requests #454: Switchbot support - BLE

nemccarthy, 2019-10-27:
**Describe the problem you have/What new integration you would like**

It would be great to build a switch component which could be used to control SwitchBot devices over BLE.

https://www.switch-bot.com/

**Please describe your use case for this integration and alternatives you've tried:**

Switchbot devices seem to have a few connectivity issues with HomeAssistant and the RaspberryPi BLE transmission isn’t that powerful. 

Having an ESPhome act as a BLE wifi bridge to home assistant would be great. The API for the switchbot seems pretty simple: https://github.com/OpenWonderLabs/python-host

Would be great start to doing BLE transmission to control other BLE output devices. 

**Additional context**

OttoWinter, 2019-10-28:
Just FYI, I would say this is unlikely to happen in the nearer future. ESPHome currently only handles reading BLE passive ADV packets. The device noted here requires BLE connections.

SchmDan, 2020-06-04:
What's the current state on this request? I would really love this component!

plantoschka, 2020-07-31:
I would also love this.

ronytomen, 2021-03-13:
I would like to add that someone has written the following:
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32

This includes CurtainBot support.

marithpl, 2021-04-29:
Is any chance for Switchbot Curtain Support?

firstdominator, 2021-05-19:
Any updates on this? Anyway to use an esp32 with esphome to control switbot bot without an hub?

Thanks!

firstdominator, 2021-05-19:
> I would like to add that someone has written the following:
> https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32
> 
> This includes CurtainBot support.

Did you managed to make it working with MQTT broker in home assistant?

marithpl, 2021-05-19:
> > I would like to add that someone has written the following:
> > https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32
> > This includes CurtainBot support.
> 
> Did you managed to make it working with MQTT broker in home assistant?

Yes I did.
But next few days should be released version with MQTT Auto Discovery, so devices should automatically add to MQTT Integration.

mplacona, 2021-05-19:
> Yes I did.
> But next few days should be released version with MQTT Auto Discovery, so devices should automatically add to MQTT Integration.

Do you think it’s a possibility that the above can be made into an ESPHome integration?

marithpl, 2021-05-19:
Im not a dev :(
But I wish to have it in ESPHome

firstdominator, 2021-05-28:
Any news on this ? Since esphome release 1.18.0 there is BLE client support. So this means that adding switchbot might be feasible.

Almost here guys!

ry8s, 2021-06-14:
Another bump! I'm also trying to get switch boys and curtains going, I might try the MQTT route but it would be awesome if these were setup in espHome!

randysnel, 2021-07-14:
Warning note upfront:  I'm no expert in ESPhome/BLE and are not experienced in this work on getting stuff to work.

That being said, i've tried to connect to a switchbot from esphome. I've used the Android app BLE Scanner on my phone and held it near my switchbot curtain motor to discover the MAC-address of the Switchbot. This was fairly easy, I'm not aware of another method to gain this information. For what is worth, I have two switchbot motors for my curtains which open from the middle.  

I've added the following code to my ESPHome:

```
esp32_ble_tracker:
  scan_parameters:
    active: false #set to true when connecting new devices

ble_client:
  - mac_address: D2:9B:5D:XX:XX:XX
    id: switchbot_a

```

Below is the log output from ESPHome, it is the same for both switchbot curtain motors:

```
[10:56:47][D][ble_client:045]: Found device at MAC address [D2:9B:5D:xx:xx:xx]
[10:56:47][I][ble_client:083]: Attempting BLE connection to d2:9b:5d:xxxx:xx
[10:56:48][I][ble_client:159]: Service UUID: 0x1800
[10:56:48][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[10:56:48][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[10:56:48][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[10:56:48][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[10:56:48][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[10:56:48][I][ble_client:159]: Service UUID: 0x1801
[10:56:48][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xa
[10:56:48][I][ble_client:159]: Service UUID: CBA20D00224D11-E69F-B800-02A5-D5C51B
[10:56:48][I][ble_client:160]:   start_handle: 0xb  end_handle: 0xffff
[10:56:48][I][ble_client:339]:  characteristic CBA20002224D11-E69F-B800-02A5-D5C51B, handle 0xd, properties 0xc
[10:56:48][I][ble_client:339]:  characteristic CBA20003224D11-E69F-B800-02A5-D5C51B, handle 0xf, properties 0x10
```

I'm trying to cross reference information from the  [ESP32 BLE gateway code](https://github.com/combatistor/ESP32_BLE_Gateway/blob/master/Switchbot_NimBLE_Client/Switchbot_NimBLE_Client.ino) but I'm not really sure how to continue in adding the correct sensor information with the correct characteristics. If someone has more knowledge or ideas, please do share as I'm fine with future testing.

marithpl, 2021-07-14:
> Personal warning note upfront: I'm no expert in ESPhome/BLE and are not experienced in this work on getting stuff to work.
> Warning note from BLE client as part of ESPHome: maximum is three clients, the code is known to be memory intensive.
> 
> That being said, i've tried to connect to a switchbot from esphome. I've used the Android app BLE Scanner on my phone and held it near my switchbot curtain motor to discover the MAC-address of the Switchbot. This was fairly easy, I'm not aware of another method to gain this information. For what is worth, I have two switchbot motors for my curtains which open from the middle.
> 
> I've added the following code to my ESPHome:
> 
> ```
> esp32_ble_tracker:
> 
> ble_client:
>   - mac_address: D2:9B:5D:XX:XX:XX
>     id: switchbot_a
> ```
> 
> Below is the log output from ESPHome, it is the same for both switchbot curtain motors:
> 
> ```
> [10:56:47][D][ble_client:045]: Found device at MAC address [D2:9B:5D:xx:xx:xx]
> [10:56:47][I][ble_client:083]: Attempting BLE connection to d2:9b:5d:xxxx:xx
> [10:56:48][I][ble_client:159]: Service UUID: 0x1800
> [10:56:48][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
> [10:56:48][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
> [10:56:48][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
> [10:56:48][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
> [10:56:48][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
> [10:56:48][I][ble_client:159]: Service UUID: 0x1801
> [10:56:48][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xa
> [10:56:48][I][ble_client:159]: Service UUID: CBA20D00224D11-E69F-B800-02A5-D5C51B
> [10:56:48][I][ble_client:160]:   start_handle: 0xb  end_handle: 0xffff
> [10:56:48][I][ble_client:339]:  characteristic CBA20002224D11-E69F-B800-02A5-D5C51B, handle 0xd, properties 0xc
> [10:56:48][I][ble_client:339]:  characteristic CBA20003224D11-E69F-B800-02A5-D5C51B, handle 0xf, properties 0x10
> ```
> 
> I'm trying to cross reference information from the [ESP32 BLE gateway code](https://github.com/combatistor/ESP32_BLE_Gateway/blob/master/Switchbot_NimBLE_Client/Switchbot_NimBLE_Client.ino) but I'm not really sure how to continue in adding the correct sensor information with the correct characteristics. If someone has more knowledge or ideas, please do share as I'm fine with future testing.

The Switchbot BT MAC address you can find in Switchbot App, so you don't have to scan network around you.
The MAC address you can display by clicking tree dots in right top corner.
[image]

randysnel, 2021-07-15:
I tried quite a few things with the BLE_client function, however the Switchbot itself keeps disconnecting, possibly due to energy conservation reasons. I managed to read the battery with the code below, however due to the disconnects the sensor status in HA switches from the current percentage to 'unknown' until the ESP connects to the Switchbot again and requests the value again. That value is stored in the HA sensor for 1-3 seconds until it is overwritten to unknown again due to the disconnect cycle.

```
ble_client:
  - mac_address: D2:9B:5D:xx:xx:xx
    id: switchbot_a
    
sensor:
  - platform: ble_client
    ble_client_id: switchbot_a
    name: "Switchbot A battery"
    service_uuid: '1800'
    characteristic_uuid: '2A00'
    device_class: battery
    icon: 'mdi:battery'
    unit_of_measurement: '%'
```
Reading values of the Switchbot curtain motors is fun on its own of course, but from the code i found at [this repository](https://github.com/OpenWonderLabs/python-host/blob/master/switchbot_py3.py) seems we need to write to a service to call an action on the Switchbot like close or open. See line 96-112 for specific calls and required format.

In the documentation of the [BLE Client of ESPHome](https://esphome.io/components/ble_client.html) I can only observe a sensor function as a usable component, which can only read as far as i can tell.  With further digging it seems that there more components which are not mentioned on the BLE_Client page itself: 

- [BLE Client Switch](https://esphome.io/components/switch/ble_client.html) available as well, but the only thing that can do is enable or disable a specific BLE Client as specified in the ESPHome code with mac address & id. This might be useful if you want to bypass the limit of 3 ble_clients concurrent connections limit. 
- [BLE Signal strength sensor](https://esphome.io/components/sensor/ble_rssi.html) to measure the signal strength of a BLE device
- [Xiamoi BLE sensor](https://esphome.io/components/sensor/xiaomi_ble.html) which is specific for Xiaomi BLE sensors
- [BLE presence](https://esphome.io/components/binary_sensor/ble_presence.html) component to see if a BLE device is present.
- [BLE Scanner](https://esphome.io/components/text_sensor/ble_scanner.html) the scanner for devices nearby
- [BLE  Tracker Hub](https://esphome.io/components/esp32_ble_tracker.html) creates a global hub so that you can track bluetooth low energy devices
- [BLE Beacon](https://esphome.io/components/esp32_ble_beacon.html) to create your own BLE beacon

Looking at the feature requests of ESPhome regarding BLE, I cannot observe a request for writing to BLE services for the BLE_client function. In [this pull request](https://github.com/esphome/esphome/pull/1177) I can observe that the work for writing to services is something that could be done in the future as an enhancement. The codebase/supporting libraries of ESPHome supports writing to services as I can observe from other platforms that can control a device through BLE and ESPHome.

Currently I cannot see a method to manage the Switchbot curtain motors from the ble_client function due to the missing write function. It seems a custom platform could be written to support the Switchbot functions more natively in ESPHome, however that is not something I am able to undertake.

iMicknl, 2021-07-27:
@randysnel I am looking into this as well! Unfortunately I don't have a SwitchBot Curtain yet, but I am looking to buy them for my new house in a two months. It should be doable to add this to ESPHome as well and have all functionality, I just wonder how the retries should be implemented since I read that the BT LE connection is not super reliable.

It seems that they broadcast the sensor data in the advertising service data;

https://github.com/OpenWonderLabs/node-switchbot/blob/6427e72e2d590864014635a732f4f0a65f598f00/lib/switchbot-advertising.js#L178-L201

Would that mean that the only issue is the 'write' functionality, which should be implemented in C++ most likely? I could have a look at that in the coming months, if no one has fixed it beforehand.

devWaves, 2021-07-31:
Hi Everyone, I wrote the ESP32 MQTT code https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32

If people have questions I am also on the HA forums.

if people are working on an ESPHome switchbot integration that's awesome. Implementing all the features of my code may be tricky though, but I can provide info if needed

I am still new to HA and ESPHome. I'm no C++/arduino expert either. I can give you some info that I learned while experimenting with switchbots though...

1. They require a connection to send commands. In my code I only keep the connection active for the duration of the command action, then I disconnect after

2. Subscribing to BLE notifications of the device will return the response of the command sent to it. Possible values range from 1,5 = success, 3 = busy , 7,9 etc = error

3. Getting the latest status/state/attributes of the device can be done via a passive BLE scan. This is information that is being constantly broadcasted by the device and does not require a "connection". The meter/temp sensor ONLY need a passive scan and no connection

4. The only attributes to collect that required a temporary connection were for the bot (firmware, hold secs,num of timers). The curtain attributes can all be collected passively without a connection, same for meter

5. You will need to implement retry logic. My code can retry on all of these issues:
a) retry connection until success
b) retry send until success
c) retry if bot is busy until success
d) retry if the bot or curtain doesnt send a response (no response may still trigger bot, so this you can turn off if needed). Curtain doesnt really matter if you resend the same command with no response

6. if you want to do something like set the hold second value on the bot and then after call the press/on/off command, you can keep the connection active between commands, and then disconnect. This will make it action much faster

cheers

ppanagiotis, 2021-08-17:
Any update with cover and `ble_client` ?

Nardol, 2021-11-24:
In a first time, it could be great to have meters support as they only use BLE advertising.
I still not really understand how to develop with ESPHome so not able to add it as a native component, but I found a website where only the code has been understandable for me :slightly_smiling_face: but so useful: https://xoops.hypweb.net/modules/UsersWiki/person/nao-pon/blog/2021-03-17

I use it with success.
A code example:
```yaml
esphome:
  name: "Meters Hub"
  platform: ESP32
  # Reste has to be adapted for your ESP
  board: esp32dev
  platformio_options: 
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L

logger:
  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
  level: NONE

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  ap:
    password: !secret ap_key

captive_portal:

# API Home Assistant
api:
  password: !secret api_pass

ota:
  password: !secret ota_pass

sensor:
  - platform: template
    name: "Hygrometer"
    id: hygrometer
    unit_of_measurement: '%'
    accuracy_decimals: 0
    device_class: humidity
    icon: "mdi:water-percent"
  - platform: template
    name: "Thermometer"
    id: thermometer
    unit_of_measurement: '°C'
    device_class: temperature
    accuracy_decimals: 1
    icon: "mdi:thermometer"
  - platform: template
    name: "RSSI"
    id: rssi_meter
    unit_of_measurement: 'dB'
    device_class: signal_strength
    accuracy_decimals: 0
    icon: "mdi:bluetooth"
  - platform: template
    name: "Battery"
    id: battery_meter
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
esp32_ble_tracker:
  on_ble_advertise:
    - mac_address: xx:xx:xx:xx:xx:xx # MAC address in lowercase
      then:
        - lambda: |-
            for (auto data : x.get_service_datas()) {
                if(data.data.size() == 6) {
                  float temperature = (float)(data.data[4] & 0b01111111) + ((float)(data.data[3] & 0b00001111) / 10);
                  if (!(data.data[4] & 0b10000000)) {
                    temperature = -temperature;
                  }
                  int8_t humidity= data.data[5] & 0b01111111;
                  int8_t battery = data.data[2] & 0b01111111;
                  int8_t rssi=x.get_rssi();
                  id(hygrometer).publish_state(humidity);
                  id(thermometer).publish_state(temperature);
                  id(rssi_meter).publish_state(rssi);
                  id(battery_meter).publish_state(battery);
                }
            }
```

Things could be better with a native ESPHome component, maybe.

iMicknl, 2021-12-04:
I just received my Switchbut Curtain U-Rail 2 (and a 1) and I will try to port the ESP32 mqtt code to an ESPHome component, however I could use some help.

Would there be someone willing to tackle this with me? In my experience it moves way faster forward if it isn't just a side project that I tackle alone. 😄

Stephan296, 2021-12-10:
@iMicknl i want to help you!

iMicknl, 2021-12-10:
@Stephan296 can you ping me on Discord? (iMick #1903). I am quite busy, but would love to work on this when I have some time. (currently waiting for my curtains to be delivered actually...)

nmajin, 2022-01-04:
Linking https://github.com/esphome/esphome/pull/2776 since it was mentioned there might be some code functionality for this.

jeroenterheerdt, 2022-03-30:
@iMicknl did you ever get any further with this? I'd love to see this integrated somehow

iMicknl, 2022-03-30:
> @iMicknl did you ever get any further with this? I'd love to see this integrated somehow

Not really... I have to admit that my SwitchBots are currently not mounted to the rails at all, had some other priorities lately.

However, I would be interested to see if we can tackle this together with some devs. Happy to contribute.

Ulrar, 2022-04-21:
Hi,

I'm not familiar with ESPHome's code base, but I do have a pair of switchbot curtains mounted and working so I'd be happy to help, if I can do anything. Guess I'll start reading [these](https://esphome.io/custom/custom_component.html).

ppanagiotis, 2022-04-24:
[This](https://github.com/esphome/esphome/pull/3398) is almost what we want. If the connection problem resolved we can have switchbot (bots / covers) at esphome :dancers:

Ulrar, 2022-04-24:
What atr you calling cover, the switchbot curtains ?
That's great news. I wonder if these connections issues are why I've been having random delays in commands using switchbot mqtt ble esp32

ppanagiotis, 2022-04-24:
Yes curtains! 
And yes I think this is the reason of the delay. mqtt2ble is trying to reconnect before each action. So this adds the extra delay you are describing. I hope something similar can be done with esphome. (I prefer the delay than fail :-p)

ppanagiotis, 2022-04-30:
For anyone interested to debug some steps.

1. git clone git@github.com:rbaron/esphome.git
2. cd esphome/
3. git checkout delayed-ble-write-action
4. python3 -m venv .venv
5. source .venv/bin/activate
6. python3 setup.py install
7. create a file `testing_switchbot.yaml`

```yaml

esphome:                                                                           
  name: switchbots_test
  platform: ESP32                                                                  
  board: nodemcu-32s                                                               
                                                                                   
wifi:                                                                              
  ssid: "YOUR_SSID"                                                            
  password: "YOUR_PASSWORD"                                                           
  ap:                                                                              
    ssid: "FALLBACK"                                                          
    password: "FALLBACK_PASSWORD"                                                           
                                                                                   
captive_portal:                                                                    
                                                                                   
api:                                                                               
                                                                                   
ota:                                                                               
                                                                                   
# Enable logging                                                                   
logger:                                                                            
                                                                                   
esp32_ble_tracker:                                                                 

ble_client:                                                                        
  # switchbot device macaddress
  - mac_address: XX:XX:XX:XX:XX:XX
    id: switchbotone                                                               
  # curtain device macaddress
  - mac_address: YY:YY:YY:YY:YY:YY                                                
    id: curtainone
                                                                                   
button:                                                                            
  - platform: template                                                             
    name: "switcbotonebutton"                                                                 
    on_press:                                                                      
      - ble_client.ble_write:                                                      
          id: switchbotone                                                         
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
          # List of bytes to write.                                                
          value: [0x57, 0x01] 

switch:
  - platform: template                                                             
    name: curtainoneswitch
    turn_on_action:                                                                
      - ble_client.ble_write:                                                      
          id: curtainone                                                        
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
          # List of bytes to write.                                                
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x00]                    
    turn_off_action:                                                                
      - ble_client.ble_write:                                                      
          id: curtainone                                                     
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
          # List of bytes to write.                                                
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x64]                    

```

8. esphome testing_switchbot.yaml compile
9. esphome testing_switchbot.yaml upload
10. esphome testing_switchbot.yaml logs

stich86, 2022-05-18:
@ppanagiotis what's the current situation with that build?
I wanto to move 3 ESP (2 8266 and 1 32) to a single build an remove the [SwitchBot-MQTT-BLE-ESP32](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32).

I just need to send command to a Switch Bot Curtain. Is it stable?

thx

ppanagiotis, 2022-05-18:
> @ppanagiotis what's the current situation with that build? I wanto to move 3 ESP (2 8266 and 1 32) to a single build an remove the [SwitchBot-MQTT-BLE-ESP32](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32).
> 
> I just need to send command to a Switch Bot Curtain. Is it stable?
> 
> thx

Haven't tested status of the curtain / button, but open / close works great! I remove all mqtt-ble and have only esphome for 5 curtains + 1 button.

pokemon81, 2022-05-19:
how can I include this source in the official esphome of homeassistan? I only use the homeassistant own version. So that cihd sensors all appear in one interface.

fabiosoft, 2022-05-27:
You can try using external component (still on GitHub dev) or follow progress here: https://github.com/esphome/esphome/pull/3398

> how can I include this source in the official esphome of homeassistan? I only use the homeassistant own version. So that cihd sensors all appear in one interface.

TheFes, 2022-06-09:
> For anyone interested to debug some steps.
> 
> 1. git clone [git@github.com](mailto:git@github.com):rbaron/esphome.git
> 2. cd esphome/
> 3. git checkout delayed-ble-write-action
> 4. python3 -m venv .venv
> 5. source .venv/bin/activate
> 6. python3 setup.py install
> 7. create a file `testing_switchbot.yaml`
> 
> ```yaml
> esphome:                                                                           
>   name: switchbots_test
>   platform: ESP32                                                                  
>   board: nodemcu-32s                                                               
>                                                                                    
> wifi:                                                                              
>   ssid: "YOUR_SSID"                                                            
>   password: "YOUR_PASSWORD"                                                           
>   ap:                                                                              
>     ssid: "FALLBACK"                                                          
>     password: "FALLBACK_PASSWORD"                                                           
>                                                                                    
> captive_portal:                                                                    
>                                                                                    
> api:                                                                               
>                                                                                    
> ota:                                                                               
>                                                                                    
> # Enable logging                                                                   
> logger:                                                                            
>                                                                                    
> esp32_ble_tracker:                                                                 
> 
> ble_client:                                                                        
>   # switchbot device macaddress
>   - mac_address: XX:XX:XX:XX:XX:XX
>     id: switchbotone                                                               
>   # curtain device macaddress
>   - mac_address: YY:YY:YY:YY:YY:YY                                                
>     id: curtainone
>                                                                                    
> button:                                                                            
>   - platform: template                                                             
>     name: "switcbotonebutton"                                                                 
>     on_press:                                                                      
>       - ble_client.ble_write:                                                      
>           id: switchbotone                                                         
>           service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
>           characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
>           # List of bytes to write.                                                
>           value: [0x57, 0x01] 
> 
> switch:
>   - platform: template                                                             
>     name: curtainoneswitch
>     turn_on_action:                                                                
>       - ble_client.ble_write:                                                      
>           id: curtainone                                                        
>           service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
>           characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
>           # List of bytes to write.                                                
>           value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x00]                    
>     turn_off_action:                                                                
>       - ble_client.ble_write:                                                      
>           id: curtainone                                                     
>           service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
>           characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
>           # List of bytes to write.                                                
>           value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x64]                    
> ```
> 
> 8. esphome testing_switchbot.yaml compile
> 9. esphome testing_switchbot.yaml upload
> 10. esphome testing_switchbot.yaml logs

I'm using a Switchbot Curtain, is it also possible to prove a yaml for that? I tried the MQTT setup, but that one is not working reliable for me.

Ulrar, 2022-06-09:
Hey,

Here's a working config for ESPHome : 

```yaml
substitutions:
  device_name: esp32-relay-master-bedroom
  device_id: esp32_relay_master_bedroom
  sbcmb: SwitchBot Curtain Master Bedroom
  sbcmb_nospace: switchbot_curtain_master_bedroom

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

esphome:
  name: $device_name
  includes:
    - switchbot.h
    - switchbot_curtain.h

external_components:
  - source:
      type: git
      url: https://github.com/Ulrar/esphome-temp.git
      ref: ble_write_action
    components: [ ble_client ]

logger:

ota:
  password: !secret esp32-relay-master-bedroom-ota
  
api:
  password: !secret esp32-relay-master-bedroom-password
  encryption:
    key: !secret esp32-relay-master-bedroom-key

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_pass
  fast_connect: !secret wifi_iot_fast

mdns:
  disabled: true

sensor:
  - platform: template
    name: $sbcmb Right Battery
    id: ${sbcmb_nospace}_right_battery
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
    update_interval: 180s

  - platform: template
    name: $sbcmb Left Battery
    id: ${sbcmb_nospace}_left_battery
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
    update_interval: 180s

cover:
  - platform: template
    id: ${sbcmb_nospace}_cover
    device_class: curtain
[... 95 lines cut ...]
            batteryEnt->publish_state(battLevel);
        }
    }
}
```

You need to define the mac address of your switchbot curtain (and it's groupped friend, if you're using two). I'm using the secrets `curtain_master_bedroom_mac` and `curtain2_master_bedroom_mac` above, with the left curtain being the master that gets the commands.

Place the .h files in your ESPHome directory, or you can also just copy the functions into your yaml directly if you want. I use separate files to make it easier to define a lot of different curtains.
I also have a file for contact sensors, if anyone is interested.

I've been using the above on 3 curtains for a while and it works like a charm, the first time it sends an open / close after startup is a bit slow but afterwards it works well. It's a bit slower than the MQTT solution though since it doesn't keep the connection open, but I believe it'll get the batteries to last longer.
I used nRF Connect on Android to find the mac address of the second switchbot, but you can also ungroup them to find it from the official app.

What works : 
- Open / Close
- The slider reflects the position correctly in home assistant (within 10%)
- The logbook reflects the openning / opened states in real time
- The battery level for both switchbots even when groupped, but you need to find the mac address of both

What doesn't work (can be solved if / when rbaron makes the value field templatable in his ble_client PRs) : 
- Setting the position using the slider. You can only use the full open / close buttons
- The stop button

Hope that helps @TheFes

TheFes, 2022-06-09:
@Ulrar great, thanks! I will give it a try!

TheFes, 2022-06-09:
@Ulrar 
I tried your code, adjusted it a bit to my settings (I only have on swithbot curtrain, so I could get the MAC from the app).
```yaml
substitutions:
  device_name: curtain
  device_id: curtain
  sbcmb: SwitchBot Curtain
  sbcmb_nospace: curtain
  device_type: switchbot

esp32:
  board: lolin32_lite
  framework:
    type: arduino

esphome:
  name: $device_name
  includes:
    - include/switchbot.h
    - include/switchbot_curtain.h

external_components:
  - source:
      type: git
      url: https://github.com/Ulrar/esphome-temp.git
      ref: ble_write_action
    components: [ ble_client ]

logger:

ota:

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: on
  ap:
    ssid: "${device_name} ${device_type} Hotspot"
    password: !secret hotspot_pass

mdns:
  disabled: true

sensor:
  - platform: template
    name: $sbcmb Right Battery
    id: ${sbcmb_nospace}_right_battery
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
    update_interval: 180s

  - platform: template
    name: $sbcmb Left Battery
    id: ${sbcmb_nospace}_left_battery
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
    update_interval: 180s

cover:
  - platform: template
    id: ${sbcmb_nospace}_cover
    device_class: curtain
    name: ${sbcmb}
    optimistic: false
    has_position: true
    open_action:
      - ble_client.ble_write:
          id: ${sbcmb_nospace}
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x00]
    close_action:
      - ble_client.ble_write:
          id: ${sbcmb_nospace}
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x64]

esp32_ble_tracker:
  id: $device_id
  scan_parameters:
    interval: 320ms
    window: 320ms
  on_ble_advertise:
    - mac_address: !secret mac_curtain_floris
      then:
        - lambda: |-
            switchbot_curtain( &x
                              , id(${sbcmb_nospace}_cover)
                              , id(${sbcmb_nospace}_left_battery)
                              );
    - mac_address: !secret mac_curtain_floris
      then:
        - lambda: |-
            switchbot_battery_only(&x, id(${sbcmb_nospace}_right_battery));

ble_client:
  - mac_address: !secret mac_curtain_floris
    id: $sbcmb_nospace
    maintain_connection: false
    
button:
  - platform: restart
    name: Restart $device_name

switch:
  - platform: safe_mode
    name: use_safe_mode
```

I now get this error when I try to validate:
`ID 'curtain' of type esp32_ble_tracker::ESP32BLETracker doesn't inherit from ble_client::BLEClient. Please double check your ID is pointing to the correct value.`

Any idea where this comes from, and how to resolve it?

Ulrar, 2022-06-09:
If you have only one switchbot you should rename one of the battery entities and remove the other one, and remove the second lambda (the battery only one) in the ble advertise callbacks. No need to publish the same value twice.

As for your error I've never seen it, but my bet is it's because you're using `curtain` for both your device id and your curtain ble id. I would change one or the other to have different values, it looks like it's picking up the wrong variable because they have the same name. Just change the device_id to something like `esp32-curtain` for example

TheFes, 2022-06-09:
Thanks, think I got it now.
Uploading in progress.

TheFes, 2022-06-09:
Almost there, one minor issue, everything is reversed.
It shows open when closed, and open closes it, and closing opens it.

I can of course switch the open and close actions, but that won't fix the state.
It is a curtain going from right to left, so basically similar with a the right curtain of a double curtain

Ulrar, 2022-06-09:
Ah, I see. For double curtains the position sent by the bots and what home assistant expect is reversed, which is why I 'invert' it with that line : `float(100 - (data.data[3] & 0b01111111)) / float(100);`

You can probably just use `float(data.data[3] & 0b01111111) / float(100);` instead, I don't have a single curtain bot to test but that should reverse it for you.
You'll probably also have to reverse the conditions after that to make sure the open and close states match.

TheFes, 2022-06-09:
Wait just half a second.. The state is correct, only the actions should be inverted :) I can fix that :)

TheFes, 2022-06-09:
One more question, what does the safe mode do?

Ulrar, 2022-06-09:
Just makes it easier to flash. If you define three ble clients or put too much code in there it won't leave enough resources to do full OTAs, if you start seeing your uploads timeout in the middle just turn safe mode on and retry.
Not sure what it does exactly, I think it prevents most stuff from loading to keep the memory free for the OTA

stich86, 2022-06-09:
@Ulrar why the stop command is not working? if you know.. thx

Ulrar, 2022-06-09:
As far as I can tell the switchbots don't have a `stop` command. I imagine we could implement stop by sending it a command to move to the position it's already at, but we can't do that without templating the `value` field, same as to implement the slider.
Currently rbaron's PR don't implement that field as templetable so we can't do either

stich86, 2022-06-09:
> As far as I can tell the switchbots don't have a `stop` command. I imagine we could implement stop by sending it a command to move to the position it's already at, but we can't do that without [...]

ok thanks for sharing
let's wait :)

Ulrar, 2022-06-09:
> I actually think your code was already correct, I left the inverted calculation in, and the open and close actions are also like you had. Not exactly sure what was wrong the first time, but with a single curtain the icons are also a bit confusing.
> 
> I think it will also differ if you use the bot for a single curtain from left to right, might be that you don't need to invert the state then.

The [documentation](https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/curtain.md) says 
```
The distance between the final position of the device and the origin (%)
```

So presumably it'll work the same regardless of the direction, because the bot knows where the origin is. But I can't test it, just a guess !
For your icons the [esphome template cover doc](https://esphome.io/components/cover/index.html) does say you can define the icon, so you can probably just change the icons it's using to match your setup ?

stich86, 2022-06-09:
> The icons for the open and close action in HA are a bit confusing, because they are made for double curtains. There is no device class for a single curtain, so this will have to do [image]
> 
> Thanks a lot for your help!

I've same behaviour with an awing

TheFes, 2022-06-09:
You can change the device class in HA to better match your cover type.

The switchbot.h and switchbot_curtain.h files are like Ulrar posted. The ESPHome yaml I used is:
Please note I added the inlcluded files in a separate folder in the esphome folder. I also added configuration for the AP and web portal.
```yaml
substitutions:
  device_name: curtain_floris
  device_id: curtain_floris
  sbcmb: SwitchBot Curtain
  sbcmb_nospace: sb_curtain
  device_type: switchbot

esp32:
  board: lolin32_lite
  framework:
    type: arduino

esphome:
  name: $device_name
  includes:
    - include/switchbot.h
    - include/switchbot_curtain.h

external_components:
  - source:
      type: git
      url: https://github.com/Ulrar/esphome-temp.git
      ref: ble_write_action
    components: [ ble_client ]

logger:

ota:

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: on
  ap:
    ssid: "${device_name} ${device_type} Hotspot"
    password: !secret hotspot_pass

web_server:
  port: 80

mdns:
  disabled: true

sensor:
  - platform: template
    name: $sbcmb Battery
    id: ${sbcmb_nospace}_battery
    unit_of_measurement: '%'
    device_class: battery
    accuracy_decimals: 0
    icon: "mdi:battery"
    update_interval: 180s

cover:
  - platform: template
    id: ${sbcmb_nospace}_cover
    device_class: curtain
    name: ${sbcmb}
    optimistic: false
    has_position: true
    open_action:
      - ble_client.ble_write:
          id: ${sbcmb_nospace}
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x00]
    close_action:
      - ble_client.ble_write:
          id: ${sbcmb_nospace}
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B
          value: [0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x64]

esp32_ble_tracker:
  id: $device_id
  scan_parameters:
    interval: 320ms
    window: 320ms
  on_ble_advertise:
    - mac_address: !secret mac_curtain
      then:
        - lambda: |-
            switchbot_curtain( &x
                              , id(${sbcmb_nospace}_cover)
                              , id(${sbcmb_nospace}_battery)
                              );

ble_client:
  - mac_address: !secret mac_curtain
    id: $sbcmb_nospace
    maintain_connection: false
    
button:
  - platform: restart
    name: Restart $device_name

switch:
  - platform: safe_mode
    name: use_safe_mode
```

TheFes, 2022-06-09:
Future project, add this code to the ESP32 I already have in use for my MiScale :)

stich86, 2022-06-09:
> Future project, add this code to the ESP32 I already have in use for my MiScale :)

amazing.. with this integration I can finally merge 3 ESP in just one :D

Ulrar, 2022-06-09:
One caveat is you can only have three ble_client connections per ESP32, I believe. But from testing it seems you can receive as many ble advertisements as you can fit, looks like that's not a limiting factor.
So in my case I'm splitting my switchbot curtains in groups of three per device, and I get my ble advertisement per room on one of them, works well

Ulrar, 2022-06-09:
> Nice work. Any Pull Request anytime soon ;)? Seems like things are working based on some testing.

We just need rbaron's PRs to be merged.
- https://github.com/esphome/esphome/pull/3398
- https://github.com/esphome/esphome/pull/3434

They do work very well, at least for me.
Once that's done maybe we can look into making a switchbot component, but I'm not sure what that entails or if it would make sense for this.

stich86, 2022-06-16:
i've test the integration, it works but it's a little bit slower (about 7s to send the command).

Is it normal?

Ulrar, 2022-06-16:
It's a bit slower for me but only 1 or 2 seconds usually, although sometimes it can take a few more.
That's more a question for rbaron I suppose but I think it's just the time it takes for the esp to establish a connection.

You can make it faster by keeping the connection open (the boolean under ble_client) but that will be at the cost of battery, and like with everything the switchbots randomly disconnect anyway so it's inconsistent. Not sure why they can't stay connected

DerTiger1, 2022-06-18:
@Ulrar, would it be possible to integrate [github://myhomeiot/esphome-components](url) to your [https://github.com/Ulrar/esphome-temp.git](url)? The ble client in your fork is just able to handle 3 ble connections.
Another problem is, that [https://esphome.io/components/esp32_ble_tracker.htm](url) crashes my esp devices, if I try to read ble values from the same devices that are used by ble_client.ble_write:

```
`ble_client:                                                                        
  # switchbot device macaddress
  - mac_address: !secret bot_Markise_ausfahren
    id: switchbot1  #Markise `ausfahren``
```

```
`  - platform: template        #https://github.com/esphome/feature-requests/issues/454#issuecomment-1113985592                                                     
    disabled_by_default: true
    name: "Markise ausfahren [${esp32_name}]"                                                                 
    on_press:                                                                      
      - ble_client.ble_write:                                                      
          id: switchbot1                                                        
          service_uuid: CBA20D00-224D-11E6-9FB8-0002A5D5C51B                       
          characteristic_uuid: CBA20002-224D-11E6-9FB8-0002A5D5C51B                
          # List of bytes to write.                                                
          value: [0x57, 0x01]`
```
 This code causes my devices to crash:
```
`#    - mac_address: !secret bot_Markise_ausfahren #BOT Markise ausfahren
#      then:
#        - lambda: |-
#            for (auto data : x.get_service_datas()) {
#                if(data.data.size() >= 2 ) {
#                  int8_t battery = data.data[2] & 0b01111111;
#                  id (${esp32_name}_battery_bot_markise_ausfahren).publish_state(battery);
#                  }
#                }`
```

But all other switchbot ble readings are working:
```
esp32_ble_tracker: #https://esphome.io/components/esp32_ble_tracker.html war 100
  scan_parameters:
    interval: 320ms #320
    window: 40ms #320
  on_ble_advertise:
    - mac_address: !secret contact_Fenster_Bad #Fenster Bad
      then:
        - lambda: |-
            switchbot_contact( &x
                             , id(${esp32_name}_fenster_bad_contact)
                             , id(${esp32_name}_fenster_bad_motion)
                             , id(${esp32_name}_fenster_bad_battery)
                             );
```

Ulrar, 2022-06-22:
> 

Hey,

Not sure how I'd go about doing that, the changes in that repo aren't mine.
I've just merged together both of @rbaron's PRs and removed the const.py change to make it importable directly from yaml, I think merging it with the component in your link would take actual know how.
That said I do agree that there's no real reason to be limited to 3 devices if they're disconnected between commands, so it'd be neat to have. Maybe rbaron can remove that limit in a future PR, if these get merged some day.

As for the crash the only reason I can see for it is if `data.data[2]` is out of bounds, which seems possible if your device isn't actually a switchbot (or if it's not advertising the battery state there for some reason, old firmware ?).
The condition you have above should probably be `> 2` or `>= 3`, since arrays are indexed from 0. As is it's possible for it to go through the condition and then crash trying to access `data.data[2]` I'd say.

georaspi, 2022-07-09:
Great to see the amazing progress on this guys, many thanks for your effort and looking forward to have that integrated soon in the Esphome release!

[21 more replies]

closed 2022-11-22 by nagyrobi