esphome/feature-requests #456: Support OEM-style Tuya dimmers (labelled QS-WiFi-D01-TRIAC)

tortfeaser, 2019-10-29:
**Describe the problem you have/What new integration you would like**
I'd like it if ESPHome supported Tuya dimmer labelled QS-WiFi-D01-TRIAC

**Please describe your use case for this integration and alternatives you've tried:**
These are affordable, well packaged mosfet (not Triac) dimmers very suitable for home integration with ESPHome. While they work with Tasmota, the ESPHome API makes them a better option for control of these in Home Assistant.

**Additional context**
Tasmota's page for setting these up is at https://blakadder.github.io/templates/qs-wifi_D01_dimmer.html

OttoWinter, 2019-10-30:
According to the website you linked, you can try something like this for your configuration (using a custom float output)

```cpp
// Save this file as qs_wifi_output.h in your config dir
#include "esphome.h"

class QSWiFiOutput : public FloatOutput, public UARTDevice {
 public:
  QSWiFiOutput(UARTComponent *parent) : UARTDevice(parent) {}

  void write_state(float state) override {
    write_byte(0xFF);
    write_byte(0x55);
    write_byte(uint8_t(state * 255));
    write_byte(0x05);
    write_byte(0xDC);
    write_byte(0x0A);
  }
};
```

```yaml
esphome:
  includes:
    - qs_wifi_output.h

logger:
  baud_rate: 0

uart:
  id: uart_bus
  tx_pin: GPIO1
  baud_rate: 9600

output:
- platform: custom
  type: float
  lambda: |-
    auto my_output = new QSWiFiOutput(id(uart_bus));
    return {my_custom};
  outputs:
    id: custom_float

light:
- platform: monochromatic
  output: custom_float
  name: QS WiFi Dimmer
```

tortfeaser, 2019-10-30:
Thanks Otto. I'll give that a go and report back.

tortfeaser, 2019-10-30:
Tried this, got the following compile result:
```
INFO Reading configuration /config/esphome/wifi_dimmer_2.yaml...
...
Compiling /data/wifi_dimmer_2/.pioenvs/wifi_dimmer_2/src/main.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:153:15: error: 'my_custom' was not declared in this scope
       return {my_custom};
               ^
src/main.cpp:153:24: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::output::FloatOutput*>'
       return {my_custom};
                        ^
src/main.cpp:154:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
*** [/data/wifi_dimmer_2/.pioenvs/wifi_dimmer_2/src/main.cpp.o] Error 1
========================= [FAILED] Took 41.91 seconds =========================
```
I think I found the bug though. Here's my attempt at a fix. This compiled:
```
esphome:
  name: wifi_dimmer_2
  platform: ESP8266
  board: esp01_1m
  includes:
    - qs_wifi_output.h

wifi:
  ssid: !secret wifi_SSID
  password: !secret wifi_password

# Enable logging
logger:
  level: VERBOSE
  baud_rate: 0

substitutions:
  hostname: "wifi_dimmer_2"

# Enable Home Assistant API
api:
  password: !secret default_password

ota:
  password: !secret default_password
  
# used just for testing, otherwise no need for it
web_server:
  port: 80

uart:
  id: uart_bus
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600

output:
- platform: custom
  type: float
  lambda: |-
    auto my_output = new QSWiFiOutput(id(uart_bus));
    return {my_output};
#    return {my_custom}; # I think this is wrong, and the line above is correct?
  outputs:
    id: custom_float

light:
- platform: monochromatic
  output: custom_float
  name: $hostname
```
This worked, I can dim and toggle the light from HA.

The remaining functionality available in Tasmota (and OEM) is for a momentary switch connected to GPIO13 counting AC pulses. I'm sure implementing this isn't trivial.

Thanks for your help and for your great product.

DeanoXX, 2019-11-21:
I have been playing with these before - but struggled to flash a couple (when Tuya-convert wasn't working)., Just got a fresh couple. Tuya-converted in moments. This is my yaml.

-Exposes the light to HA
-Push Button - Single Click - Toggle Light
-Push Button - Hold - Cycles through Dim->Bright->Dim etc.

No custom components required

Finally got something to replace old X10 LD11s :-)

`esphome:
  name: esp_dim01
  platform: ESP8266
  board: esp01_1m

wifi:
  domain: .local
  ssid: 'xxxx'
  password: 'xxxx'
  manual_ip:
    static_ip: 192.168.1.69
    gateway: 192.168.1.254
    subnet: 255.255.255.0

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    sensor: ERROR
    duty_cycle: ERROR
    binary_sensor: ERROR
    light: ERROR

# level: VERBOSE
# Enable Home Assistant API
api:

ota:

web_server:

time:
  - platform: homeassistant

substitutions:
  switch_id: "dim_01"

# globals:
# Dummy light brightness tracker Global

globals:
  # Dim direction for Switch 1: 0=Up (brighten) 1=down (dim)
  - id: g_direction_1
    type: int
    restore_value: no
    initial_value: "1"
  # Counter for time pressed for switch 1
  - id: g_counter_1
    type: int
    restore_value: no
    initial_value: "0"
  # initial brightness

# Uart definition to talk to MCU dimmer
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  stop_bits: 1
  baud_rate: 9600

sensor:
  - platform: wifi_signal
    name: "${switch_id} WiFi Signal Sensor"
    update_interval: 60s
  # Primary template sensor to track Brightness of light object for "on_value" sending to MCU dimmer
  - platform: template
    name: "${switch_id} Brightness Sensor"
    id: sensor_g_bright
    internal: true
    update_interval: 20ms
    # Ensure on_value only triggered when brightness (0-255) changes
    filters:
        delta: 0.8
    # Read brightness (0 - 1) from light , convert to (0-255) for MCU
    lambda: |-
      if (id(light_main).remote_values.is_on()) {
        return (int(id(light_main).remote_values.get_brightness() * 255));
      }
      else {
        return 0;
      }
    # On Change send to MCU via UART
    on_value:
      then:
        - uart.write: !lambda |-
            return {0xFF, 0x55, (char) id(sensor_g_bright).state, 0x05, 0xDC, 0x0A};
        - logger.log:
            level: INFO
            format: "Sensor Value Change sent to UART %3.1f"
            args: [ 'id(sensor_g_bright).state' ]
  # Sensor to detect button push (via duty_cycle of 50hz mains signal)
  - platform: duty_cycle
    pin: GPIO13
    internal: true
    id: sensor_push_switch
    name: "${switch_id} Sensor Push Switch"
    update_interval: 20ms

binary_sensor:
  #Binary sensor (on/off) which reads duty_cyle sensor readings.
  - platform: template
    id: switch1
    internal: true
    name: "${switch_id} Switch Binary Sensor"
    # read duty_cycle, convert to on/off
    lambda: |-
      if (id(sensor_push_switch).state < 95.0) {
        return true;
      } else {
        return false;
      }
    # Short Click - toggle light only
    on_click:
      max_length: 300ms
      then:
        light.toggle: light_main
    # Generic On_Press - log press, toggle DIM Direction and reset press interval counter
    on_press:
      then:
        - logger.log: "Switch 1 Press"
        - lambda: |-
            if (id(g_direction_1) == 0) {
              id(g_direction_1) = 1;
            } else {
              id(g_direction_1) = 0;
            }
            id(g_counter_1) = 0;

# Dummy light output to allow creation of light object
output:
  - platform: esp8266_pwm
    pin: GPIO14
    frequency: 800 Hz
    id: dummy_pwm

# Primary Light object exposed to HA
light:
  - platform: monochromatic
    default_transition_length: 20ms
    name: "${switch_id} Light"
    output: dummy_pwm
    id: light_main

switch:
  - platform: restart
    name: "${switch_id} Restart"

# Polling object for long press handling of switch for dim/brighten cycle
interval:
  - interval: 20ms
    then:
      - if:
          condition:
            binary_sensor.is_on: switch1
          then:
            # Ramp rate for dim is product of interval (20ms) * number of intervals
            # Every 20ms Dimmer is increased/decreased by 2/255
            # Lower limit = 10%
            # Upper limit = 100%
            # 100% - 10% = 90% = 230/255. Therefore 230/2 * 20ms = 2.3 seconds for full range
            # At full/min brightness - further 16x20ms = 0.32 Seconds "dwell" by resetting counter to 0
            # Initial pause for 16x20ms = 0.32s to allow "on_click" to be discounted 1st
            # g_direction_1 = 0 (Increasing brightness)
            # g_direction_1 = 1 (decreasing brightness)
            # g_counter_1 = Interval pulse counter

            lambda: |-
                    float curr_bright = id(light_main).remote_values.get_brightness();
                    id(g_counter_1) += 1; 

                    // If max bright, change direction
                    if (curr_bright >= 0.999 && id(g_direction_1) == 0) {
                      id(g_direction_1) = 1;
                      id(g_counter_1) = 0;
                    }
                    
                    // If below min_bright, change direction
                    if (curr_bright < 0.1 && id(g_direction_1) == 1) {
                      id(g_direction_1) = 0;
                      id(g_counter_1) = 0;
                    }

                    if (id(g_direction_1) == 0 && id(g_counter_1) > 15) {
                      // Increase Bright
                      auto call = id(light_main).turn_on();
                      call.set_brightness(curr_bright + (2.0/255.0));
                      call.perform();
                    }

                    else if(id(g_direction_1) == 1 && id(g_counter_1) > 15) {
                      // Decrease Bright
                      auto call = id(light_main).turn_on();
                      call.set_brightness(curr_bright - (2.0/255.0));
                      call.perform();
                    }`

tortfeaser, 2020-03-16:
Hi @DeanoXX . This is great. 

I'm going to try and change the behaviour so it when you hold the switch it either increases or decreases to the max/min and stops there rather than going 'around the top'. 

Do you have a method to expose the dimmer brightness to HA? It would be good to be able to set brightness with a HA automation or from the HA interface. HA doesn't seem to have a value for "${switch_id} Brightness Sensor"

DeanoXX, 2020-03-16:
The config uses the light object "light_main" to expose the light to HA - once added in HA via the ESPHome integration as normal, you can control/query the light as you would any other HA light.

To remove the "pause and change direction" just remove/comment the 2 "change direction" sections...
```
              // If max bright, change direction
              if (curr_bright >= 0.999 && id(g_direction_1) == 0) {
                id(g_direction_1) = 1;
                id(g_counter_1) = 0;
              }

              // If below min_bright, change direction
              if (curr_bright < 0.1 && id(g_direction_1) == 1) {
                id(g_direction_1) = 0;
                id(g_counter_1) = 0;
              }

```

tortfeaser, 2020-03-16:
Thanks @DeanoXX 

I can't get that lot to compile, I'm not sure about the syntax error.

Here's what I have:

```
interval:
  - interval: 25ms
    then:
      - if:
          condition:
            binary_sensor.is_on: switch1
          then:
            # Ramp rate for dim is product of interval (25ms) * number of intervals
            # Every 25ms Dimmer is increased/decreased by 2/255
            # Lower limit = 10%
            # Upper limit = 100%
            # 100% - 10% = 90% = 230/255. Therefore 230/2 * 25ms = 2.9 seconds for full range
            # At full/min brightness - further 16x25ms = 0.4 Seconds "dwell" by resetting counter to 0
            # Initial pause for 16x25ms = 0.4s to allow "on_click" to be discounted 1st
            # g_direction_1 = 0 (Increasing brightness)
            # g_direction_1 = 1 (decreasing brightness)
            # g_counter_1 = Interval pulse counter

            lambda: |-
              float curr_bright = id(light_main).remote_values.get_brightness();
              id(g_counter_1) += 1; 

            #  // If max bright, change direction
            #  if (curr_bright >= 0.999 && id(g_direction_1) == 0) {
            #    id(g_direction_1) = 1;
            #    id(g_counter_1) = 0;
            #  }
            #
            #  // If below min_bright, change direction
            #  if (curr_bright < 0.1 && id(g_direction_1) == 1) {
            #    id(g_direction_1) = 0;
            #    id(g_counter_1) = 0;
            #  }

              if (id(g_direction_1) == 0 && id(g_counter_1) > 15) {
                // Increase Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright + (2.0/255.0));
                call.perform();
              }

              else if(id(g_direction_1) == 1 && id(g_counter_1) > 15) {
                // Decrease Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright - (2.0/255.0));
                call.perform();
              }
```

But that gives the error:
[image]

Any ideas?

DeanoXX, 2020-03-17:
Everything in the lambda block is passed straight through to the C compiler. "#" is not a valid comment - so you need either a C-style comment block with
```/*
Comments
*/
```
or preface each line with "//" for a single line C comment rather "#" for the YAML comment

(I agree that the "#" YAML comment should be filtered from the code passed to the Compiler..has caught me several times - but it isn't)

tortfeaser, 2020-03-17:
Thanks for that. Sorry.

tortfeaser, 2020-03-18:
So just commenting those sections out worked to stop it going 'around the top', but holding the button while it dimmed resulted in the light going to 0% then back to 100% and down again.

I fiddled around and came up with this, which shows the behaviour I was after:
```
            lambda: |-
              float curr_bright = id(light_main).remote_values.get_brightness();
              id(g_counter_1) += 1; 

              // If below min_bright, change direction
              if (curr_bright < 0.1 && id(g_direction_1) == 1) {
                auto call = id(light_main).turn_on();
                call.set_brightness(0.1);
                call.perform();
                id(g_counter_1) = 0;
              }

              if (id(g_direction_1) == 0 && id(g_counter_1) > 15) {
                // Increase Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright + (2.0/255.0));
                call.perform();
              }

              else if(id(g_direction_1) == 1 && id(g_counter_1) > 15) {
                // Decrease Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright - (2.0/255.0));
                call.perform();
              }
```

tortfeaser, 2020-04-02:
This works great now. I've installed these for five circuits in an extension I've built. They all work fine, except when I had to shutdown the Home Assistant server. Then each of the dimmers turned off after about 10 minutes. I use the switch to turn them on again, HA still shutoff, and they turn off without me using the switch after 10 minutes.

Can this behaviour be fixed in the code? It would be good for the family to have the lights be operable by switches as normal as possible if the HA server goes down.

Here's what I'm using:
```
esphome:
  name: wifi_dimmer_5
  platform: ESP8266
  board: esp01_1m

substitutions:
  device_name: "wifi_dimmer_5"

wifi:
  ssid: !secret wifi_SSID
  password: !secret wifi_password
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name} Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    sensor: ERROR
    duty_cycle: ERROR
#    binary_sensor: ERROR
#    light: ERROR

# Enable Home Assistant API
api:
  password: !secret default_password

ota:
  password: !secret default_password

# used just for testing, otherwise no need for it
web_server:
  port: 80

time:
  - platform: homeassistant

# globals:
# Dummy light brightness tracker Global

globals:
  # Dim direction for Switch 1: 0=Up (brighten) 1=down (dim)
  - id: g_direction_1
    type: int
    restore_value: no
    initial_value: "0"
  # Counter for time pressed for switch 1
  - id: g_counter_1
    type: int
    restore_value: no
    initial_value: "0"
  # initial brightness

# Uart definition to talk to MCU dimmer
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  stop_bits: 1
  baud_rate: 9600
[... 108 lines cut ...]
              float curr_bright = id(light_main).remote_values.get_brightness();
              id(g_counter_1) += 1; 

              // If below min_bright, change direction
              if (curr_bright < 0.1 && id(g_direction_1) == 1) {
                auto call = id(light_main).turn_on();
                call.set_brightness(0.1);
                call.perform();
                id(g_counter_1) = 0;
              }

              if (id(g_direction_1) == 0 && id(g_counter_1) > 15) {
                // Increase Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright + (2.0/255.0));
                call.perform();
              }

              else if(id(g_direction_1) == 1 && id(g_counter_1) > 15) {
                // Decrease Bright
                auto call = id(light_main).turn_on();
                call.set_brightness(curr_bright - (2.0/255.0));
                call.perform();
              }
```

dclobato, 2020-08-21:
If I´m not wrong, when you use API, if the module lost connection with HA for a period of time (don´t remember 5 or 10min), it will reboot.  Maybe it´s what´s happing with you

nerdosity, 2021-02-16:
https://esphome.io/components/esphome.html#esp8266-restore-from-flash this could help, I think.

closed 2022-07-05 by nagyrobi