esphome/feature-requests #457: Add pzem004t total energy value (Energy field)

rradar, 2019-10-31:
**Describe the problem you have/What new integration you would like**

It would be great (after pzem004t support arrived :rocket:) to also have support for the **total energy value** which is stored in the pzem004t and doesn't reset with a power cut.

This is a super useful value. It's included in all versions of the pzem004t (the old v1+2 and the new v3). I use it to 'sync' my pzem004t to my mains energy meter :man_scientist: 

**Please describe your use case for this integration and alternatives you've tried:**

I tried to get the same information with the integration sensor. Problem is that I have my `update_interval:` set to `1s` for the pzem004t component to get real time information. When I add a `restore: true` to the `integration` component I'm worried it will wear out the flash from the esp really quick because it will write the value every second to the flash of the esp (if I understand the docs right)

**Additional context**

https://github.com/esphome/feature-requests/issues/49

rradar, 2019-10-31:
Actually I really tried to get this working with the integration

```
  - platform: integration
    name: "${upper_devicename} Energy Meter" 
    sensor: pzem_power
    time_unit: h
    restore: false #don't! pzem_power is updated every second!
    filters:
      # Multiplication factor from W to kW is 0.001
      # Energy Meter reads 2272kWh right now
      - lambda: return x * (0.001) + 2272;
    unit_of_measurement: kWh
    icon: mdi:counter
```

But the output isn't useful. Don't know why their is spikes and why it actually lowers the value again. :1234: 

[image]

My old setup with espurna (already replaced by esphome) was using the information directly from the pzem004t and reporting it via mqtt. It just goes one way :arrow_up: 

[image]

OttoWinter, 2019-10-31:
Thanks for creating a new feature request for this!

To answer some of the issues here:

> When I add a restore: true to the integration component I'm worried it will wear out the flash from the esp really quick because it will write the value every second to the flash of the esp

If this is an ESP8266 and you don't explicitly set `esp8266_restore_from_flash` to true in your config, it will be stored in RTC memory which does not have issues with wearing out the flash.

I don't know if the PZEM004T stores the accumulated value in a non-volatile area too (you can check by disconnecting+reconnecting to power). But if it does, it will have around the same write cycles as the ESP8266 because the technology is quite standard.

--------

As for the graphs you're seeing:

 - The first big jump appears to be adjusting the constant in the lambda.
 - The up/down thingies are concerning - but I would say that points more to a bug that can be fixed. My guess would be that it's in the PZEM data reading, could you attach a graph of the PZEM power value during that timeframe?

rradar, 2019-11-01:
Thank's for your reply @OttoWinter ! Actually I just found out that I use still the custom component for the pzem004t v3 I'm using right now. I will get it running with latest esphome beta, test it again and report here. I also have a pzem004t v2 to help with testing!

rradar, 2019-11-03:
Ok, I did upgraded my sensor yesterday to 1.14.0 using the pzemac component. Here are some graphs:

[image]
[image]
[image]
[image]

and my yaml part:

```
sensor:
  - platform: pzemac
    frequency:
      name: "${upper_devicename} Frequency"
      unit_of_measurement: hz
    voltage:
      name: "${upper_devicename} Voltage"
    current:
      name: "${upper_devicename} Current"
    power_factor:
      name: "${upper_devicename} Powerfactor"
    power:
      name: "${upper_devicename} Power"
      id: pzem_power
    update_interval: 5s

  - platform: total_daily_energy
    name: "${upper_devicename} Total Daily Energy"
    power_id: pzem_power
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
    unit_of_measurement: kWh
    icon: mdi:counter

  - platform: integration
    name: "${upper_devicename} Energy Meter" 
    sensor: pzem_power
    time_unit: h
    filters:
     # Multiplication factor from W to kW is 0.001
      - lambda: return x * (0.001) + 2282;
    unit_of_measurement: kWh
    icon: mdi:counter
```

for me it looks good now. Still I think this feature would improve the accuracy dramatically. Because the pzem004t is not limited to a 5 second (or 1 second) read interval and I guess the total energy captured directly inside the pzem004t is therefor much more accurate.

SeByDocKy, 2019-11-03:
Definitively the real energy field should be directly integrated

r-jordan, 2019-11-28:
It would be also great to add the energy sensor for the PZEM-004T v3.0 (pzemac).
Is there any way to add the action for resetting it (command 0x42)?

lettore, 2020-03-18:
Any news about reading the energy meter directly from the PZEM-004T?
Previously I had a Mysensor board attached to the PZEM and I was able to get Voltage, Current, Active Power and Energy.
As you already done the first 3, it should be easy to add also the Energy value.
Here the protocol I used in my Arduino Sketch, don't know if it will help
[https://github.com/mysensors-rus/PZEM004T](url)

SeByDocKy, 2020-03-18:
Have been integrated from last dev build... Works fine for me.

pietrozanc, 2020-03-23:
Hi guys it will be possible to implement the reading of the total energy also on the oldest version of pzem004t (v1) ?

gromar88, 2020-05-07:
The energy value in new dev version works great - thank You! I just wonder how I should modify code to get energy value in kWh since it shows in Wh.

closed 2020-05-26 by glmnet