esphome/feature-requests #461: Custom save to flash interval for Total Daily Energy/Integration and others?

rradar, 2019-11-03:
**Describe the problem you have/What new integration you would like**

While using one of the widely available power monitor plugs (sonoff pow r2, blitzwolf shp6 or a esp+pzem004t) and adding a total daily energy component or a integration component like this:

```
sensor:
  - platform: total_daily_energy
    name: "${upper_devicename} Total Daily Energy"
    power_id: power

  - platform: integration
    name: "${upper_devicename} Total Energy" 
    sensor: pzem_power
    time_unit: h
```

always results in a rest of the values when the device is upgraded or power cycled.
[image]

This is the **default behavior** because save to flash is disabled.

Problem is that I'm forced to use the same `update_interval` to save this values to flash as the power monitoring component has set. For example using `pzemac` (the same is true for all other components like `hlw8012` or `cse7766`)

```
sensor:
  - platform: pzemac
    power:
      name: "${upper_devicename} Power"
      id: power
    update_interval: 1s
```

enabling _`esp8266-restore-from-flash`_ (https://esphome.io/components/esphome.html#esp8266-restore-from-flash) and set the `restore` to `true`:

```
sensor:
  - platform: integration
    name: "${upper_devicename} Total Energy" 
    sensor: pzem_power
    time_unit: h
    restore: true
```

will result that the value is written every second to flash.

To don't wear out the flash to quickly (the esphome docs mention around 100 000 write cycles. This would result in only **28 hours** when writing every second) but still get a real time feed of the data over to home assistant a option set the save interval would be very useful.

Something like this:
```
sensor:
  - platform: integration
    name: "${upper_devicename} Total Energy" 
    sensor: pzem_power
    time_unit: h
    restore: true
    save_every: 1h
```

The expected live of the flash module on the esp would then be **over 10 years!**

I guess this could be also good for other sensor which produces a lot's of output.

**Please describe your use case for this integration and alternatives you've tried:**
Some hardware like the pzem004t have a energy usage counter included: #457 

**Additional context**

michalsteyn, 2019-12-13:
Very much looking for a solution to this too!

glmnet, 2019-12-21:
I believe a more general flash update rate with some kind of sliding expiration is more suitable. 

Components will instruct if they are saving periodic data or a switch state 

Also it’d be nice to have a write rate protection, ie if more than 40 writes in a minute then the writer would error and disable itself.

glmnet, 2019-12-27:
> **Additional context**
> 
> One more thing would be to be able to call the saving of data to flash with a switch or something. So just before the ota hit the save to flash button and their will be almost no loss of data/values...
> 
> Or directly including it in the ota routine? If saving to flash is enabled and a ota is coming in, first thing (before accepting ota) is to save all values to flash.... then ota... then restore... profit!

This should be working already, ESP8266 will always save state to the non wearable so called RTC memory when rebooting so this should work. Can you confirm otherwise?

rradar, 2019-12-30:
> This should be working already, ESP8266 will always save state to the non wearable so called RTC memory when rebooting so this should work. Can you confirm otherwise?

As you can see in the picture above the total_daily_energy sensor didn't survived the upgrade:

[image]

But I'm not sure right now what module it was. It could have been a esp-m3 (esp8285) module which is known to sometimes need a hard power cycle to upgrade. The soft reset (which is the only reboot the RTC memory survives if I'm right) does sometimes not work proper with esp8285's from my experience.

marc-gist, 2020-02-08:
I'm also interested in some kind of solution. Would this work on an ESP32?
Another idea would be to have the value be recovered from a retained MQTT or other external source. that way it won't have any possibility of wearing out flash.

rdehuyss, 2020-02-13:
What about storing to flash ONLY when power loss is detected? It can easily be done using a resistor and 2 capacitors.

See https://www.youtube.com/watch?v=i2YsKacRZOs

benneq, 2021-01-30:
Is there any internal function to trigger the "save to flash"?
I'd like to have a Button (binary_sensor), that should be pressed to save the current values to flash. Could be simply done using a on_press lambda function that executes the "save to flash" function.

jhenkens, 2021-06-23:
@benneq I like that idea.

I have no existing knowledge of the underlying persistence code, but my naive thoughts were that the following would be ideal:

1. Allow persisting either every value, or manually via an action call.
1. Allow persisting on a per-component level, either to rtc memory or flash+rtc
1. Persist time written
1. When restoring on boot, can check value from rtc and flash, and restore newest value

My use case that I'm looking at - a run-hour meter for an engine monitor I'm developing. Could hack together some stuff via globals and templated sensors, and get close but not perfect to the above. I care about "engine hours", so persisting constant values isn't strictly necessary (some data loss is okay), but less loss is better!

Maybe I'll take a look at the code and see what's doable. I've got a few other things I'm looking at too within esphome, so it might fall behind those.

Getslow6, 2022-10-28:
I have solved this issue as follows:

```
sensor:
- platform: pulse_counter
  pin: 
    number: ${GPIO_sensorpulse} # Any digital input
    # inverted: true
  id: water_pulsecounter
  unit_of_measurement: 'L/min'
  accuracy_decimals: 0
  name: 'Water flow'
  update_interval: 10s
  total:
    unit_of_measurement: 'L'
    name: 'Total water usage'
    accuracy_decimals: 0
    icon: "mdi:water"
    device_class: water
    id: water_total

# The water total will be restored from this global value.
# This global value will be stored in flash when the device shuts down
# The total value will be restored using the global value on startup
# By using the global variable we can reduce the amount of writes to the flash.
globals:
  - id: global_watertotal
    type: int
    restore_value: true

esphome:
  on_boot:
    then:
      - pulse_counter.set_total_pulses:
          id: water_pulsecounter
          value: !lambda 'return id(global_watertotal);'
  on_shutdown:
    then:
      - globals.set:
          id: global_watertotal
          value: !lambda 'return id(water_total).state;'

# Set pulse counter total from home assistant using this service call:
api:
  services:
    - service: set_total_water_usage
      variables:
        water_meter_reading: int
      then:
        - pulse_counter.set_total_pulses:
            id: water_pulsecounter
            value: !lambda 'return water_meter_reading;'
```

ssieb, 2022-10-28:
You should also use an `interval:` to save the value regularly, as you can't guarantee that you'll get a nice reboot.

flaviut, 2023-01-18:
Looks like this has been implemented in https://github.com/esphome/esphome/pull/2119!

[Documentation & sample config](https://esphome.io/components/esphome.html#adjusting-flash-writes).

You will need to [set `restore: true` on your `integration` sensor](https://esphome.io/components/sensor/integration.html#configuration-variables). If you [are using `total_daily_energy`, `restore` is enabled by default](https://esphome.io/components/sensor/total_daily_energy.html#configuration-variables).

closed 2023-01-18 by flaviut