esphome/feature-requests #467: Expose binary sensor press/release/click/double_click/etc automations as HA events

brandond, 2019-11-09:
**Describe the problem you have/What new integration you would like**
I would like to be able to use different types of clicks as automation triggers in HA

**Please describe your use case for this integration and alternatives you've tried:**
The HA Xiaomi Binary Sensor integration exposes button click details. The switches will all fire xiaomi_aqara.click events with a click_type field that can be used to determine the type of the button press - long_click_press, long_click_release, hold, single, double for single switches; or single, double, long, both, double_both, long_both for double-switch devices.

I would like to use similar functionality for ESPHome binary sensors, preferably without having to manually configure each individual click_type as a binary sensor automation. Since I am a MQTT user, I would like this to be supported for both connectivity types (MQTT and API).

**Additional context**
https://www.home-assistant.io/integrations/binary_sensor.xiaomi_aqara/
https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/xiaomi_aqara/binary_sensor.py#L430

OttoWinter, 2019-11-09:
Kind of a duplicate of https://github.com/esphome/esphome/issues/115

I don't think that solution would be true to ESPHome's idea of extensibility.

Yes, I understand being able to parse button press types is something useful, but:

 - This would need to be opt-in - you don't want your MQTT binary sensors pushing data to some new topic if you haven't explicitly enabled it to.

    - But then, if you need to add stuff to each binary_sensor anyway, why not create an `on_click`/etc that you copy to each binary sensor.

    - I know the issue then is of course that you'd need to *copy* a piece of YAML to each binary sensor - but that means the base issue is the missing YAML templatization, and not binary sensors.

 - There's no way to generate events in HA from MQTT - you can listen for mqtt messages on a topic though, but that means user-action as they need to figure out what topic it's sending to.

 - In order for this to be in the core binary sensor model, there would need to be some sort of event bus like in HA - but there is no such thing in ESPHome, so that would need to be done first.

brandond, 2019-11-09:
I was hoping there was an existing JSON schema for sending events (as opposed to states) to HA via MQTT. If not I guess that would be something to put in a PR for in HA, and then use here?

brandond, 2019-11-10:
Related: #472

bdr99, 2023-12-18:
Home Assistant now supports event entities as a cleaner and more user-friendly way to expose events. Event entities would be the best way to implement this feature.