esphome/feature-requests #468: Adaptive thresholds for ESP32 Touch Pad component

loongyh, 2019-11-09:
**Describe the problem you have/What new integration you would like**

The threshold for the `esp32_touch` component can only be set once during initialization. My application requires sensing through 3mm plastic and another 3mm thick glass and as such even after setting the `low/high_voltage_reference` options to maximize sensitivity, the raw sensor values during idle and touched state still differs by only around 10. As the idle state raw sensor values are quite stable (deviation of +/-2), I set the threshold at the [(lowest observed idle state value) - 2] and it works fine.

However the problem comes when the environment changes - if the air-conditioning is on for example, humidity drops and that causes the idle state value to rise substantially, causing the touchpad to no longer respond to taps.
In ESP-IDF, the threshold can be set on runtime with the `touch_pad_set_thresh()` function. Perhaps the threshold can be made to adapt to environmental changes with the following algorithm on a loop:
1. Poll the raw sensor value every second and record it.
2. Once 10 consecutive readings within +/-2 are recorded, set the [(lowest recorded reading) - 2] as the new threshold with `touch_pad_set_thresh()`.
Of course, the above values need not be hardcoded and can be configurable in the yaml.

**Please describe your use case for this integration and alternatives you've tried:**

Tried experimenting with all the config variables of the esp32_touch component, to no avail.

**Additional context**

With this algorithm, the touchpad threshold will be self-calibrating and adaptive over time and there will not be a need to manually set the threshold to a fixed value.

loongyh, 2019-11-09:
Seems like PR [#631](https://github.com/esphome/esphome/pull/631) already added `set_threshold` and `get_value` methods.

torfbolt, 2019-11-13:
I would propose using an exponential moving average & variance for this, instead of a highest frequency filter.
The loop would look something like this:

```
mean = (1 - alpha) * mean + alpha * x
variance = (1 - alpha) * variance + alpha * (x - mean)^2
sigma = variance^0.5
output = abs(x - mean) > factor * sigma
```

Alpha is the smoothing constant and can be derived from `sample_period / smoothing_period`.
The factor defines the threshold, something like 6 should be a sensible default for a 3.5ppm false positive rate (assuming Gaussian distribution).

Maybe this could also be a general binary sensor component that uses the values of an analog sensor as input.

torfbolt, 2019-11-14:
If you do not want to have dependencies to math libraries, this can also be simplified to

```
mean += alpha * (x - mean)
variance += alpha * ((x - mean) * (x - mean) - variance)
output = (x - mean) * (x - mean) > factor * factor * variance
```

torfbolt, 2019-11-14:
To avoid the problem of touch events seriously messing with the mean and variance values, it might make sense to use a median filter instead of a mean filter. In that case the loop would look like this:

```
median += (x > median) ? epsilon_m : -epsilon_m
stddev += ((x - median) * (x - median) > stddev * stddev) ? epsilon_s : -min(epsilon_s, stddev)
threshold = median - factor * stddev
```

The problem is then to find good initialization values. I would propose an initialization period where we use mean filtering as above with a relatively large alpha defined by the init period length and suppress events:

```
alpha = 2 * sample_period / init_period
threshold = 0
```

After that we can properly initialize the median filter:

```
median = mean
epsilon_m = median * sample_period / filter_period
stddev = sqrt(variance)
epsilon_s = stddev * sample_period / filter_period
```

Of course this could also be just a method for auto-defaults and the user can override the values for `epsilon_m` & `epsilon_s`

OttoWinter, 2019-11-14:
@torfbolt Interesting median calculating, I've not heard of that method before. If I understand correctly the median is calculated by successively moving the median "closer" to the value.

Ended up implementing the mean+variance method and it's working well - maybe even too well lol: it triggers when my fingers even get close to the pad (but not touching it), changing the factor for dynamic offset calculation would probably solve that.

As for initialization values: Ideally we wouldn't have any - problem with initialization sequence is that every time the device reboots it takes some time for touch detection to work.

Ultimately though mean and median are mostly the same thing - mean just moves "faster" to the current value when it changes rapidly whereas the median algorithm you mentioned would move at a constant rate.

All solutions will by definition behave weird in one condition: the touch pad is held down for a long time. After a long enough time, any algorithm is going to think the new value means the touch pad is inactive. So it's expected that scenario will break adaptive threshold - but it's unlikely to occur IRL anyway.

torfbolt, 2019-11-14:
Yep, maybe for this use case it makes more sense to use a user-defined fixed offset. If we use the filter just for the median, we can instantly initialize

```
epsilon_m = x_0 * sample_period / filter_period
median = x_0
```

 after the first measurement. The constant change rate is what makes the median filter more robust to outliers.

Of course all of these algorithms require that event time << filter time constant. But I think that is an acceptable limitation. The user would then just have to supply offset & filter time constant as parameters.

loongyh, 2019-11-15:
Wow ok just read through both your replies and it took some time to digest. I don't have a background in statistics so not sure I fully understand how both your proposed algorithms work, however I'll go ahead to test and follow up with some results :)

> The mode (what this implementation is currently using) is just not a good property to use for these statistics. Consider this: During the interval, the user touches the pad for some time, and half the values are now around 800 and the other half around 850.

Even though my current implementation seems to work on the surface, from the home assistant history logs, I did notice two light switches in unoccupied rooms were switching on and then off by itself at random times even after setting the offset to a point where the sensitivity is significantly affected. Unless freaking people out is part of the feature, perhaps using just the mode is indeed not a good way to do adaptive thresholds.

[image]

> Ended up implementing the mean+variance method and it's working well - maybe even too well lol: it triggers when my fingers even get close to the pad (but not touching it), changing the factor for dynamic offset calculation would probably solve that.

That's exactly what my use case require, through 3mm plastic (the led backlight diffuser) and 3mm glass. For my case, the main challenge when working with this level of sensitivity is dealing with the noise.

> Another factor that should be considered here is number of variables. If we have lots of variables the user can adjust to tweak adaptive threshold, that will be poor user experience.
>
> Ideally, we'd have
>
> as little variables as possible
> ... that are easy to understand
> ... have defaults that work well
> ... still work fine for detecting touches

Fully agree! Initially I intended to keep things simple, thus assumed a simple mode filter would suffice. It's too simple in fact that I could write up the [docs](https://github.com/esphome/esphome-docs/pull/412) within the same day. While both your proposed algorithms may work much better, it seems to be for now, understanding which configuration parameters to input or even how it works at all does require some background in statistics. I didn't even know what standard deviation was yesterday. I foresee writing a comprehensible doc may be yet another challenge ;)

Ok I'll get to work on testing both your algorithms now... Be right back with the results

loongyh, 2019-11-16:
I've finished [implementing the exponential mean algorithm](https://github.com/esphome/esphome/compare/dev...loongyh:esp32touch-adaptive-thresholds-exp-mean), however I'm unsure how max_variance is derived so I set it to be the previous variance +10%. Did I do the implementation correctly? A set of 10 readings is collected over a 10 second period to calculate mean and variance to decide if threshold would be adjusted, and set the log to print out variance and max_variance every calculation iteration to see how it changes. Tested it out on one of my switches with `offset: 7`:

```
[15:19:24][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:19:24][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:25][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:25][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:25][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:19:26][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:26][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:26][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:26][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 805
[15:19:27][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:27][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:27][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:27][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 805
[15:19:28][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:28][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:28][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:28][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:29][D][esp32_touch:185]: PrevMaxVariance: 4
[15:19:29][D][esp32_touch:186]: Variance: 4
[15:19:29][D][esp32_touch:190]: NewMaxVariance: 4
[15:19:29][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:29][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 805
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 807
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:33][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:34][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:19:34][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 805
[15:19:34][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:34][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 807
[15:19:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:19:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:19:40][D][esp32_touch:185]: PrevMaxVariance: 4
[15:19:40][D][esp32_touch:186]: Variance: 4
[15:19:40][D][esp32_touch:190]: NewMaxVariance: 4
[15:19:40][D][esp32_touch:141]: New threshold set: 797
[15:19:40][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:19:40][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[... 226 lines cut ...]
[15:20:34][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 802
[15:20:35][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:36][D][esp32_touch:185]: PrevMaxVariance: 2
[15:20:36][D][esp32_touch:186]: Variance: 2
[15:20:36][D][esp32_touch:190]: NewMaxVariance: 2
[15:20:36][D][esp32_touch:141]: New threshold set: 797
[15:20:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:36][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[... cut ...]
[15:20:37][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 803
[15:20:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 805
[15:20:38][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
[15:20:39][D][esp32_touch:129]: Touch Pad 'touch_1' (T4): 806
```

I'll roll it out to the rest of the switches and test it over a day or two.

loongyh, 2019-11-17:
Seems like it doesn't work - some lights still got triggered on by itself - especially when the air-conditioning is switched on.

OttoWinter, 2019-11-17:
I'll try to answer the questions later, but just FYI this is the diff I tested with (includes some cleanup too). Currently uses EMA for mean/variance but changing to median shouldn't be hard.

```diff
diff --git a/esphome/components/esp32_touch/binary_sensor.py b/esphome/components/esp32_touch/binary_sensor.py
index ceb48171..f2e3609e 100644
--- a/esphome/components/esp32_touch/binary_sensor.py
+++ b/esphome/components/esp32_touch/binary_sensor.py
@@ -3,7 +3,7 @@ import esphome.config_validation as cv
 from esphome.components import binary_sensor
 from esphome.const import (CONF_NAME, CONF_PIN, CONF_THRESHOLD, CONF_ADAPTIVE_THRESHOLD,
                            CONF_OFFSET, CONF_INTERVAL, CONF_SAMPLES, ESP_PLATFORM_ESP32,
-                           CONF_ID)
+                           CONF_ID, CONF_ALPHA)
 from esphome.pins import validate_gpio_pin
 from . import esp32_touch_ns, ESP32TouchComponent
 
@@ -35,30 +35,34 @@ def validate_touch_pad(value):
 
 ESP32TouchBinarySensor = esp32_touch_ns.class_('ESP32TouchBinarySensor', binary_sensor.BinarySensor)
 
+CONF_MAX_STDDEV = 'max_stddev'
+CONF_SENSITIVITY = 'sensitivity'
 ADAPTIVE_THRESHOLD_SCHEMA = cv.Schema({
-    cv.Required(CONF_OFFSET): cv.positive_int,
-    cv.Optional(CONF_INTERVAL, default='2s'): cv.positive_time_period_seconds,
-    cv.Optional(CONF_SAMPLES, default=10): cv.positive_not_null_int
+    cv.Optional(CONF_ALPHA, default=0.01): cv.zero_to_one_float,
+    cv.Optional(CONF_MAX_STDDEV, default=10): cv.positive_float,
+    cv.Optional(CONF_SENSITIVITY, default=10): cv.positive_float,
 })
 
 CONFIG_SCHEMA = binary_sensor.BINARY_SENSOR_SCHEMA.extend({
     cv.GenerateID(): cv.declare_id(ESP32TouchBinarySensor),
     cv.GenerateID(CONF_ESP32_TOUCH_ID): cv.use_id(ESP32TouchComponent),
     cv.Required(CONF_PIN): validate_touch_pad,
-    cv.Exclusive(CONF_THRESHOLD, 'threshold_group'): cv.uint16_t,
-    cv.Exclusive(CONF_ADAPTIVE_THRESHOLD, 'threshold_group'): ADAPTIVE_THRESHOLD_SCHEMA
+    cv.Required(CONF_THRESHOLD): cv.uint16_t,
+    cv.Optional(CONF_ADAPTIVE_THRESHOLD): ADAPTIVE_THRESHOLD_SCHEMA,
 })
 
 
 def to_code(config):
     hub = yield cg.get_variable(config[CONF_ESP32_TOUCH_ID])
-    if CONF_ADAPTIVE_THRESHOLD in config:
-        conf = config[CONF_ADAPTIVE_THRESHOLD]
-        var = cg.new_Pvariable(config[CONF_ID], config[CONF_NAME], TOUCH_PADS[config[CONF_PIN]],
-                               conf.get(CONF_OFFSET), conf.get(CONF_INTERVAL),
-                               conf.get(CONF_SAMPLES))
-    else:
-        var = cg.new_Pvariable(config[CONF_ID], config[CONF_NAME], TOUCH_PADS[config[CONF_PIN]],
-                               config[CONF_THRESHOLD])
+    var = cg.new_Pvariable(config[CONF_ID])
     yield binary_sensor.register_binary_sensor(var, config)
     cg.add(hub.register_touch_pad(var))
+    cg.add(var.set_touch_pad(TOUCH_PADS[config[CONF_PIN]]))
+    cg.add(var.set_threshold(config[CONF_THRESHOLD]))
+
+    if CONF_ADAPTIVE_THRESHOLD in config:
+        conf = config[CONF_ADAPTIVE_THRESHOLD]
+        cg.add(var.set_adaptive_alpha(conf[CONF_ALPHA]))
+        stddev = conf[CONF_MAX_STDDEV]
+        cg.add(var.set_adaptive_max_variance(stddev*stddev))
+        cg.add(var.set_adaptive_sensitivity(conf[CONF_SENSITIVITY]))
diff --git a/esphome/components/esp32_touch/esp32_touch.cpp b/esphome/components/esp32_touch/esp32_touch.cpp
index 099ef1e2..252fe0ec 100644
--- a/esphome/components/esp32_touch/esp32_touch.cpp
+++ b/esphome/components/esp32_touch/esp32_touch.cpp
@@ -1,5 +1,6 @@
 #include "esp32_touch.h"
[... 181 lines cut ...]
 
+  void process_(uint16_t value, bool should_print, bool setup_mode);
+
   touch_pad_t touch_pad_;
   uint16_t threshold_{};
   uint16_t value_;
-  uint16_t offset_;
-  uint16_t interval_;
-  uint16_t sample_size_{};
-  uint16_t count_{};
-  uint32_t last_run_{};
-  std::unordered_map<uint16_t, uint16_t> samples_;
+
+  // Values used for adaptive threshold
+  float adaptive_mean_ = 0.0f;
+  float adaptive_variance_ = 0.0f;
+  // Config values for adaptive threshold
+  float adaptive_sensitivity_ = 0.0f;
+  float adaptive_alpha_ = 0.0f;
+  float adaptive_max_variance_ = 0.0f;
 };
 
 }  // namespace esp32_touch

```

loongyh, 2019-11-18:
Just tested the above diff on one of my light switches. I needed to set sensitivity to around 2-3 for it to register touches. Problem is I can't get the light to stop turning on and off even after trying different values of alpha and max_stddev.

torfbolt, 2019-11-18:
For the touch sensor it probably makes sense to only use a fixed (configurable) offset for the detection threshold and the above code only for the adaptive mean value, so without using the variance at all. And of course pretty small values for alpha, I would start testing with something like 1e-3.

loongyh, 2020-07-06:
Just an update - I have since switched out to external hardware capacitive sensors and thus abandoned this. Anyone interested to continue this please do so.

smurfix, 2021-05-15:
It might make sense to simply have two mean values. One long term with a time constant of a minute or so, and one short term with a time constant of a quarter second. Then
```
if (touch) {
    if (long_avg-short_avg < thresh_off)
        touch = false
} else if (long_avg-short_avg >= thresh_on)
    touch = true
```
with `thresh_off<thresh_on` of course, so that there's some hysteresis here.