esphome/feature-requests #470: Support for TuyaMCU non dimmer devices

danpowell88, 2019-11-09:
**Describe the problem you have/What new integration you would like**
ESPHome currently does not support TuyaMCU non dimmer devices

**Please describe your use case for this integration and alternatives you've tried:**
There are a number of devices such as fan controllers that use TuyaMCU that will not work using the dimmer component

**Additional context**
Tasmota recently started supporting TuyaMCU devices that werent dimmers such as fan switches etc.

https://github.com/arendst/Tasmota/wiki/TuyaMCU#tuyasend
https://github.com/arendst/Tasmota/wiki/TuyaMCU#tuyareceived

OttoWinter, 2019-11-09:
Related: https://github.com/esphome/esphome/pull/820#issuecomment-548618214 and https://github.com/esphome/esphome/pull/820#issuecomment-548980938

brandond, 2019-11-10:
I don't have a non-switch permanently powered device but I might pick one up. @danpowell88 is there a particular one you've got in mind?

danpowell88, 2019-11-10:
I have some of these 

https://www.brilliantsmart.com.au/wp-content/uploads/2019/08/Smart-Remote-99111.pdf

Ive managed to figure out the command with Tasmota here (https://community.home-assistant.io/t/australia-electrically-certified-hardware/32074/787?u=thecubical)

and also one of 

https://www.ebay.com.au/itm/Smart-WIFI-Touch-Home-Automation-6-Speed-Fan-Wall-Switch-Google-Home-Alexa-IFTT/123618202331
[image]

and the tasmota config here (https://community.home-assistant.io/t/australia-electrically-certified-hardware/32074/787?u=thecubical)

danpowell88, 2019-11-10:
@OttoWinter not sure if I should raise a separate ticket as well, support for MCU based power monitoring would be awesome as well

danpowell88, 2019-11-25:
Even support for light switches that dont have a dimmer would be quite useful!

paravoid, 2019-12-30:
If I may expand the scope a little bit…

Tasmota's wiki has some [documentation](https://tasmota.github.io/docs/#/TuyaMCU?id=tuya-protocols) pieced together from various sources regarding their protocol and their basic data types. With a cursory look, it looks like ESPHome actually does have some of the fundamentals of the protocol implemented (booleans, ints, enums, but not strings) but not exposed to users. It looks like these can only used by the lights module and in a very specific configuration.

Howver, Tuya's WiFi modules can be used to control pretty much any type of device, and with all kinds of function mapping (and even custom functions). From what I could piece together on the web and e.g. from [Tuya's own developer docs](https://docs.tuya.com/en/iot/device-development/access-mode-mcu/develop-introduction-wifi-/wifi-mcu-overview) they provide a panel where implementors can implement all kinds of different configurations for their devices and to communicate with their own MCUs. For example, see:
[image]
[image]

More concretely: I own a dehumidifier that uses a Tuya module (which from what I gather, it's pretty common; this was one of 3-4 brands in the local appliances store that used Tuya). I managed to run tuya-convert and deploy its intermediate firmware on it successfully, and I'm thus now able to flash a custom firmware. However, while ESPHome, Tasmota and Espurna all implement (parts of) the Tuya protocol, none allow the flexibility of configuring a device with multiple controls. They mostly support light switches and lightbulbs and would thus be unable to control the device.

It'd be neat if ESPHome supported exposing these primitive data types directly to YAML, so that users could configure their own device accordingly. Bonus points if there is an easy way to figure out what these commands are, e.g. through logging :)

patfelst, 2020-02-04:
I would also like this to be updated to work with non-dimmable devices - thanks! 

I manged to get it working on a 2-gang [Elite Glass Wall Switch](https://www.brilliantsmart.com.au/smart-products/electrical/smart-elite/). It works fine in Hassio if I set `dimmer_datapoint: 0`, **except that the control has a redundant dimmer slider**. 

Here's the relevant part of my ESPHome yaml:
```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

light:
  - platform: "tuya"
    name: "Elite Wall Plate Sw 1"
    dimmer_datapoint: 0
    switch_datapoint: 1
  - platform: "tuya"
    name: "Elite Wall Plate Sw 2"
    dimmer_datapoint: 0
    switch_datapoint: 2
```

grea09, 2020-03-13:
Ehlo, I see that there's recent development on that front. I looked around on the Tasmota side and I would like to improve that platform so that I can make esphome work with my diffuser (light color and all). Where should I start ? Thanks for the awesome work so far !

rmooreID, 2020-09-25:
> More concretely: I own a dehumidifier that uses a Tuya module (which from what I gather, it's pretty common; this was one of 3-4 brands in the local appliances store that used Tuya). I managed to [...]

I am also hoping to control a dehumidifier with a Tuya module. I am a beginner at coding but I've got a couple years of experience at using ESPHome for connecting devices to Home Assistant.

I'm just starting to research what it would take to get a usable Tuya dehumidifier integration started. I can't imagine I'll be much help on the coding side but I'd be glad to help out however I can whether it's testing or verifying that features work, etc.

I only just started researching this today but here's the documentation that gave me some hope: https://tasmota.github.io/docs/TuyaMCU/#dehumidifier

And the e-etailer website says: "Vacplus dehumidifier provides the WIFI remote via the Tuyasmart APP" which tells me that the unit is able to run some Tasmota-type custom firmware, yeah?

shockerengr, 2021-01-13:
I'll add to this, I recently purchased a generic relay board [https://www.amazon.com/gp/product/B089QCT8L5/] that uses the tuya MCU to control the relays.  I've been able to get it controlled, but it's pretty clunky.
One of the datapoint id's is an ENUM that controls the relay behavior between normal, inching and interlock.  So far, the only way I've been able to set the ENUM to a defined state (to ensure consistent behavior) was to define a TUYA FAN component, with the switch definition going to a bad address.  Then I could use the "speed" setting to backdoor an ENUM value to the MCU.

just allowing ability to send arbitrary values to the different MCU id's would probably cover a lot of these new use cases, the underlying communication looks to be there.

sjwright, 2021-04-21:
> just allowing ability to send arbitrary values to the different MCU id's would probably cover a lot of these new use cases, the underlying communication looks to be there.

I'd wish to second this. I'm tinkering with a Tuya-based kettle and being able to send arbitrary values to an enum would be extremely useful—even if it was the most basic scaffold available through a template / lambda. Thanks!

sjwright, 2021-04-21:
For anyone who wants to set enums, here's a disastrously hacky way of doing it:

```    turn_on_action:
      - lambda: |-
          tuya::TuyaDatapoint datapoint{};
          datapoint.id = 4;
          datapoint.type = tuya::TuyaDatapointType::ENUM;
          datapoint.value_enum = 0x2;
          tuya_tuya->set_datapoint_value(datapoint);
```

In the above example, datapoint 4 is being set to an enum value of 2 (0x2). Again, this is a terrible hack and could stop working at any time because it relies upon aspects of the underlying C++ code which is subject to change. But it does work right now.

jasimancas, 2021-06-28:
Hi, it's possible add this device?
https://templates.blakadder.com/zemismart_YH002.html

scrhall, 2021-08-01:
I have a difuser, with a rgb ligth, (14char datapoint), it is posible to configure?

Nardol, 2021-08-16:
It might be good to have enum supported alone. For example with my Maxcio diffuser I have an enum to define timer option.
For the moment I have set it as a sensor and change the value using the function provided by @sjwright but having it as a simple enum sensor would allow a clean use.
Thanks @sjwright for the tip :slightly_smiling_face:

Nardol, 2021-08-18:
Oops, the `lambda: tuya_tuya->set_datapoint_value` does not work anymore since ESPHome 2021.8

Nardol, 2021-08-19:
Sorry for multi-comment, but it looks like a tuya enum device would be really necessary.

Until now, I used a tuya sensor for the enum but since esphome/esphome#2059 it throws the following error each time I switch my diffuser off:
```txt
[10:26:26][E][tuya:504]: Attempt to set datapoint 103 with incorrect type
```

Which is logical but could be avoided if enum were not limited as they are for now.

closed 2022-07-01 by nagyrobi